Ralim/IronOS #551: TS100 heats up then cools immediately.

Red-Fox25, 2020-01-17:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**

* **What is the current behavior?**
When I plug my TS100 into a fully charged 3s or 4s battery (that's all I own) it will heat up to the maximum temperature like it should but will only stay there for a second or two before immediately cooling down

Detailed screen next to the voltage also changes to VBA but I have been unable to locate what that means if anything at all.

* **What is the expected behavior?**
iron stays hot so I can actually solder a joint.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1.
2.
3.

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.07
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information**

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2020-01-17:
Hi,
A bit more information is required here:

* Is the soldering iron exiting to the main menu?
* Does the screen go black?
* Does the iron warn about tip disconnection?
* Does the iron show the low input voltage warning?
* What is your input voltage during this?
* Is your battery voltage dropping too low?
* What is the current rating of your battery?
* What temperature are you going up to ?

Red-Fox25, 2020-01-17:
Is the soldering iron exiting to the main menu? everything seems to work fine
Does the screen go black? no
Does the iron warn about tip disconnection? no 
Does the iron show the low input voltage warning? no
What is your input voltage during this? 12.1 on 3s and 16.2 on 4s
Is your battery voltage dropping too low? I don't think it is, it stays at 12v
What is the current rating of your battery? 50c
What temperature are you going up to ? 850

Red-Fox25, 2020-01-17:
https://www.youtube.com/watch?v=QYJPme2jiBU&feature=youtu.be

once it says VBA next to the voltage it has started to cool off.

Ralim, 2020-01-17:
Looks like you are trying to drive the tip all the way up to 450C. Can you please test a sane temperature first (~660F).

Red-Fox25, 2020-01-17:
that was apparently the issue since it seems to go way above where it's set, it hovers around 700 when it's set to 660

Ralim, 2020-01-17:
Someone else has reported a bug in the U.S.A conversions (fahrenheit) which could be contributing.
Most likely you are asking your iron to go higher than the system can do with the current calibration and range.

whitequark, 2020-01-17:
I just hit the same issue, with a 20 V power source rated at 45 W. I have the temperature set to 450 °C so it's not a problem with Fahrenheit conversions. What does "VBA" mean?

Also, if I press a button once the screen is frozen with the "VBA" indication, the iron starts working properly again, except the problem happens once again above around 420 °C. Seems to work fine with lower temperatures.

whitequark, 2020-01-17:
I think it might be very briefly exceeding the rating of the battery at these high temperatures and then crashing, since I can see it get quite close to 45 W when above 420 °C. That's fine, but the "VBA" (or is it "BA" next to "V"?) should more clearly explain the nature of the problem.

whitehoose, 2020-01-17:
I'd be interested to know what you are soldering and with what solder.
Or are you just winding the poor thing up to see at what point the tears start (in the interest of science). 

End of day its your iron therefore your rules, but a little bit of me wants to know if it's experimentation (first thing I do with a new car is wind it up until the wheels sound like a strangled cat  - or just that you've never had the benefit of experienced mentoring.

I started soldering in 1968 with an old electric iron and started work in 72 when I "did" a course at the local tech college.  We used a stove and copper slug attached to an iron stake.  
There was a lot of talk about starting low.  it takes practice to judge temp based on how the solder reacts to the hot tip.

We did have electric  thermostatic irons - they were mechanical and not particularly accurate. but once set it stayed at temp quite nicely,  But most irons were designed to operate at "the ideal temp" there were no adjustments, just faith that the manufacturer had got it right.

I had a gas stove and my Iron was a 6oz slug of copper.  Thermostatic control was done by eye and smell.  If you needed more "oomph" you increased the thermal mass of the tip (the biggest we had was 10 oz), not it's temperature.
We worked to an ideal of 330 +/- 10c.
Typical temperatures for 37/63 or 60/40 lead/tin are around 180c melt (see the data sheet) that gives a working temp anywhere from 300-350c.  In a real world - the thermometer is only a (very) rough approximation due to many environmental reasons.

An exact reading isn't particularly important as long as you know what you're looking at.  Thin tracks and small components/pads with a low thermal mass wont soak up the heat while a massy slug of copper (I use a 20g slug of copper as my "standard") on the ts80.  You can see the ripples form as the iron works to maintain it's settings.  Then the puddle relaxes and the surface smooths as the solder melds with the copper's molecules, the solder climbs the wire to form a "cone" round it and the job's done.

Most solder manufacturers (leaded and lead free) would be wriggling with discomfort thinking about >400c for real world application.  I'd certainly fail labwork on the sort of kit I regard as "typical"
In the field I'd still be uncomfortable about leaving a potential fail on a real job (not so much a potential as a dead cert its not going to see the summer).

whitequark, 2020-01-17:
> I'd be interested to know what you are soldering and with what solder.

It's not soldering exactly. I was desoldering a device in a [TI's USON package](https://www.ti.com/lit/ds/symlink/sn74lvc1t45.pdf#page=31). It's a tiny 1x1 mm thing that was soldered using lead-free on a board that heatsinked pretty badly. I would normally remove it with hot air, but that specific board had plastic connectors that would melt. In those cases I set the iron to somewhere around 400..450°C, add a blob of solder on a tip like BC2, and heat up the entire part harshly. That works very well.

I had to do similar things to remove an [ONSemi UDFN18](https://eu.mouser.com/datasheet/2/308/MG2040-D-107505.pdf#page=8) package, that is only doable with a KF tip. It was right adjacent to a plastic connector so I couldn't have done it with hot air even with a lot of masking.

This of course stresses the tip a lot and should not be done for a long time. I do this rarely and only for a few seconds at a time. But when I do it, it's indispensable.

Red-Fox25, 2020-01-17:
> I'd be interested to know what you are soldering and with what solder.
> Or are you just winding the poor thing up to see at what point the tears start (in the interest of science).
>
> End of day its [...]

Honestly I find this comment kind of silly. If there is absolutely no reason to ever turn the temp up that high then why would anyone even bother programming it to be able to do so? It's a brand new iron and I'm just playing around with it to see what it can do.

closed 2020-01-18 by Ralim

whitehoose, 2020-01-18:
Red
 _I find this comment kind of silly. If there is absolutely no reason to ever turn the temp up that high then why would anyone even bother programming it to be able to do so?_

I did make provision for the joy of just seeing what happens in the first paragraph, that's human nature  - we've all tried the meltdown setting to see if it will.

As I explained, Turning it up wasn't an option to most until relatively recently - I'm a part time instructor - and since it became an option it's becoming worryingly common for people to routinely crank the volume up to 11 or 19 on the dial  because of course more is always better!!
You see it in comments here too - a worrying number of would be techs are routinely running their irons above 400c.  When vibration is a major factor, adding excessive heat isn't a good idea  

_why would anyone even bother programming it to be able to do so?_
At work we've asked this too - because we're seeing the number of failed repairs due to brittle unions increasing too.  I was always taught that if you need more thermal penetration you increase the tip mass not the heat.
Desoldering is a different issue - it always takes more to soften the site - but you then turn it down again when fixing in replacement components. 

Just because you can, doesn't necessarily mean you should.
I was just curious to see who was asking.  Thanks for your input.

PR #554 (2020-01-18): Fix: Remove tip disconnection warning. Fix leading zeros (again)
Please try and fill out this template where possible, not all fields are required and can be removed.

* **Please check if the PR fulfills these requirements**
- [x] The commit message makes sense
- [x] The changes have been tested locally
- [ ] Are there any breaking changes

* **What kind of change does this PR introduce?**
(Bug fix, feature, docs update, ...)
Fix:
Remove tip disconnection warning. 
Fix leading zeros (again)

Closes #551 

* **What is the current behavior?**

Setting the target temperature too high can cause th [... cut]