Ralim/IronOS #555: Might it be possible to prevent accidental cross-flashing?

Nazosan, 2020-01-18:
(Sorry I lost the template by accidentally clearing it.  And GitHub is being really nice and saving my blank draft for me so I can't restore it.  This is a sort of mix of a bug report in that it doesn't limit and feature request in that I think it should.)

I got a little too enthusiastic upon reading the description for this latest firmware and downloaded and flashed too quickly without paying attention.  In so doing I accidentally flashed a TS100 firmware onto my TS80.  Luckily this is not a fatal thing, but since the display comes on and shows stuff it almost seems like it worked ok which could potentially confuse the issue in diagnosing it.  Some sort of super simple checking mechanism either in the flash itself or on startup (maybe a configuration option that says "is TS80" or "is TS100" producing an error if the wrong one is set?) producing an error onscreen and not attempting to continue incorrectly?

Obviously this is user error and not high priority, but perhaps something to consider at least.

Ralim, 2020-01-18:
Yeah this is something that could be implemented, though is it really worth it is a good question?
As it causes no harm I haven't focused on it. But if people wanted, it could be added.

Technetium1, 2020-01-18:
If there is an option to avoid bricking devices and it's not a lot of effort I think it's worth adding.

Ralim, 2020-01-18:
@Technetium1 It wont ever brick the device per se, just doesnt let you heat up the tip as the tip heater control is on a different pin.

Nazosan, 2020-01-19:
Well, I can testify firsthand that it at least didn't brick mine this time.  I do wonder if it's not impossible some feature in a future firmware could adjust voltage a certain way or something like that and potentially have the ability to harm hardware, but I'm no expert on what is involved and couldn't say for sure.  It just struck me that the potential may exist for this someday mostly.

closed 2020-05-29 by Ralim