Ralim/IronOS #556: Soldering iron tip becomes red hot

whitequark, 2020-01-18:
* **I'm submitting a ...**
  - [x] Bug report

* **What is the current behavior?**
I reflashed the iron and went to see if #551 was fixed. I set the temperature to 450 °C (and in case it's relevant, I configured the power limit to 45 W, which is what my power supply is rated at). The temperature rapidly climbed to about 420 °C and then slowed down to a very shallow slope (about 1 °C per 5 s) while the power stayed at around 30 W. Once the (displayed) temperature was about 425 °C, I noticed the tip was red hot and brightly glowing red, well visible under my bright work light. I did not have a pyrometer at hand but by the color of the glow I estimate the actual temperature was around 650-700 °C.

* **What is the expected behavior?**
Soldering iron tip does not become red hot.

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.08
  - PCB Version: (isn't shown per instructions above?)
  - Power Supply: 20 V, 2.25 A

whitequark, 2020-01-18:
In case you're wondering, the tip actually started sugaring at the shoulder between the shaped end of the tip and the cylindrical portion, with the "blisters" characteristic for stainless steel overheated in air, and a layer of green patina. In spite of that I managed to recover it by some vigorous tinning, so it's still perfectly usable. That was an exciting experience though :)

Ralim, 2020-01-18:
Eugh, sorry 😬 
Actually saw this on twitter before the email :|.

I'll dig a bit more into it and get a fix out soon.

In explanation of whats happening:

Because the tip temp is effectively thermocouple feedback + handle temp, it will rise to approximately the ADC maximum (i.e. maximum that can be measured), and then can rise slightly as handle (cold junction) temp also rises.

I *thought* I had this constrained in my testing here. I'll get back to you when I have a patch figured out.

whitequark, 2020-01-18:
> Eugh, sorry 
> Actually saw this on twitter before the email :|.

I don't mind. Bugs happen. This one didn't even cost me any tips (well, I guess the life of that BC2 is reduced, but probably not by much).

I'm actually impressed just **how** hot it got. I now want to take a tip and heat it until the point of destruction while measuring the temperature. Is it actually going to melt the stainless at one point?

> Because the tip temp is effectively thermocouple feedback + handle temp, it will rise to approximately the ADC maximum (i.e. maximum that can be measured), and then can rise slightly as handle (cold junction) temp also rises.

Does this mean TS100 just can't get any higher than 420°C around room temp, and 400°C at zero (probably the lowest temp it makes sense to support)? I imagine then limiting the temperature selector to 420°C at most and also throttling the loop at ADC maximum makes the most sense, but you understand this better.

Ralim, 2020-01-18:
>  I now want to take a tip and heat it until the point of destruction while measuring the temperature. Is it actually going to melt the stainless at one point?

Dont know if it will melt the stainless, but they do get a *very* nice cherry red very quickly. Haven't yet been game to destroy a tip just yet.

> Does this mean TS100 just can't get any higher than 420°C around room temp, and 400°C at zero (probably the lowest temp it makes sense to support)

This would be most likely the path I'll go down long term (limiting to 420°C). However at the moment, i felt keeping the 450 as the max was easier. What the actual max is per iron also depends on the ADC offset calibration, and the tip type itself (not all tip types have the same slope, but most of them are quite close, especially the Miniware ones). The non-linear Hakko tip does let you get slightly higher on the same hardware technically.

I've added ADC limiting and fixed a race condition now :) If you have some time could you sanity check this (attached) fixes the issue for you. It works for me, here, but then again I missed the issue the first time around...
[TS100_red_fix.zip](https://github.com/Ralim/ts100/files/4080146/TS100_red_fix.zip)

whitequark, 2020-01-18:
> If you have some time could you sanity check this (attached) fixes the issue for you.

Works for me.

PR #558 (2020-01-18): Fix Acting as red light
Please try and fill out this template where possible, not all fields are required and can be removed.

* **Please check if the PR fulfills these requirements**
- [X] The commit message makes sense
- [X] The changes have been tested locally
- [ ] Are there any breaking changes

* **What kind of change does this PR introduce?**
(Bug fix, feature, docs update, ...)
Fix ability to overdrive output and cook the tip into a red light source

Fixes #556 
Fixes #545
closed 2020-01-18 by Ralim