Ralim/IronOS #564: Compatible Device(s) [Not a bug]

jellewie, 2020-01-21:
This is just a few adapters that support/ not support the latest TS-80 firmware, I really like how the [WIKI](https://github.com/Ralim/ts100/wiki/TS-80) states what adapters work with the firmware.
So I tasked myself to try every QC3 adapter I will find from now, and post my results.

QC3 is stated to be 9V3A or 12V1.5A max (that's 18Watt, so that is the benchmark to beat)

- [SDC-18W](https://i.imgur.com/tRSfW4Q.jpg) china adapter came with TS80 - 9.89V2.41A=24Watt

- [3q52 cat adapter](https://i.imgur.com/SiSTxHI.jpg) - NO 12V!!, 9.15V*2.15A=20Watt
The TS80 will restart if you try to heat up on 12V with this adapter.

ScruffR, 2020-04-03:
This "issue" seems the closest I found for reporting supplies/powerbanks that work or not.

I tested a Charmast 10400mAh power bank which turns of 20-30 seconds after sleep mode has been engaged or when a lower temperature is selected and the cooldown takes more than 20/30 seconds.

I also tested a Omars Powerbank 24000mAh which even switches off while the device *keeps* the set temperature (~3W power draw ?!?!?!)

jellewie, 2020-04-03:
Xstorm voyager QC3 output is max 25Watt (else the powerbank triggers short circuit protection) 9V is more stable than 12. Also the powerbank has low load cutoff, so when the Ts80 isn't heating for a while the powerbank cuts al the voltage. The powerbank also has a boot time; 5V1A max for 2 minutes, then restarts with QC and PD enabled (this will turn of the iron again..)

jellewie, 2020-04-03:
I bought a aliexpress DC to QC3 board, this was eble to power the bank at max power, (given the input can handle it). As suggested somewhere in this github wiki thingy. 
I combined this with a 20V PD adapter (so this module would convert PD to QC) the only down side note is that the PD module would step down and the closer the voltage the better (20-12 worked, but to 9 it was getting toasty) 

QC3.0 QC2.0 USB DC-DC Buck Converter Charging Step Down Module 6-32V 9V 12V 24V to Fast Quick Charger Circuit Board 3V 5V
https://a.aliexpress.com/_B1QCxD

whitehoose, 2020-04-03:
The sleep "thing" is always going to be a problem.  The painful fact is that powerbanks are designed principally as support for phones and other devices that take a consistent reasonably high current. 
They arent intended to be used for the very different intermittent consumption an iron has in normal use.

It would actually make more sense to follow the design intention of powerbanks and use your powerbank to extend the running time of an an intermediate reservoir of power (like an s4) and in turn use that to satisfy the iron's intermittent demands.

I use an AOKoda or ISDT BattGo BG-8S Smart Battery Checker with 4s batteries.  Both adaptors can control up to 6s which would cover both the TS80 and TS100 voltage ranges.  You can use them either as a QC3 controller or just as a voltage conditioner, so if something goes wrong the Iron should survive.  Not the only solution by a long shot - but I know they are both compatible with my kit and use patterns.
the ISDT is a bit expensive so really the AOK is the best all round solution - and its 100% compatible with both stock and ralim firmwares.

From a pure power delivery point of view - it's always going to be easier to buck than boost.
Even the bigger capacity powerbanks start as only 3.7v cells which are boosted either to 5v or qc3 so any subsequent processing is done with a pulsed DC supply carrying a pulse frequency of anything from 20kHz up.  I have a couple of cheap USB2 banks where the output frequency of the bank heterodynes with my attempts to connect a second boost device to get 12v, the sum of the oscillators cause a low frequency pulsed output. which is useless unless you want a flashy LED.
Built in qc3 - 9v or 12v can confuse matters more.  Its little wonder there are so many problems.

It doesn't take much maths to see that persuading a 3.7v source to chuck out 20v2A takes a lot of doing and few powerbanks can deliver the currents required to support both an iron and all the extra circuitry.
Coasting downhill from a true steady DC  14.8v or 22.2v is a far easier conversion task.

jellewie, 2020-04-03:
> The sleep "thing" is always going to be a problem. The painful fact is that powerbanks are designed principally as support for phones and other devices that take a consistent reasonably high current.

I've had powerbanks before without the sleep issue (old, lost them, and no QC)
for example all my T-Link powerbanks didn't had this 'smart' feature and the output was always 5V. So there might be some out there that do work.

Also a lot of powerbanks have weird issues, like the Xstorm I mentioned, this doesn't boot into QC on a cold boot right away. I just want to get these things into a picture. I only brought mine since the store employee allowed me to test it.

jellewie, 2020-04-03:
#592  RAVPower PD Pioneer Model: RP-PB172 only with Custom by Ralim

whitehoose, 2020-04-03:
I realised what you're doing - and its worth doing.  the trouble is smart will become commonplace - I just bought a bag of powerbank shells £0.99p each.  the box is battery shaped and the pcb has protection, charging AND a torch :+1: they are much better value than the stock protection boards on fleabay - however this latest batch are all "smart" and fall asleep at the drop of a hat with no override (that I've found so far).  You can workround by adding a LED and switch to prevent it dropping off - but it increases the cost and wastes power.  The story on QC3 is that many chinesium mfrs have reverse engineered qc3 to avoid paying qualcomm royalties - the result is its turned compatibility into a bit of a lottery and its not going to be fixed.

The sleep option is an excellent idea ... if you're passing on a big chunk of pixies and want to screw every last one out of a charge - I imagine if we were having this conversation on a phone forum wed be saying what a good idea it is.

I got my ts80 specifically to be portable, I've worked out of a van for many years and my solution (not the only one) was lead acid cells in a backpack - lots of magic smoke for a days work out and about.
I hoped the TS80 and a powerbank would help my aching back ... it didn't.  4s is a little lighter .. it just has a tendency to go nuclear if damaged.

The drone crew already have the power and have adopted the TS irons without issues.  Once you have a qc3 controller the lipo cells are much cheaper per Ah than most high capacity powerbanks and once you have that controller they are more versatile.

Anyone setting up now should seriously consider this approach before spending load of cash on kit that may not work ... its not for everyone, if you've already got a drawer full of banks its not quite the bargain - but it's worked for me without problems for best part of a year.
Another alternative is to use powertool batteries, most QC3 adapters are 7-36v - so construction workers who already use the batteries can add a ts80 to a system they've already adopted.

DebtFromAbove, 2021-02-09:
Hi. I can confirm that Original TS80 works with the RavPower PD Pioneer 2000mAh power bank work with Ralim firmware 2.14.1, using the non-Power Delivery USB-A port, the iron successfully negotiates QC3.0 and starts heating at 9V. The stock firmware (v1.06 and v1.07) did not work with this power bank. The model number of this power bank is RP-PB201. Great f/w. Thanks so much

closed 2023-07-21 by Ralim