Ralim/IronOS #565: Cooked thermocouple.

RVduino, 2020-01-23:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
While uploading the TS100_EN firmware I ended up with a fried soldering tip whch seems to now have a bad thermocouple. Not sure how the temp calibration works but I deleted your firmware upgrade and reloaded the original v2.18 from the miniware site and my other tips still work fine. Seems like your firmware cooked my soldering tip.

* **What is the expected behavior?**

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1.
2.
3.

***Video of problem if hard to reproduce***
When you power up the OLED displays Sens-Error with the bad tip. 

[image]

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information**

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2020-01-23:
Hi,
This sounds like you have damaged your tip.
My firmware cannot really do anything to the internal hardware to make it fail any more or less than the default firmware.

Can you please specify which firmware build of mine you uploaded to the unit, what temperature were you trying to operate at, what power supply source are you using?

Most likely, somehow you have damaged the tips internal junction, you can verify this by measuring the resistance across the two contacts on the tip with an ohmmeter, it should read ~8-8.5 ohms.
If its reading open circuit/really high its most likely your heating coil has failed open. So far this has only occurred when operating above recommended temperatures ( > 400C) or using higher voltage sources than recommended which can push too much current through the tip.

RVduino, 2020-01-23:
Thanks for responding. Perhaps it's just a really odd coincidence? The tip is def bad and the thermocouple resistance is jumping all over the place and mostly open. I've been using this tip for a  few weeks now and I was working through the Temp Calibration in the latest version 2.08.1 when this happened. Perhaps I somehow overadjusted the temp offset or messed things up in the Temp Cal function? I have no idea, but I wasn't aware at any time that it got overheated. In fact I never could get it to heat up at all with your firmware. But I think that might be due to the damaged tip. But still not sure how it got damaged, the connection lugs look fine and I've tried to spin it around etc. No go. I reloaded the old firmware and as soon as it tries to heat up the bad tip the temp reads 400C and it shuts off and gives the Sens Error. I am using the 19V power supply that came with it. When I put other tips in with the old firmware it heats up fine. I'm a little paranoid to try new firmware now since I seemed to find a way to kill my tips.

Ralim, 2020-01-23:
Could be odd coincidence

Temp calibration does not require applying heat to the tip at all, it measure the room temperature measurement of the tip and the handle to calibrate.

If you did in fact manage to get a really wrong offset into the temp calibration its possible you drove the tip into an overheat, but i've taken many tips to glowing red hot without damage ?

fricy, 2020-02-21:
Is this bug supposed to be fixed in 2.08.1?
Because I also have a cooked BC2 tip. I was operating it at 300 C, and did not notice the tip being hotter than usual, let alone red hot. It was idling for a couple of minutes in the holder, and when I picked up it was reading 415 C, and wouldn't respond to adjustments. When I replace the tip I had to temporary revert to the original Mini fw for the iron to work, calibration or factory reset did not work for some reason. Now I'm back to 2.07 with a different tip, and it seems to work.
The fried tip is open circuit, the good tip reads ~7.7 ohms.

On a side note: odd coincidence, but auto rotate doesn't work on my iron. However when it fried the tip, before I did the factory & fw reset I noticed that autorotate started to work for me. First time ever. And now it's again back to the default "Nope, you don't need the autorotate m8." Very weird, thought I might mention it. PCB v1.
Oh, and  the tip wasn't an original mini, so it could have been a low quality part, and it just died on me by pure chance. No idea. Wasn't used too much, and mostly around 300 C with an ocassional boost here and then.

closed 2020-11-01 by Ralim