Ralim/IronOS #567: TS100 does not heat up above 410-420 degrees

bodja, 2020-01-27:
* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**
Soldering Iron does not heat above 410-420 degrees though 450 is the target.

* **What is the expected behavior?**
Should be possible to heat up to 450.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***
1. Install https://github.com/Ralim/ts100/releases/download/v2.08.1/TS100_EN.hex
2. Power the soldering iron.
3. Set the temperature to 450.
4. Wait until the indicator shows that tip is ready.
5. Check the temperature.

* **What are you running:**
  - Firmware Version: 2.08.1
  - PCB Version: how can I figure it out?
  - Power Supply (Voltage and Current Rating) : 20V 4.5A

Ralim, 2020-01-31:
This is depending on how the current code calculates the maximum temperature that the ADC can measure.

I need to check this on other tips before i say too much more (could be a code bug), but at least on my personal main iron, this is where the reading range on the iron actually runs out.

metal03326, 2020-03-27:
I have exactly the same issue with TS80. In previous firmware the issue was even bigger. Even though temperature was set to 450, it reached only 410-420 and the tip started glowing hot. This version removed the glowing, but temperature never reaches 450 and stays 410-420.

Firebie, 2020-03-28:
Ralim is using next formula for tip calc in Celsius:
"(TipVoltage - CalibrationOffset)/24.9 + HandleTemp", where CalibrationOffset is [600..1000], HandleTemp in [20..35].
From my point of view, this has to be changed to:
"(TipVoltage - TipOffset)/22.0", where TipOffset is [200..500].

Ralim, 2020-03-29:
@Firebie  Why did you remove the HandleTemp in your version?

Firebie, 2020-03-29:
On my checks - on your stock firmware - during calibration - HandleTemp is 20C, later, during soldering run, HandleTemp is around 35C degree. And I don't see why it has to affect calculation of TipTemp in soldering session.
But it can be used to find TipOffset during equilibrium calibration:
(TipVoltage - TipOffset)/22.0 = HandleTemp
goes to:
TipOffset = TipVoltage - HandleTemp*22

Ralim, 2020-03-29:
The handle temperature is included because of cold junction compensation of the thermocouple integrated into the tip. The material change point (cold junction) is in the handle, so it is required to compensate for the temperature of this point.

The most likely cause of this is:
* I think for some tips the uV per deg C is wrong and I'm testing this
* Its quite possible before when you had it set to 450C it wasnt actually reaching 450C

Most likely is a combination of both 😅

Firebie, 2020-03-29:
If we accept that tip thermocouple is linear - we do not need cold junction compensation (and is there any proof that it has the same type as the one is used in the tip?).

Ralim, 2020-03-29:
@Firebie 

I'm not an expert on this at all, but i did to as much reading as I could, and miniware also models their code in the same way (last I read it). If you believe otherwise, can you _please_ point at documentation so I can understand :) 

It's less of "accept" and more of a few people measuring the tip's response.

And yes we do need cold junction - it has nothing to do with the "curve" or calibration used for the tip.

From my understanding of thermocouples in how they are created; they produce a voltage in response to temperature as a result of the theromoelectric effect.
Because of the nature of the tips construction there is a second junction created from another pair of dissimilar metals where the connections on the tip mate to the handle; where these effectively create a potential that opposes that created in the tip. The metals used in the handle are similar to those used in the tip as far as I'm aware.

To compensate for this we are adding the handle temperature, and this is done at the stage where both are in degrees C, because they definitely do not have the same response. 😅

Firebie, 2020-03-29:
I still don't understand how in our case Cold Junction Compensation (CJC) have to work.
At the calibration phase (equilibrium) we have handle temp 20C. So, air temp is 20C. Later, during soldring, air temp is still 20C, while handle temp is 35C, so we have +15C surplus. How this is correct?

Ralim, 2020-03-29:
Air temp _never_ comes into the equation **after** calibration.

Because what we care about is the temperature at the tip, and that is measured relative to the temperature in the handle.

If tip temp is X, handle temp is Y; the adc channel reads (X-Y)*(op-amp gain)*(ADC conversion factor).
X and Y are the only unknowns in that equation.
Y is measured using the TMP36 ic that is mounted between the two contacts onto the tip via a separate channel. 
Then can factor the ADC readings down to (X-Y) and substitute Y in by this second measurement channel.

Firebie, 2020-03-29:
'getTipRawTemp' - it returns  (X-Y)(op-amp gain) - i.e. already CJC adjusted value?

Firebie, 2020-03-29:
I will stay with my own opinion that we do not need soldering mode handle temp compensation (as it has nothing to do with actual tip temp) and can use TipOffset (from my formula) calculated during calibration phase (equilibrium) - which effectively does same compensation (similar to CJC), but in a software way.

Ralim, 2020-03-29:
If you use your formula, it doesnt work if ambient changes or handle temp changes (during use).

You can test this by disabling the compensation, then measuring tip temp on a hakko measurement unit then using for like 45 mins so handle warms, and the time temp will have dropped by the amount the handle warmed up.

Firebie, 2020-03-29:
Lets look to the problem from the other side - if tip thermocouple produces same uV for the same temperature of tip, irrespectivly of what current temp handle has (+20 or +35) - we can convert this tip uV to C - we do not need handle temp and CJC at all.

Ralim, 2020-03-29:
[image]

Yes but keep in mind that the "tip" we are talking about here is the tip of the iron, and cold junction is because of the join between the removable tip part of the iron and the copper sleeves inside the handle. (See picture, the "tip" that makes uV is on the left, but all of this is about those two joins in the middle).

Both of those joints **also** act as thermocouples, and this whole process is about trying to remove their influence. Since the voltage we read at the ADC is the sum of both of those junctions + the "tip".

Firebie, 2020-03-29:
If those U2/U3 junction thermocouple type characteristic is linear and uV/C coeff not far from 22 (which,  actually, doesn't matter a lot as handle temp is not high) - their influence is already compensated by formula: "(TipVoltage - TipOffset)/22.0 + (U2 + U3)/22" which leads to  "(TipVoltage - (TipOffset - U2 - U3))/22.0", so, we can consider that  U1  and U2 are already part of TipOffset, as we see correct temp on display.

Ralim, 2020-03-29:
`tipvoltage =(tipTemp*22) + (handleTemp*Y*2)` Where Y=gain of U2/U3 which is assumed to be ~= the tip/2. Therefore `tipvoltage=(tipTemp*22)+(handleTemp*22)`.

Then we have `adcReading= tipvoltage+calibrationOffset == [(tipTemp*22)+(handleTemp*22)] + calibrationoffset`

Then we can re-arrange that to:
`tiptemp =(adcReading-calibrationOffset - (handleTemp*22))/22`
Which is why we need to know the handle temp (CJC) and the calibration offset (ADC offset + op-amp offset).

TL;DR tipoffset is done assuming both U1,U2,U3 are equal so that we _ignore_ CJC and only measure the ADC+op-amp offset. tip-offset does *not* include U2+U3.

Firebie, 2020-03-29:
TipVoltage = TipTemp * 22 + ColdJunctionTemp * 2 * 22
TipVoltage = (TipTemp + ColdJunctionTemp * 2) * 22
TipVoltage / 22 = TipTemp + ColdJunctionTemp * 2
TipTemp = TipVoltage / 22 - ColdJunctionTemp * 2

Firebie, 2020-03-29:
Wait, U2/U3 are connected in a way that they will compensate each other V potential, so, their influence has to be 0 to the TipTemp.

Ralim, 2020-03-29:
Sadly they are not actually, as they are different junctions (as far as I understand it). If they did cancel out there wouldn't be hundreds of IC's dedicated to exactly this function.

Even if your previous equation you still have the handle temp component...

Firebie, 2020-03-29:
If they compensate each other, formula becomes:
TipVoltage = TipTemp * 22 + (ColdJunctionTempU2 - ColdJunctionTempU3) * 2 * 22
TipVoltage = TipTemp * 22
TipTemp = TipVoltage / 22

Firebie, 2020-03-29:
U2 and U3 are created from same set of metals: tip connector metal + copper (or whatever of gold color) on PCB board , but connected in opposite directions - so, they are compensate each other.

Ralim, 2020-03-29:
https://www.maximintegrated.com/content/dam/images/design/tech-docs/4026/4026Fig01b.gif

Please read this maxim article at the least
https://www.maximintegrated.com/en/design/technical-documents/app-notes/4/4026.html

You are forgetting that we are talking about the two metals inside the tip assembly, and where they expose to meet the copper.

There are actually extra junctions here too; the copper to metal sleeve is the same on both sides so it cancels out. But the inside of the sleeve are two different metals (when the join to the sleeve) which is the junction that we are concerned about.

Typically it is modelled as just the one as one of the base metals is fairly unreactive (as I understand it).

Please, read some datasheets, im possibly wrong here, but don't want to discuss this back and forth without using accredited sources for this rather than finger pointing 😅 as that's not productive.

Firebie, 2020-03-29:
In general - why we need CJC.
Thermocouples (TC) tables(!) designed in a way that TC has 0 V at 0 C.
In old times scientists has been using same TC, included in an opposite way and inserted into 
melting ice (at 0C) bath - and from here, I think, we have name for this second TC - Cold Junction Compensation (CJC).
Nowadays this CJC can be achieved by several ways.
In a case of software implementation of TS100 we can easily achieve this by just substructing TipOffset.

Firebie, 2020-04-02:
I have to admit, that my understanding of Thermocouples and Cold Junction Compensation behavior was wrong :-)

closed 2020-09-14 by Ralim