Ralim/IronOS #568: Protection for faulty (shorten) tip

Firebie, 2020-01-28:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [x] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Feature for TS100

* **What is the current behavior?**
In a case of shorted tip and powerfull PS TS100's mosfet can be short out.

* **What is the expected behavior?**
Limit too high current with, lets say, 3A only.

* **What is the motivation / use case for changing the behavior?**
Prevent TS100 damage

Firebie, 2020-01-28:
Will something like this help?
https://gist.github.com/Firebie/3f7bd9684a9235e89368fce00d348c22

Ralim, 2020-01-28:
Hi,
Unfortunately this won't help at all.

In the irons, you can simplify the heating circuit to :

(Power +)--- (resistor in tip) --- (N-FET)---(Power -)

In normal operation, the N-FET is PWM'ed on and off so that the average current is the desired power (by estimating using approximate tip resistance).

This means that current in the circuit is actually only two values; 0 or (24V/8.5 ohms).

The maximum current in the ts100 is around 3A, (24V), and both the tip and the MOSFET are rated for this.

If you run the iron at around 26V you will tend to make the MOSFET fail to overcurrent eventually. If you run the iron at any higher voltage than that, you will make the internal DC/DC fail.

In the case of a tip failing, it is most likely that the tip has been stressed previously (extended use above 400C) and has eventually failed. Or manufacturing defects.

This firmware doesn't magically run the iron harder per se, just that it's pid is tuned for slightly more overshoot, so it ramps up faster.

closed 2020-01-28 by Firebie