Ralim/IronOS #569: version-screen on activated autostart and 'heat-stop'

discip, 2020-02-03:
Good evening,

* **I'm submitting a ...**

 - [x] bug report

* **Do you want to request a *feature* or report a *bug*?**
.

I want to report a bug.

* **What is the current behavior?**

Currently the iron stops heating and besides that it shows the version-screen, when autostart is set to either T or S and you push and hold the B-button to stop heating.

* **What is the expected behavior?**

On pushing and holding the B-button the iron stops heating and shows the idle screen.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

Set autostart to T or S and unplug the iron.
Plug the iron in again and watch it heating up.
Now push and hold the B-button to stop the iron from heating.

* **What is the motivation / use case for changing the behavior?**

It would match the behavior of the iron if the autostart mode was set to F.

* **What are you running:**

  - **TS80**

  - Firmware Version: v2.08.1

  - PCB Version: 2

  - Power Supply (Voltage and Current Rating):

voltage | amperage

-|-

3.6 - 6.5V |   3A

6.5 -   9V |   2A

9   -  12V | 1.5A

Thank you in advance

kind regards

Firebie, 2020-02-04:
[TS80_EN.hex.autostart.fixes.zip](https://github.com/Ralim/ts100/files/4153801/TS80_EN.hex.autostart.fixes.zip)

discip, 2020-02-05:
@Firebie 
Thank you very much that solved the odd behavior!
ps.:
Is this based on the latest commit?

closed 2020-02-05 by discip