Ralim/IronOS #571: iron starts heating although autostart is set to sleep

discip, 2020-02-03:
Good evening,

* **I'm submitting a ...**

 - [x] bug report

* **Do you want to request a *feature* or report a *bug*?**
.

I want to report a bug.

* **What is the current behavior?**

The iron starts heating, when it gets powered up, regardless of beeing moved or not.

* **What is the expected behavior?**

The iron does not start heating, when it gets powered up, while not beeing moved, until it gets moved.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

Set autostart to S and unplug the iron.
Plug the iron in again and watch its behavior.

* **What is the motivation / use case for changing the behavior?**

If this is left at its current state, there is no sense in having the option to choose between T and S in the autostart settings.

* **What are you running:**

  - **TS80**

  - Firmware Version: v2.08.1

  - PCB Version: 2

  - Power Supply (Voltage and Current Rating):

voltage | amperage

-|-

3.6 - 6.5V |   3A

6.5 -   9V |   2A

9   -  12V | 1.5A

Thank you in advance

kind regards

Firebie, 2020-02-04:
[TS80_EN.hex.autostart.fixes.zip](https://github.com/Ralim/ts100/files/4153828/TS80_EN.hex.autostart.fixes.zip)

discip, 2020-02-05:
@Firebie 
I tried the attached file, but unfortunately it did not help. :(
The iron still starts to heat up immediately after being plugged in.
Tank you for trying anyway.
I hope you will get this to work as you did with #569.

Firebie, 2020-02-05:
Try this:
[TS80_EN.hex.autostart.fixes.3.zip](https://github.com/Ralim/ts100/files/4158176/TS80_EN.hex.autostart.fixes.3.zip)

discip, 2020-02-05:
Sorry but this did not solve the problem either.

Firebie, 2020-02-05:
Try this:
[TS80_EN.hex.autostart.fixes.4.zip](https://github.com/Ralim/ts100/files/4158251/TS80_EN.hex.autostart.fixes.4.zip)

discip, 2020-02-05:
Nope, unfortunately no obvious effect.
Thank you for being persistent. 👍

Firebie, 2020-02-05:
I am running out of ideas, try this:
[TS80_EN.hex.autostart.fixes.5.zip](https://github.com/Ralim/ts100/files/4158312/TS80_EN.hex.autostart.fixes.5.zip)

discip, 2020-02-05:
@Firebie 
Thank you for your effort so far. 👍 
But this one did not work either.
ps.: Even if you don't get this to work, you tried and that is what counts in the end. 🥇

Firebie, 2020-02-05:
Are you sure that the iron is steady during power connection? If it is not, accelerometer will trigger out of sleep behavior.

discip, 2020-02-05:
Yes I am.
I even clamped it down to eliminate every possibility of moving. ;)

Firebie, 2020-02-05:
We have to call in main developer :-)

Firebie, 2020-02-05:
I have a new idea:
[TS80_EN.hex.autostart.fixes.6.zip](https://github.com/Ralim/ts100/files/4159506/TS80_EN.hex.autostart.fixes.6.zip)

discip, 2020-02-05:
You are really determined! I appreciate that. 👍 
But despite the fact that the display now says "Sleeping...", it still shows underneath that it heats up and the tip gets hot as well.
I think you are getting closer. 👍

Firebie, 2020-02-05:
During sleep iron uses sleep temperature (150C by default).

Firebie, 2020-02-05:
Try this:
[TS80_EN.hex.autostart.fixes.7.zip](https://github.com/Ralim/ts100/files/4160715/TS80_EN.hex.autostart.fixes.7.zip)

Firebie, 2020-02-05:
Better try this:
[TS80_EN.hex.autostart.fixes.8.zip](https://github.com/Ralim/ts100/files/4160746/TS80_EN.hex.autostart.fixes.8.zip)

discip, 2020-02-05:
You finally did it.
Awesome! 👍 
I am going to run some scenarios and if everything will work, this issue is terminated.

discip, 2020-02-08:
@Firebie 
Sorry, but I have to reopen this, because I noticed that the iron starts to heat, without being moved, if the sensitivity is set to 6 or greater.

Ralim, 2020-02-08:
That's normal if you have sensitivity set high, as its sensitive to pick up the noise in the sensor of the iron itself.

discip, 2020-02-08:
And is there anything, that could be done to fix this behavior, since I need the sensitivity to be 7 for my application. But I can't use this in combination with autostart sleep, because it is triggered way to early.

I just had an idea of giving the autostart sleep its own sensitivity.
Could that be done?

Ralim, 2020-02-08:
Having multiple sensitivity levels could be done, but makes the code even more interlinked which makes debugging harder. A different type of filter could help a little bit here, but not massively as some of the accelerometers are noisier than others.

Could you explain more about your use case?
It feels like you a wanting something different to what sleep mode is for...

discip, 2020-02-08:
For the most part I am soldering tiny parts and in that case there is not as much movement, so I have to set the sensitivity higher (for me 7 was enough) to keep the iron at its required temperature.
And autostart in sleep mode is just a very convenient way of using the device, because in most cases I do have to prepare things after I plug the iron in, so the tip is not heated for nothing.
I hope I could make my situation clear enough.

By the way: I am impressed and grateful for what you have achieved. 🥇 
And I do not take your work for granted.
I just hope to help to improve this already great device.
Thank you so much so far

Firebie, 2020-02-08:
2 levels of motion sensitivity (one for autostart - less sensitive and one for general - more sensitive) will not help, as iron will never come back to sleep mode due more sensitive second mode.

discip, 2020-02-11:
@Firebie 
Good evening,
after your post I was confused for a bit, because the setting: sensitivity = 7 worked flawlessly for me in the past (2.06 RC5). And after positively testing it for assurance just some minutes ago.
@everyone
My assumption is: Because the temperature is relatively low (approximately 21°C) after the iron is being plugged in (Wich I definitely appreciate! cudos to @Firebie 👍 Please, please keep it like this.), there is more 'noise' (as statet by @Ralim). And later while working the sleeping temperature is much higher (90°C in my case) so there is not as much noise. And the iron sleeps with the sensitivity set even to 9.
So I think the solution would be: To make the sensitivity dependent on the temperature, if that makes sense. :)
Thank you in advance
regards

Firebie, 2020-02-11:
Try this
[TS80_EN.hex.autostart.fixes.9.zip](https://github.com/Ralim/ts100/files/4187989/TS80_EN.hex.autostart.fixes.9.zip)

discip, 2020-02-11:
Thank you so much.
This is it! 👍 🥇 
You did it!
Works as expected!

closed 2020-02-11 by discip

Firebie, 2020-02-11:
9 or 10?

discip, 2020-02-11:
Sorry, its 10.

discip, 2020-02-18:
Good evening @Firebie,
would you mind to share the latest compiled build?
Or help me with compiling it myself?
kind regards

Firebie, 2020-02-18:
Here:
[TS80_EN.hex.autostart.fixes.11.zip](https://github.com/Ralim/ts100/files/4221651/TS80_EN.hex.autostart.fixes.11.zip)

discip, 2020-02-20:
Oh thank you so much.
It works s intended, buuuut ...
the scrolling description unfortunately is missing the newly added 'O-parameter '.