Ralim/IronOS #575: Error when try run make_translation.py

playday3008, 2020-02-06:
Run python3 Translation\ Editor/make_translation.py
Making ../workspace/TS100/Core/Src/Translation.cpp from .
Traceback (most recent call last):
  File "Translation Editor/make_translation.py", line 510, in <module>
    defs = loadJson(os.path.join(jsonDir, "translations_def.js"), True)
  File "Translation Editor/make_translation.py", line 20, in loadJson
    with io.open(fileName, mode="r", encoding="utf-8") as f:
FileNotFoundError: [Errno 2] No such file or directory: './translations_def.js'
##[error]Process completed with exit code 1.

Ralim, 2020-02-06:
The script assumes you are running it in its folder
if you `cd` into the `Translation Editor` folder first it will work.

playday3008, 2020-02-07:
ok, i create .yml file for github actions, and try setup autocompile after push

closed 2020-02-07 by playday3008