Ralim/IronOS #577: TS80 draws more than 24W

Ellipsis753, 2020-02-10:
**I'm submitting a:**
Bug report

**Do you want to request a *feature* or report a *bug*?:**
Bug

**What is the current behavior?**
TS80 will draw more than 24W when heating (if set to 24W mode).

**What is the expected behavior?**
Should never draw more than 24W.

**What are you running:**
  - Firmware Version: 2.07
  - PCB Version: Unknown
  - Power Supply (Voltage and Current Rating) :12V

Hi. I'm looking to get the full 24W out of my TS80. I've done this by buying this:
https://www.amazon.co.uk/gp/product/B078GJNXBF/
This is meant to be able to provide 12V at 2A. (24W)
After changing my iron to 12V mode, it will cut off when heating the tip. When details are shown, this seems to happen because the current draw goes above 24W.

As a workaround, could someone recommend a USB QC3 power supply I can buy that is able to get the full 24W out of the TS80 and is compatible with this firmware?

whitehoose, 2020-02-10:
Can you say how much over 24w?  Last time I measured a tip =4.5ohms so at 12v, 4.5 ohm =2.66A - 4ohm =3A
if the tip drops just 1/2 ohm that results in 36w so it might help the brain shed to know just what more means in this case, bear in mind that its not that accurate, the electronics aren't measuring precise values it's more of a programmer's guestimate.  You really need a USB tester like the ruideng meter
The AOKoda QC3 charger acts as a good voltage conditioner and it's only £10 ish if you shop round. Try bangood, fleabay or any of the usual chinesium emporiums you'll get it for less.
I sort of get the idea of why - but if you read the specs and look at real world performance, using a bog standard qc3 9v output will melt lead quite impressively.  12v is far too close to the limit of the specs for my tastes.  Its a reasonable iron I'd rather keep it working than remembering the puddle.
I keep saying this - but if you are trying to melt a ton of lead onto an iron girder, you've bought the wrong iron.  The 80 is quite impressive - I can lead solder onto a 20g slug of copper without much effort at 380c.  If you are taking your work above 400c you risk damaging components and burning the solder into a brittle nub ... its not good practice if you're serious about soldering.
If it's "scientific curiosity" or you want to get into the record books - its your iron - but its going to have a short sad life if you drive it hard as a matter of course.  Just because you can doesn't mean it's without consequences.

Ellipsis753, 2020-02-10:
Hello Whitehoose. Thanks for the reply.

I'm not trying to get especially high temperatures. Just want the iron to heat as fast as possible and maintain temperature well. Would you recommend not using the 12V mode in the options? (9V does work without issues for me.)

I've ordered a ruideng meter. So I'll include readings when it arrives. AOKoda QC3 looks to be battery powered only - so not super useful for a permanent setup for me.

My tip appears to be 4.2ohms.

whitehoose, 2020-02-11:
Hi
The ruideng is an important part of my work kit.  Also handy for testing capacities of batteries etc all round easier than an AVO for USB kit.
The AOKoda I use my 80 with a 4s Lipo.  I cant remember what I was thinking, but no, not a lot of good with the adaptor you bought - it would be fine with any 7.5v or above PSU though,  I got so tired with the incompatibilities of QC3.  The AOK has been rock solid so I bought a couple and use use it powered either by the 4s cells or my big chunky transformer for any flavour of USB.  It's qc3 but also delivers 5v USB2.
Finally
Despite all the problems I've had with portable qc3 supplies I've never considered the 80 to be slow ... or under powered.  My test is a 20g slug of copper which the 80 coped admirably with.  I don't need more - so for me it's not been an issue.  I bought my 80 almost as a toy - Thinking it wouldn't stand up to even my modest needs, I was wrong it's more than adequate.  My problems were all with QC3 compatibility - and the AOK solved that.  Early in my search for compatible portable power.  Tried a standard USB2 power brick connected to a boost converter then to a qc3 adapter I over volted and killed the iron - so for me 12v is a red flag I choose not to - because I don't need to.

I totally understand some people buying an 80 as a "project", nothing wrong there - but I think they understand the risks of taking their kit to the edge.
In my case it's part of my toolkit and while I'll try somethings - I want to see how long I can keep it going

Ralim, 2020-02-11:
Hey,
Because of how the hardware works, 9V is the main supported voltage, with 12V (24W), the micro controller is trying to keep the _average_ below 24W. The peaks will be closer to 36W, most QC controllers do not mind this (often they just let the voltage sag).

Some controllers will trip over current protection from this (technically they are all meant to.. but most do not).

My general advise is to run 9V, unless you _need_ 12V. Once your in 12V YMMV, especially because of how _not a spec_ the QC spec is.

Firebie, 2020-02-11:
We recently added a power limit option in the settings - there is a good chance to try it.

Ralim, 2020-02-11:
Keeping in mind said limit only limits PWM duty cycle it won't limit the peak.

That said, if your QC is shutting down due to over temp it most certainly will help

Ellipsis753, 2020-02-11:
The newest firmware with a power limit set to 24W actually seems to totally solve this issue for me.
Thank you.

closed 2020-02-11 by Ellipsis753