Ralim/IronOS #578: ts80 usb C-C cable

LianAarts, 2020-02-12:
Hi I just got the ts80, I'm really loving it. I am also enjoying the custom software! 

The only problem I have is that I can't find a C-C cable that works, I have tried every combination but can't seem to get it working.

What does work for me:
-The original cable that came with the ts80 and a QC3 adapter(usbA) or power bank(usbA).
-The original cable connected to a satechi usb hub that is connected to a QC3 adapter(usbC) or power bank(usbC)
-The original cable connected to a satechi usb hub that is connected to a QC3 adapter(usbC) or power bank(usb) (this is so weird because PD should not work maybe the satechi hub does a conversion?)
-An aftermarket usb A to C cable
-A cheap micro USB cable and a small micro to C adapter

What doesn't work:
-A usbC-C cable connected to a QC3 power adaptor
-A usbC-C cable connected to the satechi hub

I have tried multiple cable that are QC3 rated so I have no idea what the problem is.
Does anyone have a solution for this or maybe a cable that is compatible?
(sorry for the bad English)

Thanks in Advance!!

whitehoose, 2020-02-12:
Not certain I understand what you are trying to achieve here.  Talking connectors - the usb standard A normally plugs into a USB power **SOURCE**.  The source can be 1.1, 2, 3 - but should be backwards compatible and therefore, unless its been tampered with will be standing at 5v (4.4v-5.25v).  If its c compatible there is the potential to negotiate for higher voltages - but the standing voltage should be 5v.
If you see a USB 3.1 "c" connector, that would normally receive a voltage output from a power source.
A powerbank would have a USB A output and an input that could be usb mini, micro or "c" type used to charge the bank from a power source.
The basic rule of thumb is male connectors collect from a power or data source while female connectors output power and or data. (this is a guide not a rule - but it applies in most cases.

Ralim, 2020-02-13:
Hi,
Your running into the fact that the TS80 is designed for QC. And USB-PD specifies you are _not_ allowed to use QC over the same USB-C as PD.

Many power bank violate this spec, which causes confusion.

USB-PD uses pull down resistors on the CC pins to indicate that a device is connected (why power bank turns on when you connect a phone). The TS80 does _not_ have these pull down resistors, so it can not work with USB-PD sources.

USB-C output devices that implement spec properly either : have no QC2/3 on the USB-C, or they have to have no USB-PD. Ones that have both have a weird inbetween state that doesn't work with the TS80.

You can buy USB c-c cables that have these resistors in them, and then be forever confused why you cannot use those cables with anything USB-PD/ Smart.

LianAarts, 2020-02-13:
Hi, 
Thanks for the quick response!! @Ralim @whitehoose 

> Your running into the fact that the TS80 is designed for QC. And USB-PD specifies you are _not_ allowed to use QC over the same USB-C as PD.

Maybe I wasn't very clear in my post, wil link all the chargers/cables I used :)

[QC3 charger(Blitzwolf)](https://www.aliexpress.com/item/32885317092.html?spm=a2g0s.9042311.0.0.27424c4d2a0PHh)
[QC3 power bank(Blitzwolf)](https://www.aliexpress.com/item/4000074756487.html?spm=a2g0s.9042311.0.0.27424c4dtdT3NN)
[usb C-C cable(Ugreen)](https://www.aliexpress.com/item/32679851391.html?spm=a2g0s.9042311.0.0.27424c4dtdT3NN)
[the PD adapter that weirdly works when using a Satechi usb hub(apple)](https://www.apple.com/shop/product/MX0J2AM/A/96w-usb-c-power-adapter?fnode=8b&fs=f%3Dcharger%26fh%3D4595%252B45d4) This is the newer 96W adapter I have the 87W adapter.
If the product pages are correct all the used devices are QC3 compatible(except the last one of course). 
Thanks for the help!! @Ralim 

> The basic rule of thumb is male connectors collect from a power or data source while female connectors output power and or data. (this is a guide not a rule - but it applies in most cases.

I totally understand what your saying! The only logical explanation I have: when you're using a reversible usb C cable the ts80 simply isn't recognised as a power user, so the charge doesn't output power. Please correct me if I am wrong:)

The only solution I might have is to just purchase a small usb C-A adapter [like this one](https://satechi.net/collections/usb-type-c/products/satechi-usb-3-0-usb-type-c-usb-c-to-3-0-usb-a-type-a?variant=27300143433) because I know that a Satechi hub works.

Thanks for your response!! @whitehoose

whitehoose, 2020-02-13:
Well never have I ever ... makes a lot of sense ... and a mockery of the standardised USB school of thought :grinning: .  I make a lot of my own low resistance charging leads using mains grade cable, but of course qc needs data so not practical - and I've never come across c-c so not qualified to offer much of any advice.  EXCEPT to say qc3 has lost any credibility in my humble opinion.  I have it on good authority (from the guy who delivers our milk!) that QC3 is licensed to qualcom  who demand their pound of flesh from every licensee.  Of course the chinesium contingent laugh at the thought of paying for anything so they (individually) got a neighbour's kid who is studying word and excel to reverse engineer the QC3 firmware code and port it to powerbricks, soldering irons, etc etc.  with the result that it sort of works on the first device connected to 5 other contemporary devices but time has passed, the goalposts have moved and in the spirit of chinese whispers, the original story has been repeated by 10000000s of manufacturers and consequently changed beyond recognition so that you now have more chance of winning the lottery than finding an "other brand" power source that will work with anything you will ever possess (even if you bought it from amazon!!).
While I quite like my ts80 - when in a darker mood I do feel that the amount of smarts they've crammed into a small device designed simply to melt lead is like putting an i9 processor on a mallet.
Then again I got my first thermo iron in 2018! - I've managed for 64 years without.

LianAarts, 2020-02-14:
I just ordered the small adapter, I will update you when I find something interesting. Like you said I think this search can go on forever. Thanks!!

ckambiselis, 2021-08-08:
Was having the same problem as the OP. Trying to power the TS80 using a Satechi ST-TC108WM charger which has two USB-C PD ports, and according to the specs on the back of charger can supply QC 3.0 9-12V 1.5A. Connecting a Belkin 3.1 USB-C to USB-C cable capable of 100W will not turn the iron on at all, after that I connected it to a WITRN U2 USB analyzer and it shows 5V and then using the original cable from the analyzer to the iron, the TS80 wakes up and the analyzer shows the voltage change to 12V.

Then after reading the OPs second response I tried a USB-C to USB-A 3.1 OTG adapter (Deltaco USBC-1276) between the charger and the original cable and the iron woke up.

Just some info if someone else is searching for a solution, not sure if something can be done in the firmware to make it work without the adapter.

discip, 2022-04-23:
@LianAarts
Since this topic is a hardware one and @Ralim explained that this behavior is by design, this issue should be closed. 😊

thanks

closed 2022-04-23 by Ralim