Ralim/IronOS #581: turning on the firmware

paulgit123, 2020-03-04:
* **I'm submitting a Bug report
I have just uploaded the ts100 en.hex (twice) and the screen doesn't look right...
When powering the unit on, I get the screen showing 2 buttons:
one with a temp symbol
one with a wrench.
plus the pic of a plug 

Pushing the left button turns on the iron and it heats up
Pushing the right button brings up a screen that says: V2.08   18-01-20

Pushing both buttons does nothing.

Am running Windows 10
Iron on 19.5V power brick
It was working okay before the upgrade to this program but I was hoping for the extra features

Please help if you can
Thanks
Paul

Ralim, 2020-03-04:
This doesn't look like a bug, this sounds like it's working as normal.

What are you expecting?

Short press of the right button should bring up the menu, short press of the left should enter soldering mode.

paulgit123, 2020-03-04:
Hi
Thanks so much for the quick reply
So, I just tried it again and now it works just perfectly.
It wasn't going past the screen that said V2.08.....
Thanks again  
Wonderful program and I am looking forward to using it
Paul

Ralim, 2020-03-04:
You enter that screen by holding the right button, if you press it for shorter it will enter the menu.

closed 2020-03-04 by Ralim