Ralim/IronOS #582: TS100 PULSE FOR POWERBANK

Rocker7837, 2020-03-06:
* **I'm submitting a ...**
  
  - [ ] Feature request

* **Do you want to request a *feature* or report a *bug*?**

* **What is the current behavior?**
My ts100 doesn't keep my powebank on. 

* **What is the expected behavior?**
Keep the power bank on. 

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1.
2.
3.

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

My power bank need a current detected to stay on. 

* **What are you running:**
Ts100

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.08.1
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information**

Hi, I have a "ALI 21V 3.3A" power bank which need current detected to work. It uses dc barrel jack. It works for a while but then turns off. Is there a way too keep it on without having to press the button on the power bank. 

Thanks

whitehoose, 2020-03-06:
Not seen this regarding TS100s It's pretty commonplace with the TS80.  While ralim and the boffins think deep thoughts about your (if not unique) pretty near to unique problem - I bet the number of TS100 users with your powerbank are veeeery low numbers, While they are pretty commonplace with us TS80 owners.  The Dirty fix is to use your second socket to run a single LED which is sufficient to keep the magic pixies dancing - in your case It looks like you only have one so maybe add a single LED (resistored up for your normal operating voltage (21v?)) - 1-1.5k ohm across the output should do it (higher if you can).
Its not a solution - but will get you going till the head shed come up with something.

Rocker7837, 2020-03-06:
thanks for the response! il see what I can rig up over the weekend. I will probably have to make it inline with the cable otherwise it will stay on when the ts100 is disconnected if I put the led in the power bank.
Thanks again for the idea.

Rocker7837, 2020-03-06:
BTW it only happens in sleep mode.

whitehoose, 2020-03-06:
I made a very short (in my case) USB A plug-->socket with LED and test pins, a 6" "extension" lead handy for intercepting the lines, introducing resistance, meter or other power connectors.

If you have them, a back to back plug and socket of the right flavour with a LED&resistor across +/-  
-<==O==C 
If you can find the instructions there may be a sleep override.  My qc3 brick has a charge check/torch button that disables sleep if held in for 20 seconds.  Feedback from the ts80 owners club (:D) shows overrides aren't always provided I have a cheap brick that needs 40mA  to stay awake which is a lot of pixies for a single cell.

MisterFixx, 2020-05-01:
I actually have the same issue as OP, i think it would be quite nice if there was just a toggleable option for this inside the settings screen that could pulse the iron for a second or two at an interval, that way people who don't need it could just keep it disabled.
In any way, it's so much nicer than having to stick resistors down the USB ports.

closed 2020-06-15 by Ralim