Ralim/IronOS #583: TS100 - Device should tell us whether the used power supply is appropriate - (Feature Request)

Phil-2020, 2020-03-10:
* **I'm submitting a ...**
  - [ ] Bug report
  - [X] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
   - [X] Feature

* **What is the current behavior?**
Doesn't have this feature as far as i know.

* **What is the expected behavior?**
I request a function of the Ralim firmware that enables either at start-up or through a menu option to report back how the device is fairing with the power source. Things like deducing if the amperage is too high during the regular operation and during the heating up to the target temperature. 

Feature Request:

Basically a function/feedback screen that tells us if the DC power source we are using is safe for long term use and won't damage the hardware.

Also please add to the wiki what kind of power supplies can work, how well they would work and what exactly the amp ratings are usable without causing any damage. 

Specific lists of commercial laptop power supplies and how people have fared with them over a long time would be good. A database of power supplies where people can leave feedback whether it worked out for them with those specific power supplies would be very useful.

Right now one is really playing Russian roulette for the most part when it comes to the power supply.

whitehoose, 2020-03-11:
Already done for both TS80 and TS100 ...not by ralim - search RUIDENG UM34C is my personal favourite but there are a shedload of options available.  Add an electronic load and you can test power sources, power banks, batteries like a pro.

Really mate - ralim will be a lot nicer with his reply - but you are asking the impossible - certainly from a soldering iron. Also from the modding (or hacking) community, of you go off piste and use a custom firmware you accept that there are some risks involved.  If you can't accept that - go back to the stock firmware and play safe.  These pages don't come with a guarantee.  or many of the bells and whistles you can demand from a manufacturer. 

For the TS100 you need a pretty basic but quite beefy transformer or battery pack (I use a ts80 - have problems with QC3 compatibility - you are laughing all the way to the bank, the 100 requires a far more straight forward setup.  You just need an 18-24v supply, beefy enough to supply Ohms law with lots of currents - some people use lab bench PSUs some use an 18-22v power tool battery.  Others will use what they found in a skip or rescued from a microwave ... thats part of the fun of rolling your own.  Whatever you chose can (according to the laws of sod) go bang and fry your kit at any time.  My basic guide is always spend as much as you can afford ... common sense says current costs money, so if you pay peanuts you'll end up looking like a monkey - Ohms law isn't negotiable - If you stick to the specs of the iron - the tip is 8 ohms (ish) the volts are 18-24(ish)  if you buy a 18v PSU rated at 100A the iron can only take I=V/R s worth - so you don't have to be exact - 
https://imgmgr.banggood.com/images/upload/2015/11/TS100/TS100%20Soldering%20Iron%20Instruction%20Manual.pdf
a ballpark PSU based on these numbers says get yourself a good quality power source of 18-24v capable of delivering 5A and you'll not be far out.  Which and how much is up to you.  But sounds to me as you'll be wanting decent quality kit which if you pay a fair price should give you fair performance.  Stay mainstream and you'll generally get something that's does what it should.

discip, 2022-04-23:
@Phil-2020
Since this was answered and for some devices (TS80P) the available space is already exhausted, it will most likely never make its way into IronOS. 😓
So please consider closing this. 😅

thanks

resistancelion, 2025-01-27:
Well, i was trying to investigate that, and tried every power unit i had at my disposal, so i've broke my iron

[image]
[Works as it was in DFU/micro-usb mode tho... 🤯 ]

Guess, i will invest even more in this issue then

resistancelion, 2025-01-28:
> If I'm not mistaken, this is due to the poorly executed soldering of the OLED connections. I'm pretty sure you'll find this hint somewhere here in the repo.

No, i've killed it with bad power supply