Ralim/IronOS #591: TS80 Slow Heatup

techied, 2020-03-18:
* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
Using the TS-80 with included Type C cable and QC3.0 power supply, the custom firmware heats much slower than the stock firmware.

* **What is the expected behavior?**
The custom firmware heats as fast as the stock firmware

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Heat the iron
2. Observe heat time

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**
TS80, V2.08 18-01-20, included Type C cable and QC3.0 power supply.

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.08
  - PCB Version: doesn't display
  - Power Supply (Voltage and Current Rating) : CY-QC30 5v/3a, 9v/2a, 12v/1.5a

Ralim, 2020-03-18:
Can you please confirm what voltage the iron is running at?

techied, 2020-03-18:
@Ralim Vin 5.1

techied, 2020-03-18:
There must be something wrong with the QC negotiation because I flashed the stock firmware back and confirmed it heats up faster (at 9v I assume)

Ralim, 2020-03-18:
Yep, so your charger did not negotiate QC to 9V correctly, so the iron is running at 5V mode. It will be noticeably slower at 5V.

Most likely cause is the charger not implementing QC to spec.

Previously this firmware has used two patterns for QC negotiation, the pattern from the miniware firmware and one designed around the _actual_ specifications.

Currently the firmware is using the latter.
This improves support with higher end power banks and more expensive adapters, but some adapters have since started having issues.

I do not own _any_ adapters that fail the negotiation.

Also worth noting is that the stock firmware _only_ lets you run in 9 or 12V, it wont run at 5V normally

techied, 2020-03-18:
Hmm, that's a shame.

But a good excuse to get a proper QC charger and/or power bank!

Thanks for your work on this firmware!

closed 2020-03-18 by techied

whitehoose, 2020-03-18:
#techied Here I go again! :grin:  One of the explanations goes  Qualcomm came up with QC3 ... and in the tradition of such things expected to be paid for their efforts.  In the chinesium empire however no one pays for not nuffun' and laughs at the very thought of copywrite and license fees.  So following tradition many companies (in the spirit of non-cooperation) searched for someone able to reverse engineer the code contained in whatever QC3 device they happened to have lying round.
Thus was born the lottery that is the QC3 we all know and love. (OK rant over)

Rather than buying a QC3 charger and or powerbank (which could well suffer the same problems caused by chinesium whispers in the code).

Here is a possible alternative route.
I use standard 4s cells along with an AOKoda QC3.0 USB charger - which is a dongle that works perfectly with the TS80 and every other QC3 device I've tried it with.  The advantage is you can use any old cells from any mfr and using the magic of the AOKoda turn it into a consistent QC3 source.
There is also the possibility of using it as a supply conditioner, the only restriction is the minimum voltage required is 7.2v so standard USB is out.
@lysdexic1 offers https://www.ebay.com.au/itm/QC3-0-QC2-0-USB-Fast-Quick-Charging-Module-DC-DC-Boost-Converter-Board-Car-Phone/392177982492 as an alternative, I have one on order.
This is a boost converter that can use standard USB.

Just a thought if you're open to alternatives.