Ralim/IronOS #592: [not a bug] RAVPower PD Pioneer Model: RP-PB172 works

opcow, 2020-03-20:
TS80 on RAVPower PD Pioneer Model: RP-PB172:

- original firmware (1.06): does not work
- ver. 1.07 from Miniware: does not work
- Custom by Ralim: works

whitehoose, 2020-03-20:
Maybe works, but in the description it makes it pretty plain that powerbanks are for phones or more broadly for devices which start by taking reasonably high amounts electrickery at reasonably high currents.  They don't make em with a short climb through the available voltages followed by long periods of wobbly feathering and then a possible sleepy session.  Very few devices I can think of follow a similar use pattern to a qc3 soldering iron that's been left on while you fit a difficult component then make a coffee and start a crossword.

If you're in need of a battery solution - think lipo, specifically think 4s or above, far better suited to the random high and low demands of a qc3 lead melter.
If you can, think qc3 wall wart but if it has to be batteries - lipo is better, you'll need a qc3 confuser to negotiate the pixie queue - but that's a low cost one off thing - and you'll get better, more consistent results than entering the qc3 lottery.
For the price of one of them pioneer pixie motels you could buy 4 decent 4s batteries and an AOKoda (req 7.2v minimum or Single Cell Lithium Battery Boost Power Module Board 3.7V 4.2V Liter 5V 9V 12V u which runs from Input voltage: 2.8-4.5V) I like the AOK - @lysdexic1 has a few good words for the other (I have one on order).

discip, 2022-04-23:
@opcow
Please close this, since it was added to #564. 😊

thanks

closed 2022-04-23 by Ralim