Ralim/IronOS #598: Weird issue

jkost3, 2020-03-26:
* **What is the current behavior?**

My TS100 works fine only with the Ralim firmware
when i return back to the stock firmware, only shows
the "Press" message  and the buttons do not respond when pressed
thus the TS100 remains cold.

any similar experience or ways i can solve this matter??

Ralim, 2020-03-26:
Going to say thats certainly a first.. Not really sure i can help much with the stock firmware im afraid

jkost3, 2020-03-26:
Hi thanks for your reply,
posting here the problem i mean i cannot return to the stock firmware
Ralim firmware though.. functions perfectly fine! maybe i'm doing something wrong?

any advices?

discip, 2020-03-26:
Good evening,
if Ralims firmware works, why would you even bother going back to stock.
The first thing I did when I received my iron, was to flash Ralims firmware!
From there on it only improved.
Every issue was ironed out and every suggestion was implemented almost instantly.
Also the support of the community is great.
I recommend you sticking with the firmware created by @Ralim.

jkost3, 2020-03-26:
hi

mostly i agree with you but i'm trying to understand... was something i did wrong
so partialy the TS100 is now damaged which Ralim's code "overlooks" and seems it works fine?

i also like Ralim's firmware but i disagree on one thing and that is the extended
temperature setting (450c) i believe if the tips were made to work up to that
temperature then also Miniware would implement it.

This tool it's something everybody should have and we should all try to improve
if we have a good idea to propose or implement if somebody has coding skills.

whitehoose, 2020-03-26:
@jkost3 
I get it, nobody wants to have to eat lobster every day.
Are you "double dipping"? Copy the file, see the .err copy again wait for the auto reboot.
Some people have said windows doesn't like dragging and drop the file but using the command line works.
If you are in windows, where is the file?  sometimes system folders don't like communicating with the outside world.  Or a system folder can be buried so deep that the path exceeds 128 or 256 characters, or it's been given an illegal character.

Try creating a new folder in windows in the root (preferably) D:\frmware or if only one partition C:\frmware and then open a system command prompt
type copy D:\frmware\ts100.hex X:\ [X:\ = whatever letter your TS100 is allocated](thats copy {space d:\ ...... X:

Remember to plug the Iron in while holding the button nearest the tip down - screen should show DFUxxxx and a file manager window should open automatically

Drag and drop the file into the window .. wait 'till the window closes and re-opens showing an ..err file - repeat and wait for the reboot.

If no joy - try the command prompt method

jkost3, 2020-03-27:
Hi!

after copying the file i always get the filename with the "RDY" extension meaning there
was no error (using drag and drop).

i've noticed that after switching a couple of times between the official firmaware and Ralim's
finally the official firmware works again! but after some time the iron becomes again non responsive
to button press.

can it be hardware issue? Ralim's code might not call some specific functions so it
runs smoothly?

whitehoose, 2020-03-28:
RDY is good. :grin:  I'm a ts80 type - but still read the 100 stuff just because and I've been hunting limping pixies most of my working life.
If the Iron becomes unresponsive with the official firmware too - I'd suspect hardware.
I've always mixed windows and linux when chasing hardware problems - this is as close as you can get with an iron ...
Silly questions (I've got 100 of ''em) is your data cable sound?  My pride is still stinging as a result of an int hr cable on a backup device freezing a server.
When it pauses does it come back with a reboot or "firm" presses or jiggles.  does it "go" before after or during use or long period of being ignored while powered.  What's on screen when unresponsive, does it feel warm(er) or not.
The $64000 question - how old is the iron?  When I killed my 80 after only a month I was honest with Banggood and they swapped it without argument. When I then asked Mini about buying spares for the corpse they sent me a new motherboard in exchange for the old one (Yes I have 2 irons now!).

jkost3, 2020-03-28:
hi

i appreciate for letting me know your experience it's very useful!!

so when it comes to hardware issue there's not much to look around
it's the STM32... i don't know if the F103T8U6 can be considered a good
microcontroller but i consider ST as a great company with great products!

my usb cables are fine..i could even try as an alternative such as to copy the firmware
from a smartphone supporting OTG so i can see if the problem can be resolved
but i don't think it's a firmware copy issue, as i said the file has RDY extension. 

about how old it is...it's a few months old and it's hardly used 3-4 hours in total!!!!
problem is i bought it from a local store and i don't expect much of support from them,
i don't know if mini can help on that.

whitehoose, 2020-03-28:
My gut feeling based on what you're saying is hardware - but it is only a hunch.
I'm in the UK and if its only a few months, I'd pay the shop a visit.  Normally I'd expect 12 months support, 1 month tends to be the swap zone then after a month it tends to be passed on to maker to fix, the ideal would be to be able to demonstrate the problem.  Does it freeze with the stock firmware? can you reset - or is it a case of after trying - it just comes right?  If you want to diy - I'd give every solder joint a good looking at.

In ralims detailed view there are lots of changing numbers - if the buttons stop do the numbers continue?  I'm about out of thoughts - if all else fails - contact manufacturer they have a forum - I'd hit that first then the tech support - the guy I spoke with (I can't find his mail) just picked up my enquiry http://www.miniware.com.cn/contact-us/ - if they tell you to forget it (which I doubt) I'm stuck

When I fried mine I was prepared to accept it was lost and just the price of trying custom stuff - I told them what I'd done and was shocked how forgiving everyone I dealt with was.
I'll be watching for udates

jkost3, 2020-03-28:
With Ralim's firmware it's impossible to demostrate the problem because it works fine.
since the stock firmware suddenly stopped responding the only thing i can think of, it's
that last time the soldering iron worked fine it might got affected by "overheat".

asking for assistance from the shop i bought it it's not a option, the shop is far
away, i have to give it back and wait who knows how long and it's the only
soldering iron i can currently use.

i've noticed that sometimes the handle can get a little hot (after long use in high temperature)
probably heat gets transfered to the handle? or the mosfet produces some heat? if that's the case then maybe a weak solder joint got expanded by the heat and cracked?

that's only speculation.... i was hopping to learn more by posting here.

in that case you're right saying i should take a close look at the PCB although i dislike re-opening 
it because the plastic case don't give me much of a confidence to play with!! (and i've noticed
a flaw with the front metal ring which makes it hard to close it back and if you don't pay attention
you might brake it.

if the soldering iron needs another soldering iron so i can fix it, it's hard case! :D

whitehoose, 2020-03-29:
Lol, if you need another iron you're screwed, open case - heat tip to 450c (that's why ralim picked 450) -  use pliers to rip the tip out and quickly make a join. :grin:
Seriously - you could access the mini support page, If its only the stock that glitches you can discount ralim's firmware as far as voiding terms of usage,  

 See what their response is and go from there.  If they want you to send it away - then you have to decide.  The other option, if Ralim's code causes no problem - pick a version you are happy with and stick with it - I use an older version from 2018 that suits my needs.

Fleabay you can get a used antex soldering iron for not much - still my favourite iron, I have a bench station I use now - but the antex is sound
I got the ts80 to replace my trusty 20 year old RS battery iron, honestly - ts80 is not as good - but I burned my boats so I manage.  I use an older Ralim firmware - its better than stock and works with 5v (thats why I use ralim's code), normally the work I do and the contract conditions would stop me using non approved hardware - but that's another issue. The irons do get hot - there are various warnings about extended use times. and heat Vs electronics pays my mortgage.
If you are consistently using <400c  I'd have to question your choice of iron.  I've been in the industry since 1970 temperature doesn't replace thermal mass - the "book" on 60/40 lead is about 340c.  If the Iron doesn't melt lead you need more thermal mass - not a higher temp which over cooks the solder burns the lead and causes brittle joints track and component leads.
I have a high wattage Iron and a heating plate for soldering aluminium substrates - that's another discussion though.

Ralim, 2020-03-29:
@jkost3

I do not follow the main firmware anymore as trying to keep track of it was a pain with the only distribution being a forum.

If their firmware doesn't load and this one does all i can really say is 🎉 . But realistically, its possible its their weird FAT32-packed-in-flash-and-ram-with-generated-table failing. I've run into this once before when messing with their firmware early on.
One way to test this theory, is if you flash their firmware and then plug it into a compute via USB the usb drive does not mount correctly and windows prompts you to format it AND the size is correct (~1.9MB off memory or less).

If that's the case, it might be possible to flash a "blank" hex which will reset the flash.
Or could use the flash dump i originally made as a backup to generate a "reset" flash for you.

To be honest though, I don't like to dig into supporting their firmware - that's what their forum is for.

The case absolutely does get warm inside, especially when running at higher voltages (~20+V). This is part of the design of the unit sadly and not a huge amount to be done here (other than it being "normal").

whitehoose, 2020-03-29:
I just opened my e-mail and "Ralim's firmware it's impossible to demostrate the problem because it works fine". hit me in the face I already said if that's the case you can hand on heart say their package is (for whatever reason) not working as expected.  Under UK consumer law that puts it 100% on miniware. 

Having slept on it I'm still thinking given all that's been said, if ralim's works 100% and the stock doesn't the question is do you pursue or "make do".  If your only reservation is 450c - you can disable boost in settings and simply don't set it manually - sorted.  As a member of the Asperger's club's OCD division, while it's not the ideal solution - I can live with it.
It would be nice to be 100% certain - but ..... Unless you are willing to live without the iron for a time that avenue isn't open.

Ralim, 2020-03-29:
The only other small thought to add here, is that my firmware tends to be the "picky" one, that gets hung up over dodgy stuff.

If this firmware is working, its almost certain that your hardware itself is fine and its their software getting the shits. You can try downloading all of the older versions of their software and trying them all until you get one that you find works.

jkost3, 2020-03-29:
@Ralim i have never had any issues when plugging it into the computer no prompts for format,
 although the disk size i think it was few kb containing the configuration file..

i thought that is normal...

i don't have "tools" to make an empty hex unless your tell me a fast way, or where i can
get your file.

@Whitehoose i'm about to go straight to mini's forum as soon as i try some of your advices.
concerning the 450c boost option yes it's disabled, are you implying i'm one more with OCD? 
i might be one, most people some short of OCD. :D

@Ralim if it's not a hardware issue and your firmware is the picky one...i don't understand
why your firmware works well on my TS100 and also the stock firmaware most likely works
fine on any TS100 than mine? 

final words, seems like it's a "partition" problem? the way the stock firmware it was built
so no matter how many times i flash the stock firmware nothing changes? what's the exactly
the disk size i should see when i connect the TS100?

whitehoose, 2020-03-29:
@jkost3 
I wasn't implying you, although I'd never use the term as an insult or dig - for anyone with a soldering iron I see it as a compliment - I totally embraced my Asperger's ocd years ago I think guys all have something that triggers them, engineering types have it in spades.  It's the manic attention to detail that gets the job done right.  It's why you're going to the mini forum - so I suppose the cap fits :grin:.
whatever you decide - I hope its also what makes you let us know what the outcome is.

jkost3, 2020-03-31:
Hi
i have sent an e-mail to their support and i've got dissapointed. 
initially it was going well! they replied promptly they asked for a video showing the issue
and the power supply voltage and also a proof of purchase, after i replied, they came back to me 
asking for the same things and also saying irrelevant stuff, such as i should change power supply if the soldering iron keeps restarting (what?..) :D fist of all none said it keeps restarting! and the 
video shows exactly what it does! 

their reply appeared as if the didn't read my e-mail and only took a look at it.

you guys offered much more support than the company who sells this tool :D :D

whitehoose, 2020-03-31:
I think I'd get back and point out their error ... It's possibly more of a problem with google translate than than anything - tell them youv'e tried different cables and power supplies as well as a lipo with no joy.

jkost3, 2020-03-31:
@whitehoose since you said so, just now i've turned my e-mail text into chinese and then from
chinese back to english and seems it gets translated just fine... :P

i'll try to explain it to them a bit more, although they have the video, they know their product
how it should function, so i don't know if it really needs much of explaining.

fact is... english is not my native language but i guess you can all understand what i'm typing. :)

@Ralim i'm waiting for some more details on how to make an empty hex or where i 
can find the one you recomended, thank you.

whitehoose, 2020-03-31:
@jkost3 at the moment as a consequence of the plague they may literally be dragging people off the streets to answer problems - 1st line support is usually done from a flowchart and in all the call centres I've ever worked in - fully trained refers to the 10 minutes someone spent on the day they started.

> fact is... english is not my native language
I just thought you had a faulty keyboard  :grin:  seriously - You make a better job of your second language than I make of my first (where's home?).

jkost3, 2020-03-31:
@whitehoose right now...it's not only their problem many countries are seriously affected
including my country's neibours..(italy from our west side) so now can you guess where i come
from?? in my country currently we can go out only for very basic stuff mostly for food shopping
and only with a permit.

thank you for your good words by the way ;)

whitehoose, 2020-04-01:
mmmmm
I hear Spain is almost as badly hit as italy - so are you in sardinia? or maybe tunisia or possibly France?
Wherever - hope you stay safe - these are bad times (or as the chinese would say, INTERESTING Times) everywhere.

jkost3, 2020-04-01:
@whitehoose nope..i'm from Greece, where are you from?

Mini didn't replied to my latest e-mail i wasn't impolite or something..
i find it reasonable to wait 1-2 days for their answer.

it is funny that i might will remain with speculations and i can't know what's wrong
with the soldering iron, i can't even try what Ralim proposed since i don't
have more details on how i should do it. 

and really i don't think it has to do with the outbreak... most provinces in china it's
like a country in size and population and not all china suffered from the outbrake
it was limited / they contained only the region (wuhan) it is considered the outbreak started
which now the media say it has recovered (i'm not from there so i can't know if 
they indeed recovered).

whitehoose, 2020-04-01:
I'm in UK - west yorkshire - Greece - I never saw that coming, (Italy is to your east!).  I know that the outbreak was in Wuhan - but the lockdown was countrywide, they do seem to have contained it well (if you can believe their reports).  We're in lockdown, but the numbers are set to increase for the next two weeks the precautions are to the end of april but the predictions say much later than that.  Every tv programme starts with a warning to stay in and separate.
[image]

Its odd they haven't replied - I found that with the time difference that if I posted by 11am my replies came next day at 7am.

I may well be wrong - but

jkost3, 2020-04-01:
@whitehoοse Greece is eastern to italy :) check google maps...

i think we went off topic...or it doesn't matter here?? ^^

they didn't replied and i don't expect much of support maybe i should feel lucky
that i found a solution by myself and it's still functional.

P.S just now i've created online an empty hex file exactly the same size as
the stock firmware...is it certain it wont damage the soldering iron??

...and as i said in an earlier post i appreciate your support you guys!!!

whitehoose, 2020-04-01:
I can't see it could damage anything.  In the past Ralim has said similar (unless you overwrite the rom which you can't do by dragging and dropping.  I found an atari site where it seems to be a standard fix for just about everything.  In the past I used to make customised boot files for DOS utility disks and windows image files and the boot loader fitted into a certain sized "box" which sat on a specific memory location - if that's missing or incorrectly formed it would not work - but would halt the process, not damage anything.

Ive never used an empty hex - so I could be off the mark - if it's way off.  Maybe Ralim could provide something that does work.

jkost3, 2020-04-01:
i don't know much...i only know it's an ARM based microcontroller it has a bootloader called DFU
something might get messed up...

i have found this topic talking about messed up bootloaders..etc
https://github.com/Ralim/ts100/issues/202

whitehoose, 2020-04-02:
I'm a PC type but pretty sure that the principle will be the same the boot sector contains pointers to the location of boot files - the and boot loader (DFU) uses that info to make the hardware boot by accessing that info.  The hex file just has the boot info to point  the bootloader (in the hardware ROM) in the right direction.  To alter the contents of the rom you'd need a hardware programmer

Using the empty file will presumably replace any corrupt boot info with zeros.  the iron will fail to boot because of a lack of info - but the boot loader will still try to attempt to boot.  Once the boot sector is flushed, the next step is to copy in the stock firmware into empty space which should then boot.

Have to say I agree ralim can't support stock firmware - but when it all goes TU I think its appropriate to offer enough guidance on how to recover an iron. I imagine its documented but the trick is finding it.  google is great when you know what to look for - but newbs tend not to be familiar with the correct terminology required to do a successful search.

Ralim, 2020-04-02:
Oh boy that went off on a tangent 😅 
No real issues though with that. (Though try to keep it to a minimum when it's long threads as we all get emails😅)

To wipe blank you need to create a blank hex file that starts at offset 0x4000 and has a length of 48kb (kilo bytes).

This will erase the user part of flash.

If you upload a hex starting at 0x0 it will come up as an error as the bootloader doesn't let you remove it

jkost3, 2020-04-06:
@Ralim you're right..i'm not going to post replies very often. 

i want to let you guys know that Mini recognized there's actualy a problem 
with my TS100 they requested the serial number and they sent me a video on
how to get it from inside the device. 

@Ralim i think it's a risk i try to flash the 48kb blank hex also i don't know how to
create that 0x4000 offset you said...since the file always starts at 0x0 and if i understood
correctly a wrong file will block the soldering iron and i couldn't remove it???

now it's up to Miniware if they will really give support on this.

Ralim, 2020-04-06:
Always feel free to comment 😅 it's nice to have the community around here 😅 so long as your not dieting an actual conversion out; I think go for it.

I can try and make the hex file for you when I get a chance 😅

jkost3, 2020-05-06:
Hi, earlier today i received a new TS100 (without the tip) 
somewhere in the middle of the back and forth e-mailing i thought it was a 
waste of time, but finally they helped with all of their powers!

I've even let them know there are strict importing rules in EU, but they managed
to  send it without any customs problems and that's so great! (though seems they
have some kind of a "warehouse" in Europe).

~~ Thank you MiniWare ~~

@Ralim do you think i should try to recover my ts100 with your code?
now i guess it's a risk free experiment...
Could you also tell me that 0x4000 offset what it is and how it works? :)
your custom firmware doesn't update the user data area?

Ralim, 2020-05-06:
@jkost3 
That's excellent news. I've found them to be good people whenever I have worked with them, even if language is sometimes a barrier.

0x4000 is the skip as the bootloader is at 0x0-0x4000 offset. If the bootloader sees you attempting to write to that section it will abort the update and not let you use it.

whitehoose, 2020-05-07:
@Ralim 
Would it be possible for you to make up an empty hex to include on the header of the firmware download page for use by all?  Reading other threads on other boards on other subjects it seems to be a widely used first step for problem solving but not one I've seen mentioned here before.  I can see that a cold reset might come in.

As with all these things it's easy if you know how, but if you don't ... just knocking one out is a bit of a mountain. 
I've not done anything like this for many years - I do a little with arduino but nothing hardcore
I don't even have a hex editor or compiler ....

Looking round the likely spots - the information is all what happened next, I can't find a word on how its created. 
I've an idea in the back of my head - but its not an area I've played in - I wouldn't want to muddy the waters by posting a badly formed duff file which didn't work.

Mini certainly fixed my problem, swapped the motherboard (foc) when I fried my ts80 ... but its not an obvious option, they don't offer spares for sale on the website either which is odd given the target audience of their kit.

@jkost3
Yay! good result ... :grinning: 

Stay safe all

Ralim, 2020-06-15:
@whitehoose 
Here is a force blank file:
https://github.com/Ralim/ts100/blob/master/Development%20Resources/force_blank_flash.hex

Flashing that will force all of the user area of the flash to 0x00 which will mean that the bootloader will erase and then flash anything you flash after that :)

The commands to make the file were basically:
```
dd if=/dev/zero of=test.bin bs=1024 count=48
arm-none-eabi-objcopy --change-addresses 0x804000 -I binary -O ihex test.bin blank.hex
```

whitehoose, 2020-06-15:
Magic, Thanks.

closed 2020-09-14 by Ralim