Ralim/IronOS #603: Blinking On detailed idle screen doesnt on cooldown

michael-blue2, 2020-04-09:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
bug

* **What is the current behavior?**
oled doesnt blink on detailed idle screen when cooling down

* **What is the expected behavior?**
oled to blink on detailed idle screen when cooling down

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1.Enabled blinking on cooldown
2.on normal idle screen the blinking works when enabled

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.09 git 04-04-20
  - PCB Version: (1/2)  1
  - Power Supply (Voltage and Current Rating) : 12v 5A

* **Other information**

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Firebie, 2020-04-10:
Should be blinking the all screen, or just, for example, "Tip:"?

michael-blue2, 2020-04-10:
Before the 2.09 update I had the 2.05, which it would blink the whole detailed screen slowly when cooling. But as I read on another threat dont remember the number that said they will implement it as a feature to select. As its a detailed screen and it can be anoying for some. So I am closing this one

closed 2020-04-10 by michael-blue2