Ralim/IronOS #604: Brand new TS100 won`t appear as USB Drive

ntz619, 2020-04-12:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
*bug*?

* **What is the current behavior?**
I tried everything to get the E:/ drive of the TS100 but it won`t appear anywhere. Tried every USB port, another cable and the device manager. 
Is it broken?
Seems like the usb port is broke or has a 
loose contact , should i have a look on it or is there a trick setting it up.

* **What is the expected behavior?**
USB Drive appears

* **What are you running:**
stock firmware, win10
On the idle screen, you can hold the settings button and it will show you the firmware version.

  -DFU: 3.45
  - Firmware Version: 2.18
  - PCB Version: V2.51A
  - Power Supply (Voltage and Current Rating) :19V 3,42A and 5V USB

Sorry if this is in the wrong category or anything but i can`t find a solution of my problem and this custom firmware is awesome :)

Ralim, 2020-04-12:
Hello,
According to your what are you running section you are using the stock firmware, not this firmware. Support for the official firmware is over [here](http://minidso.com/forum.php?mod=forumdisplay&fid=66).

v2.09 is the latest version of the firmware in this repository.

I *cannot* help much with the stock firmware as i have not used it since 2016. If holding the front button when attaching it to your computer works (screen shows DFU and pc shows a usb drive) then there is most likely nothing wrong with your device itself and instead its their firmware.

ntz619, 2020-04-12:
but it wont appear as usb drive, thats the problem. sorry for the misunderstanding

Ralim, 2020-04-12:
If it is failing to appear to show up as a drive in DFU mode, then check your cables. If that still fails you may need to talk to miniware.

ntz619, 2020-04-12:
ok, thank you :) hopefully i can fix it and try your awesome firmware 👍 and happy easter :)

ntz619, 2020-04-12:
it worked with the 5th cable wow, never thought of this easy fix :)

whitehoose, 2020-04-12:
A lot of the cheap cables seem to use wet string instead of copper.  I've collected all the usual Usb ends over the years and after taking all day to charge some batteries I now make most of my own cables.  I use twin 0.75mm copper mains cable for power leads and have decent quality multi core copper cable for data.  Having some known good quality cables to hand is always handy, especially for powe and well worth the effort.  Some of the cheap chinesium cables can be appalling,  The worst I found in my collection was reducing the charging current by 56%.

Psyc0Flyer, 2020-04-25:
You HAVE to make sure you have the soldering tip connected. I tried to do it without tip attached and I had the same issue! Now I am having an issue where I can't copy the .hex file over :/ I just received mine today and I have 3 folders already in the iron (.fseventsd, .Trashes, ._.Trashes).

satsis, 2020-05-22:
**Ralim**  On your firmware should open config.txt when connected to a PC?

Ralim, 2020-05-28:
@satsis 
As stated on the main readme, there is no usb drive config file for settings; instead all settings are in the menu.

>This firmware does NOT support the USB port while running for changing settings. This is done through the onscreen menu only.

closed 2020-05-28 by Ralim

jeskandarian, 2021-12-27:
For me the 4th cable worked. Who knew....