Ralim/IronOS #607: When reaching around 25w it powers off

michael-blue2, 2020-04-14:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report

* **Do you want to request a *feature* or report a *bug*?**
bug

* **What is the current behavior?**
the ts100 shuts off when it gets around 25w

* **What is the expected behavior?**
not to switch off

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1.put the detail screen to watch the watts
2.i insert a 19.5v 90w psu (asus laptops psu)
3. when its about 25w on heat up its shuts off
4. i reinsalled the 2.5 version and everything is good 
5. i reinstalled 2.08.1 version and the problem is there
5. when i am using a 12v psu which limits around 15w its fine

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.09 04-04-20
  - PCB Version: (1/2) 1
  - Power Supply (Voltage and Current Rating) : 19.5v 180w 9.23a asus laptop psu
2nd psu 12v 3.3 40w

* **Other information**

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Firebie, 2020-04-14:
Have you tried to enable power limit option?

michael-blue2, 2020-04-14:
> 
> 
> Have you tried to enable power limit option?

But my power supply can give up to 180w, and with on an older version it works. 2.5 its fine, after 2.5 the problem appears. I will give it just a go now to see what happens with the limit. ( But I do want some higher wats on some situations cause of long and thick wires)

Firebie, 2020-04-14:
May be 2.5 was not using more than 25W for heating?

michael-blue2, 2020-04-14:
> 
> 
> May be 2.5 was not using more than 25W for heating?

I tried it now, i start with limit of 30 and drop down to 15w and it shuts off to even 15w and needs 3-4s to boot up again. I even tried with my 12v psu to heat it up and then switch really fast psu and it stills shuts off at some point. I found out that if i hold down the boost button it doesnt shuts off for longer time. But still by 10s it will shut off. On 3.5 i can see the difference when it boots up from the 12v and 19.5 its massive, and can notice solders on big wires

Ralim, 2020-04-15:
Is there any way you can measure the voltage coming from that power supply while soldering? These kinds of random shutdowns are almost always a power supply that is not coping in some way. The TS100 bootup time is quite quick, so if its staying off for longer than about 0.5 of a second its not the TS100 itself.

Its possible your power supply has bad filtering at its output stage and the slightly different PWM base frequency of the newer firmware is causing it to have issues with the load. Note that the "power limit" option limits the _average_ power, not the _peak_ power (as the hardware does not support limiting that). 

In the past all the issues of the iron randomly shutting down have been something up with the power adapter used.

Note: If its a software crash restart it will be extremely fast and will look like its just jumping you back to the menu. Which sort of indicates its not firmware crashing but a hardware issue.

michael-blue2, 2020-04-15:
> Is there any way you can measure the voltage coming from that power supply while soldering? These kinds of random shutdowns are almost always a power supply that is not coping in some way. The TS100 [...]

Having the detailed screen on both 2.09 and 2.05 the voltage stays at 19.5-19.4v No drop at all. When it shuts off it does need time like 2-3s just for the screen to show something. IF it wasnt about the 2.05v working perfectly, I would be 100% its the psu having the problem. I dont think the psu has bad filtering, Its a legit asus psu for my gaming pc. Even i cannot test the filtering cause of the lockdown and not having the tools home. Did you change the PWM base fr after the 2.05? from 2.06 and above?

whitehoose, 2020-04-15:
You can't trust the numbers coming from the iron, it's only a guesstimate based on  software settings not a true reflection of component fact.  The only safe way to test isn't with a soldering iron you need  an AVO or other dedicated measuring instrument.  If it was the TS80 you can use a USB tester but the TS100 is best tested with a "proper" meter
Voltage regulators etc work in a knife edge.  Take a USB power brick, I regularly test outputs with a digital load and a UM34c tester.  Plug it in and slowly increase the load and when you reach the threshold current the output voltage will crash from the boosted 4.9v to anything down to 2.3-3v which is when the lithium over discharge protection kicks in.

You're using a different source - but with switch mode supplies the principle is the same. 

The separation between working and crashing is usually only micro amps (and it's consistent and repeatable), As a guess if one of the firmwares is, for whatever reason taking slightly more current than the other (possibly say a slightly tweaked PWM frequency). You'll only see that small difference with a meter.

michael-blue2, 2020-04-15:
I got a multimeter: this are my numbers
**v2.05**:

12v psu 3.3a
On startup
11.85v Vd with 1.22A draw

Asus psu 19.5v  9.23a
19.2v with 2.05a

**v2.09**
12v psu
11.75v with 1.4A

Asus psu
It hold 19.2v with 2.25a, on heat up but just before the shutdown it drops to 5v and 0amps, holds at 5v for 1-2s and then hits 19.2v again and boots up. 

How can a software on the TS100 control the psu? On 2.05 everything is good

Firebie, 2020-04-15:
lets say, that before 'tipMass' was 35, after changes it was adjusted to 45. You can try to change this to prev. value, recompile and test. But the issue is with PSU, not with TS100:
-const int32_t tipMass = 3500; // divide here so division is compile-time.
+const int32_t tipMass = 45; // X10 watts to raise 1 deg C in 1 second

michael-blue2, 2020-04-15:
> lets say, that before 'tipMass' was 35, after changes it was adjusted to 45. You can try to change this to prev. value, recompile and test. But the issue is with PSU, not with TS100:
> -const int32_t [...]

May I know in which file is located? First time i am searching this git

michael-blue2, 2020-04-15:
[image]

Any idea? First time I am making a file here

Firebie, 2020-04-15:
you need to run from /workspace/TS100

michael-blue2, 2020-04-15:
> 
> 
> you need to run from /workspace/TS100

Dump me :D 
changed the number from workspace/TS100/Core/Inc/power.hpp
from const int32_t tipMass = 45
to const int32_t tipMass = 35
Just flash it and the problem is still the same

Firebie, 2020-04-15:
25?

michael-blue2, 2020-04-15:
Stilll the same

whitehoose, 2020-04-15:
It's not a case of the software taking control.  OHMS LAW RULES EVERYTHING!!!!!!!! ... you can't actually mess with that "fact" in your case Asus V=19v, the Tip = 8 ohms so **I**=v/r = 19/8 =**2.375A** (P=i^2R=2.375^2 = 5.64A x 8 = **45.125w**).

So 45 pixies are pushed down the pipe and burned to heat your iron.
However, the people here in the head shed are cleverer than that.  So rather than just building a big bonfire you push one pixie on the fire, this warms the head up but not all the way.
you count to 10 and throw on another pixie and the same happens.  Keep doing that 1 pixie every 10 seconds and the tip gets hot ... but only ever reaches 300c ... to make it hotter - you throw the little guys on quicker.

Even though it's electrickery Capacitors, inductors and thermocouples (to name a few) need time to react - so there will always be a small latency between lighting a pixie and getting the maximum heat.  So the numbers never quite give the answer you expect.
The head shed will play with the timings to get close answers - but when the sun shines the room gets warmer as does the iron and wires so maybe you only need 42 pixies.  So being mischevious the others overheat the PSU.
Or maybe Ralim drops the pixie count because of another problem - and your PSU despite being a good quality one has developed a slightly leaky capacitor which lets more current through causing overheating mischief which makes the PSU crash.

Recently I had a powerbrick which Is fine when powering LEDs etc, but when I connected an external boost converter the internal circuit frequency hetrodyned with the external and they started up a pulsing which wasn't good.  Another boost I have worked fine - so its easy to make the wrong conclusion (phew!).  Try another PSU or introduce a buck or boost into the mix which might react better (or worse!).  The hard fact is that not every component plays nicely with every other one. Just try changing hardware variables and you might solve the problem and get even better results (or maybe you won't, but there is always the danger that if Ralim fixes your "foible" it may throw out everyone else.  Firmware can sometimes be very picky.

Ralim, 2020-04-15:
> Asus psu
> It hold 19.2v with 2.25a, on heat up but just before the shutdown it drops to 5v and 0amps, holds at 5v for 1-2s and then hits 19.2v again and boots up.
> 
> How can a software on the TS100 control the psu? On 2.05 everything is good

@michael-blue2 
Software cannot control the PSU at all.
All the TS100 can do is draw power and assume the power supply can deliver it.
Those measurements point 1000% to a faulty PSU that cannot handle the load the iron is requesting.

Adjusting the tipmass wont really help here as its most likely the inital heat-up and peak response thats killing it.

The one thing you could try is adjusting the TIP PWM Frequencies but i dont recommend this as there is a lot of the timing that is build around assumptions of those.
[here](https://github.com/Ralim/ts100/blob/2b4cf54a616b2c853b449531929c19b525660f39/workspace/TS100/Core/Src/Setup.c#L317) and [here](https://github.com/Ralim/ts100/blob/2b4cf54a616b2c853b449531929c19b525660f39/workspace/TS100/Core/Src/Setup.c#L262)
Those prescalers allow you to adjust the frequencies of the PWM. Note that adjusting these will break the PID tuning and possibly make the tip temperature control unstable.

Firebie, 2020-04-24:
I did experimental version, which for first 3 seconds after tip is started heating, slowly adds power from 0 to 100%:
[TS100_EN.SlowPowerForFirst3Seconds.zip](https://github.com/Ralim/ts100/files/4531090/TS100_EN.SlowPowerForFirst3Seconds.zip)

paulfertser, 2020-09-12:
In case anyone here has some PSUs for which power limit doesn't work (and the iron reboots), I'm asking to try if #697 helps.

Ralim, 2020-12-29:
Going to close this one out has there has been no response since April. Please re-open/comment if you want it kept open 👍🏼

closed 2020-12-29 by Ralim