Ralim/IronOS #609: wrong voltage

discip, 2020-04-17:
* **I'm submitting a ...**
   - [x] Bug report

* **What is the current behavior?**
If the iron is plugged in, it sometimes negotiates for only 5V instead for 12V, wich I have set.
While being fed with the 'wrong' voltage, the power supply makes some kind of buzzing noises.
*After unplugging the iron und plugging it in again the behavior is as expected: **12V and no noises.***

* **What is the expected behavior?**
The iron should negotiate for the voltage, wich was set (*12V in my case*).

* **Is the issue reproducible.**
As I mentioned above, this issue occurs only occasionally, so the reproduction is a matter of luck or patience.

  * ***Steps to reproduce:***
    1. Plug in the iron.
    2. Watch the display and wait for it to show the voltage, you did not set it to negotiate for.

* **What is the motivation / use case for changing the behavior?**
To make the iron function as intended.

* **What are you running:**
  - **TS80**
  - Firmware Version: **v2.09 git**
  - PCB Version: **2**
  - Power Supply (**SDC-18W**):

voltage | amperage
-|-
3.6 - 6.5V |   3A
6.5 -   9V |   2A
9   -  12V | 1.5A

Thank you in advance
kind regards

Ralim, 2020-04-17:
This is just down to the general shittyness of QC. Most likely your chargers implementation of QC 3.0 is marginal for how the negotiation is coded (which follows the spec, but then was modified to be more tolerant).

Also note that your exceeding your chargers rating at peaks using a 1.5A charger at 12V (12V spikes at ~2.5A)

discip, 2020-04-17:
Does that mean there is no solution for that?
I do not recall facing such behavior with the previous versions.

Ralim, 2020-04-17:
Generally yes, not much that can be done.
Doubly so in that often if we fix this for one power supply, we break maybe 10-20 other models :(

This is often the case if you replug quickly into some power supplies, depending how fussy they are some don't like it and go into a safety lockout (5V) as well.

discip, 2020-04-17:
Ah ok, so I will have to replug the iron in such cases. :)
Thank you for clarification.

ps: There are some minor issues about the *scrolling description text*.
e.g.:
- The Auto Start is missing the 'O' option.
- And the overall appearance of the *scrolling description text* is not consistent regarding the ending: Some are without a period but others have even exclamation marks e.g. the *'boost mode'*.

Ralim, 2020-04-17:
For all text issues with the descriptions, these are in the json files and you are 1000% welcome to open pull requests to adjust this. :)

discip, 2020-04-17:
Ok.
Thanks. :)
Than I will try to do so.

closed 2020-04-17 by discip

discip, 2020-04-17:
@Ralim
Sorry to bother, but I am struggling to open a pull request.
Would you please provide some guidance?

thank you in advance

Ralim, 2020-04-19:
@discip  you should be able to go to the language file and click the edit button up the top right and edit it in browser, and then at the bottom propose the changes as a PR as far as I'm aware?

You can also fork the repo to your account, edit there, and then open a pull request from that fork into this main one as well 😀

discip, 2020-04-20:
@Ralim
Thanks! 👍