Ralim/IronOS #611: reading and changing temperature via microusb while using, possible?

ranomier, 2020-04-18:
* **I'm submitting a ...**
  - [x] question
  - [x] maybe feature request.

Is it possible to control the ts100 via microusb while running? So eg. reading and changing temperature.

I ordered mine yesterday, still waiting. I have already an mod idea. I want to build a optional soldering station with a little raspberry and a screen.

For the cable i would use a weller or other brand cable, because these have enough pins for power and microusb and there are flexible :)

Another question is the ground pin of the microusb the same as the ground screw on top?

Sry if this is not the right place to ask this.

Ralim, 2020-04-19:
This has been asked before; what it comes down to is that it requires a fair bit of work, and lowering the clock rate that we are running the IC at (64Mhz -> 48Mhz).
I've generally avoided this as their usb isnt massively stable either (no internal crystal, so its a bit hit and miss) but it certainly could be done.

All grounds are the same (power input, the top screw, usb).

ranomier, 2020-04-21:
Sry for the late answer.

What would be the downside of a slower IC. Is it a big difference or a small one?
Can a crystal be added from the outside? And if we can add a crystal can it be avoided to change the clock rate?

Thank you for your answer :+1:

Ralim, 2020-04-21:
Its a moderate difference, but also involves re-timing all of the timers and ADC sampling periods as they are all derived from that clock, and thats mostly a pain :sweat: 
No crystals can be added as the pins are not broken out on the PCB and its a QFP package (I mean, you could _try_ but its going to be hard).

There was an idea to use the trimming register to try and force the IC to run 72Mhz off the internal clock so we could try USB but that going well outside of datasheet specifications as far as I understand, and the USB is already bad enough as it is so didnt want to have to support it. 
Adding a crystal would avoid a clock rate change, but it would change the clock rate up instead of down.

USB needs a 48Mhz clock. On the F10x series you can only use USB when your main internal clock is either 48Mhz (div by 1) or 72Mhz (div by 1.5).
When running off the internal 8Mhz MSI, the PLL can only go up to a maximum of 64Mhz as there is a divide by two on the MSI path that is _not_ on the external crystal path.

To be completely honest, 80% of the reason I haven't packed in USB to the firmware is because of how unreliable it is, and it would generate probably another 30% more github issues than are already here from other things, and I dont have the time to deal with that volume of tickets :grinning: 

I'm not against investigating it, just its awkward to do so with, and make it reliable without either down clocking (which tbh we could do) or going outside of specifications (If this was reliable enough  i would actually prefer this)

ranomier, 2020-05-03:
Again sry for the late answer and thank you for your patience. Mmmh.

Thats so sad. I realy would like to build a station xD. Is there a community, irc, discord, telegram or something esle?

ranomier, 2020-05-24:
I will close this issue for now. Thank you for your detailed answers! :star_struck:

When i got time i will look further into this. Or maybe the mini company could release a new revision/version with a crystal. Maybe i will write them a email once i understood everything a little better.

```
 _____ _                 _                        _ 
|_   _| |__   __ _ _ __ | | __  _   _  ___  _   _| |
  | | | '_ \ / _` | '_ \| |/ / | | | |/ _ \| | | | |
  | | | | | | (_| | | | |   <  | |_| | (_) | |_| |_|
  |_| |_| |_|\__,_|_| |_|_|\_\  \__, |\___/ \__,_(_)
                                |___/               

```

closed 2020-05-24 by ranomier