Ralim/IronOS #616: TS100 works with MINI F/W, blank display and inop with Ralim

Bevanj86, 2020-04-24:
My TS100 iron had previously installed Ralim (unsure of version). Started using yesterday, did ~5min of soldering.

Went to use again today, and no display - iron seemed completely dead. Tried with multiple power supplies. 

1. I flashed Ralim 2.09 english precompiled hex. Got .RDY extension - Still blank display. 

2. I downloaded and flashed standard mini firmware - iron works fine. 

3. I reflashed Ralim 2.09 - back to a blank display, and non-working iron. 

4. Reflashed original firmware and used for a while longer. With the original firmware, the display glitches out sometimes. It winds up pixelated, or missing a line. 

Any idea what could be causing this? I've checked the 3.3Vreg, and that's fine. Stable output with both firmwares. 

Also to note, in Ralim 2.09, if I plug the iron in, and try and push the button to turn the iron on, it goes to the "DFU:3.45" display for loading firmware. Very odd. 

Any ideas what is wrong with my iron? Have I got a sick STM32? 

Also, is the TS100 bootloader hex available? I have all the right tools around to replace the STM32 on the sub-board. Right now it's my prime suspect. Thoughts? 

Also, why would one firmware function (but glitchily), when the other doesn't work at all? 

Cheers,
Bevan

  - Firmware Version: 2.09
  - PCB Version: (1/2) How to tell? Mine is marked 2.58
  - Power Supply (Voltage and Current Rating) : Dewalt pack & 24V 6.5A Meanwell.

Bevanj86, 2020-04-24:
Interestingly, my issues sound virtually identical to some people in this thread: 
https://github.com/Ralim/ts100/issues/298

I've just tried pulling the accelerometer off the STM32 sub-board. Still the same - blank display with Ralim V2.09, and semi-functional with OEM firmware until the screen goes randomly glitchy.

Bevanj86, 2020-04-26:
Well.... 

I pissed around and lifted the display to measure the I2C PU resistors, and then tried uploading Ralim V2.09 again with a 3rd combination of computers and micro-USB cable, and I now have a working iron! Successful firmware load was with OSX.

I'm not sure if I've disturbed a dry joint when I was poking and prodding at the 4k7 pull-up resistors, or if it was the dud computer-cable combination. 

I'm still puzzled as to why it dropped the firmware in the first place though.

Ralim, 2020-04-26:
@Bevanj86 The only time i have seen these kinds of issues when hands on with a unit, it was one of the I2C resistors coming off the board causing intermittent issues with the I2C bus.

Generally, the STM32's dont just randomly drop the firmware thats loaded in them, so if the file was re-named correctly (i.e. programmed) then the firmware _should_ be good. 😢

Bevanj86, 2020-04-27:
Thanks Ralim, 

My suspicion also lies with the I2C pull up resistors. I might get in there and reflow the joints. The factory soldering on this iron's PCB doesn't look too flash. Rather ironic for such a fantastic soldering iron. 

Like you, I too have never seen a programmable micro just drop its firmware out of the blue. 

It is odd how someone in the other threads had a very similar report though - that their iron was working perfectly with a previous versions of your firmware, then died, and be functional with the OEM firmware. Perhaps a coincidence. 

Is what I've written above true? That your firmware conducts a readback after writing data to the display across I2C, where as the OEM firmware just conducts a write? How does your firmware behave if there is unexpected I2C data returned? Is there some form of exception handling, or do things grind to a halt? 

If there is readback on yours, but not the OEM, it might explain why I2C bus problems will allow one firmware to function, and the other not to. 

Bevan

Ralim, 2020-05-29:
My code base uses the hardware I2C, which has good and bad sides. Its not tolerant of weird hardware quirks that I2C gets with bad pullups, but it is also generally _faster_.

It also checks the ACK bit for every transfer, and will fault if its is not handled correctly.

I would be inclined to inspect the pullups, and if that fails if you can get manage to check basics like the 3.3V rail would be excellent too (it should be on the caps on the daughter board).

gutschke, 2020-06-05:
In an ideal case, if you notice that hardware I2C keeps failing, you should switch to a best-effort software implementation. It would be slower, but it would give you the option to display a message informing the user of what is wrong with their hardware.

Ralim, 2020-06-06:
I agree in an ideal case, but I2C is such a mess, that I'm inclined to _not_ support this mostly just due to developer time. There is some code in there to try and handle bus lockups and "unstick" the bus when this happens, it doesn't overly help when ACK's are failing on each write.

Happy to take a pr to add this feature naturally :joy:, but without hardware to test on nicely I dont really want to dig into this rabbit hole _yet_. Since its still just speculation as well around this. If we could 100% prove that its I2C lockups then working on this makes sense.

ArtemTomskiy, 2020-06-14:
Hello! The problem is relevant !!! I’ve rummaged around the Internet for a year in search of an answer! And so, there is also a TS100 soldering iron, and also does not work)
As with everyone, your firmware was uploaded (which I don’t remember already)
I got up normally, soldered to them rarely and every 7-10 times during operation, it just cooled down, reconnected the power, it turned on but the screen fell, decided to reboot and everything is said above, black screen or DFU 3.43
it is extremely strange that with firmware from here
http://www.minidso.com/forum.php?mod=viewthread&tid=892&highlight=TS100
It turns on and even tries to work, but the screen immediately rolls in and there is no stable work!
other firmware leads to black screen and DFU 3.43

Ralim, 2020-06-15:
@ArtemTomskiy 
Most likely this is the I2C bus issues as mentioned above ^. As this firmware enforces correct ACK.NACK. The stock firmware tends to ignore these so it will "sort of" work until something misbehaves on the bus and then the bus will crash.

I recommend opening the iron and checking that the daughter board is mounted correctly as a first step. Otherwise generally just inspecting the board may lead to hints of the damaged / playing up component.

karas288, 2020-06-22:
My new ts100 has the same issue firmware won't write the first time got to drag and drop then wait to auto reconnect then drag and drop again then it will work however if you then try to turn it on either from a power source or usb the screen will flicker then go blank I've put the stock firmware back on and it functions but gets terrible screen tearing sometimes instantly I've read on a YouTube comment that the accelerometer and display both use the same i2c lanes and that removing the accelerometer fixes it do you think its the resistor or a component as mentioned here previously or is it just a case of remove the accelerometer and it "should" work?

ArtemTomskiy, 2020-06-22:
To be honest, after I saw and read this post, I took out my TS100 again, in the hope that I could revive it, only after turning it on I found out that the screen now doesn’t work at all.
The firmware was always installed the first time, but did not work as it should.
probably this was the last attempt to revive him.

karas288, 2020-06-22:
If it is definitely a pull up resistor that's not 100% touching where it should then maybe we can re solder the resistor and get it working mine turned up this morning so it's not something I've done to it and there was a slight glitch the first time I turned it on I've opened a dispute with eBay but ideally I'd like to keep it since it comes with everything I need to get it running already and its here in front of me waiting to get to work. I can (with the stock firmware) access the config file and change the temperature etc in there and just wait for it to get hot before using it but then why buy a "smart" iron in the first place and from what I've seen/read we aren't the only ones having this issue.

karas288, 2020-06-22:
Right got it sort of working but had to use the old stock firmware but it only works for a little while before going fuzzy again

Ralim, 2020-06-22:
@karas288 
Is your iron an actual legit miniware iron ?
If it is reach out to their service email and they should help you. 

It's unknown to me what exactly is failing but the i2c bus does appear to start locking up with either the oled or accelerometer misbehaving.

I would be curious what revision of the internal board you have if you open it and get a photo.

karas288, 2020-06-23:
@Ralim I will have a look at it in the morning looks legit sounded legit but not 100% was on eBay as a kit for £53 GBP with power pack, lipo connector, stand, the iron and a choice of 1 of 3 tips think I chose the bc2 came in a black card box with silver image of the iron, logo and ts100 soldering iron written on it don't know if that helps to check legitimacy or not lol

karas288, 2020-06-29:
[image]
@Ralim is this pic ok to figure out the version?

karas288, 2020-06-29:
Just seen where it says v2.51a lol

Ralim, 2020-06-30:
Certainly looks legit 😅 

Honestly, if its misbehaving like that I would contact miniware and see what they say 😓  Hopefully they get back to you. Otherwise could be loose connection or OLED screen dying.

karas288, 2020-06-30:
Well I moved the pad that's on the back cover that sits on top of the daughter board and it seems to last longer before it goes mental so I think it's that connector that's at fault I managed to get it to turn on and heated (soldered the end of a wire as a test) only time it seemed to go dodgy again was when I tapped the end to get rid of the solder so it lends weight to my thoughts that it's the connector for the daughter board that's at fault

closed 2020-09-14 by Ralim