Ralim/IronOS #617: clean logo ts100 not working

tomadam96, 2020-04-25:
* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**
I want to remove the bootup logo. 
but when i try to flash the file (with _clean_logo.hex) it create an .err file and the logo remain.

the installation method is working (i try with the last ts100 firmware and it work perfectly), but the _clean_logo.hex create this error.

I already try with command copy from windows but not working.
I already try with linux and windows too(same problem)
DFU 3.45

***Steps to reproduce:***

1. flashing firmware  _clean_logo.hex
2. 
3.

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

pocketscience, 2020-06-04:
Also seeing the same issue from my Mac. DFU (3.45) connects and firmware install works fine, but uploading a logo file generates a .ERR response.

pocketscience, 2020-06-04:
Should mention that multiple retries didn't change the end result..

Ralim, 2020-06-14:
This is an odd one; since the clear_logo is just a logo file with a blank marker line basically; so should behave the same.

I have noticed that you are using DFU 3.45; most of the logo workflow was tested on 3.43. I'll have a look at updating my iron and see what I can find :)

Ralim, 2020-06-15:
@pocketscience 
Could you try testing flashing this file; then unplug the iron; and then flash the firmware again.

This file will erase all of the user flash :)
https://github.com/Ralim/ts100/blob/master/Development%20Resources/force_blank_flash.hex

pocketscience, 2020-06-15:
Sure, will give it a crack tonight.

pocketscience, 2020-06-17:
No dice. I installed the hex file, then reflashed with latest, but I still can't load a logo file... I tried both copying via Finder and using terminal... multiple times at each but each time got the .ERR response back... Oh well, at least the firmware installs cleanly and my TS100 is functional. If there's anything else you want me to try let me know..

Ralim, 2020-06-17:
@pocketscience 
Can you share which logo you are trying to flash? Wonder if something is bad with its hex file..

pocketscience, 2020-06-17:
The one named "001_TS100_Left.hex"

Ralim, 2020-06-17:
I would suggest trying another logo first to see if that helps but ill test that one on my units tonight too.

CSchlipp, 2020-07-06:
Got a new TS100 today and flashed the latest firmware successfully with a Windows PC. 
Upon copying over one of the prebuilt logos, I also repeatedly get an ERR-file.

squee666, 2020-07-11:
got a new ts100 today
flashed fine but didnt show

instead it corrupts the firmware text layout
like the text information for power sorces etc was missing/moved

Ralim, 2020-07-12:
@squee666 What file did you flash?
That is usually indicative of a file using the old flash offset that no-longer works :(

squee666, 2020-07-12:
> @squee666 What file did you flash?
> That is usually indicative of a file using the old flash offset that no-longer works :(

It was a self made logo. That was converted with logo editor exe.

Made sure it was the exact dimensions as the ts100 on git.
Then made it using indexed color then changed it to rgb so it was clean bw

73VW, 2020-07-15:
@squee666 I've just got the same problem as yours...

squee666, 2020-07-27:
> @squee666 I've just got the same problem as yours when trying with the old image editor ".exe" file.
> I've then reuploaded the latest firmware and used the [python converter](https://github.com/Ralim/ts100/wiki/Logo-Editor#the-python-logo-converter). (You can find its latest version when downloading the source archive under "Bootup Logo\python_logo_converter".)
> 
> It worked this way.

yeah worked for me this way 👍 
anyway to extract the logo from the ts100 again lmao mine looks almost perfect but wanted to tweak it and deleted the png and hex

Ralim, 2020-07-27:
There shouldn't be any more hosted copies of the older exe converter I believe. so not sure where its coming from. (Point me to it if its still up, as the early builds had the wrong flash offset).

Only way to really extract would be to use a debugger to read back out the flash and re-assemble that (not trivial)

campbellcole, 2020-08-01:
Yeah I had this same problem, got the exe from the repo [here](https://github.com/Ralim/ts100/releases/tag/v1.15). Will try the python script now and edit this with the results.

awkimball, 2020-08-02:
So I got a new TS100 today with DFU 3.45 and flashed the newest release firmware, everything went great. However, by default there is no boot logo and when I try to flash the 001_TS100.hex standard one, I repeatedly get .ERR. Is this related? It seems pretty similar to this issue. Thanks for any advice in advance

campbellcole, 2020-08-02:
@awkimball yeah none of the examples worked for me but using the python script with a custom image worked

PR #1232 (2022-03-13): Logo rework

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [ ] There are no breaking changes

* **What kind of change does this PR introduce?**

Support for new logo support that (a) fixes the old endian issue and (b) allows animation.
Removes logo's from this repo and instead places them in the Meta repo.

* **What is the current behavior?**

* **What is the new behavior (if this is a feature change)?**

* **Other information**:

Closes #1173
Closes #1228
Closes #617
Closes #446
Inches closer to supporting #292
closed 2022-03-13 by Ralim