Ralim/IronOS #619: [hardware support request] Custom TS100 with native USB-C Power Delivery (up to 100W)

tloki, 2020-04-28:
* **I'm submitting a ...**
  - [ ] Bug report
  - [ x ] Feature request
  - [ ] Translation

Here is a drop-in replacement of TS100 logic board that supports USB-C PD protocol - open source quick-charging protocol that is able to deliver up to 100W @ 20V. If your notebook supports USB-C charging, it is 99% PD protocol. Also, every Android phone since 2019 needs to support PD as common charging protocol (so manufacturers do not implement so much proprietary solutions and for one to be able to use fast charging on other phones). 

https://github.com/Jan--Henrik/Otter-Iron/tree/master/firmware

I think I have said enough about why we need PD support for our soldering irons. Although technically it is not "TS100" anymore, it would be nice of MiniWare to start producing this revision.

Firebie, 2020-04-28:
Or just buy a cable https://www.ebay.com/itm/USB-C-to-5525-5-5mmx2-5mm-Plug-PD-Charging-Cable-Laptop-Power-Adapter-Cord/392447289682

tloki, 2020-04-28:
Not really practical since you have to own charger with female USB jack (some of them - Apple, Lenovo, Dell don't have it).

I guess if you are trying to use it with usb triggerer this might work:

aliexpress.com/item/4000942438346.html

KieranWebber, 2020-05-14:
> Not really practical since you have to own charger with female USB jack (some of them - Apple, Lenovo, Dell don't have it).
> 
> I guess if you are trying to use it with usb triggerer this might work:
> 
> https://aliexpress.com/item/4000942438346.html

I have one of these (slightly different model) that I use with a macbook charger and a silicone extension cable I made. Works great and saves me buying a DC power brick. You do need to make an extender or custom cable though as the normal cable is too stiff and it's too much weight near the iron.

ElRoberto538, 2021-02-14:
I'm looking into tweaking IronOS to work with the Otter-Iron Pro - It should also be compatible with the Otter-Iron. If anyone's done any work on this please let me know so I'm not duplicating work.

Ralim, 2021-02-17:
@EIRoberto538
This should be quite doable, as you will mostly need to change whats in the BSP folder (make a new one for the Otter Iron).
I looked into the Otter-Iron but didnt get it finished as I didnt have time to implement control for the USB-PD interface used.

As that tip supports temperature readback while driving the tip this firmware will needlessly insert a break for that if you copy the Miniware code. You can easily optimise this by just adjusting the timings to take an ADC sample at the end of each PWM period.

Otherwise its mostly a 1:1 copy to the Miniware code, change out the STM libraries to match the new STM and update all the hardware interface code to account for new pins / timers.

ElRoberto538, 2021-02-17:
@Ralim Awesome - sounds like I'm on the right track then, that's what I've started doing. I'm currently implementing the code for BSP.c, I'm mostly using the original otter iron code for this as I'm a bit of a c/embedded programming noob, learning as I go. I'm making some progress though! Pins etc are all setup, the makefile is updated and the bsp folder is setup and tweaked (except bsp.c).

I might ask for some help/advice with optimizations once I get it running, I read some of your blog posts on how you implemented things but I think I'm lacking the needed background info at the moment.

discip, 2022-04-23:
@tloki
Since the **TS80P** and **Pinecil** are available for some time now, this can be closed. 😊

thanks

closed 2023-07-21 by Ralim