Ralim/IronOS #622: ts100 power supply wont boot the soldering iron

Nikita-Philippe, 2020-05-03:
Hi everyone.

I'm submitting a problem with my ts100.

I was trying to make a power supply for my ts100 but the dumbass i am made a short circuit. Sice that, when i plug the regular 12V 3A DC power supply, the screen keeps black, and the soldering tip doesnt heat up.
I made some tests and finally find that if i plug the microusb port to a phone charger (for example), and then plug the dc power supply, my iron heat up, totally normally.
So, my splendid brain deduced that the *5V?* wich power the stm32 doesnt exist anymore. I search for a fried components bu absolutel y didnt find it. 

My question : what is the component that step down the dc power to 5V (or less i dont know really)?

I can replace it, i have the equipment, but i cant find it, even on the schematics.

If you want more informations, or photos, just ask me, no problem.

Thanks everyone for the ranswers.

  - Firmware Version: 2.06
  - PCB Version: (1/2) : dont know ?
  - Power Supply (Voltage and Current Rating) : 12V 3A DC

Ralim, 2020-05-04:
The schematics are available over here : https://github.com/Ralim/ts100/blob/master/Development%20Resources/TS100%20V2.46%20Schematics%20V1.0.pdf

The STM32 and oled run off a 3.3V supply, so U2 manages stepping down either the USB or the DC in to the 3.3V supply required. 

if your iron works fine when powered from the 5V input but not when powered by the DC input jack, its most likely you have damaged D4 or R30.

Nikita-Philippe, 2020-05-04:
Thanks.
I check the resistor and the diode and yes, thei values are tottaly wrong. 
I find the reference of the diode, but i dont know what is the reference of the resistor...

whitehoose, 2020-05-04:
Sizing components can be a bit tricky if you're not familiar.  
The 1N4148WT diode is https://www.diodes.com/assets/Datasheets/ds30396.pdf
p4 shows it as 1.15 to 1.25 mm with 2x 0.1mm tags so 1.2+0.2=1.4mm x 0.8mm +/-0.1mm

That probably makes the 150r resistor a 0603 - but without seeing it it could be 0402 or 0805

Download and print "Example of component sizes, metric and imperial codes" as a reference. https://en.wikipedia.org/wiki/Surface-mount_technology.
The proportions are spot on once you have the scale right (it's a little fiddly but size the 1x1cm and 0.5x0.5" squares - the rest should be close enough for comparison.

Both should be easily (and cheaply) sourced

Nikita-Philippe, 2020-05-05:
Thanks for the info. 
I search and yes, i find it easily.
I close this thread, i will reopen it if it dont resolve my problem.
Thanks you very much.

closed 2020-05-05 by Nikita-Philippe