Ralim/IronOS #625: [Feature request] Drop detection

goscicki, 2020-05-13:
* **I'm submitting a ...**
  - [ ] Bug report
  - [X] Feature request
  - [ ] Translation

It is sometimes easy to unknowingly pull the iron off the table by the cord. It seems like you could use the accelerometer to detect if the iron fell on the floor and shut down to prevent a fire.

Ralim, 2020-05-13:
This has been bought up before.

It doesnt actually really help you much since the iron will still take quite a while to cool down so things will still melt / be at risk. Secondly, drop detection is a bit of a pain to detect drops vs someone just throwing iron back in the stand a bit too roughly. :sweat:

goscicki, 2020-05-13:
Thank you for taking the time to consider/respond to this!

Yeah, I know it would still be hot, but melting something for several seconds is far better than 300+ degrees for several minutes. More likely to cause a fire. What about a higher sensitivity threshold for the drop and a menu to disable it if desired?

Ralim, 2020-05-13:
Your welcome to write it :sweat: 
I did have a crack at it a while back, but found that it had a 50/50 chance of going of when putting back in my holder or pulling out of my holder.

Generally the common knowledge of don't leave _anything_ hot unattended will prevail here :)

JohnEdwa, 2020-05-14:
Hm, if it had a bunch of additional checks as well, such as keeping track of the tip temperature/heater status to see if it was just being used for soldering, what the orientation after the "drop" is compared to the last time the iron was idle to check if it's actually in a holder and so on, it might be possible to make it work well enough.

I wonder if a compromise of making the iron go to the idle temp would be unobtrusive enough, yet a good enough safety feature to be worth implementing?

goscicki, 2020-05-14:
Yeah, Idle temp would be just fine.

Vinigas, 2021-01-04:
#314

Ralim, 2021-09-07:
Going to close this and just keep #314 as this is a duplicate

closed 2021-09-07 by Ralim