Ralim/IronOS #628: Looking for ts100 fw version

Scoredx, 2020-05-21:
One time there was a version where there instead of power in watts was power in % and it was my favourite version can someone help me find it ?

Ralim, 2020-05-22:
Is it your favorite because of just that feature?

Scoredx, 2020-05-22:
No, it was very stable and I used it since I bought it

Ralim, 2020-05-22:
I do not take down old versions, so they all still exist in the releases page. Most likely look for the one where we introduced the estimated power and then try the previous ones.

Mostly curious as to what has regressed for you in later firmwares.

Scoredx, 2020-05-22:
In 2.0.6 afrer getting to designed temp I can hear beeping from under the oled

Scoredx, 2020-05-22:
The temp reading was ok for me

Ralim, 2020-05-22:
Interesting, your iron's caps must resonate nicely at the newer PWM frequency :|

Scoredx, 2020-05-22:
Like 2 of my friends have problems with temp reading too

Scoredx, 2020-05-22:
in the 2.0.6 and the newest one

discip, 2022-04-23:
@filipszewczyk99
I think you will be surprised how much better the latest release is compared to the one you are referring to.
So try the latest release and close this issue please. 😊

thanks

closed 2023-07-21 by Ralim