Ralim/IronOS #630: [Long] TS80P Thead

Ralim, 2020-05-22:
Creating an issue to track the upcoming TS80P.

_I DO NOT HAVE ONE YET_.

So anything said here is speculation at the moment 😁 

My current understanding I have pieced together is that it is possible to just add support for the USB-C to the current TS80 build. I believe it is a USB-C interface attached to I2C, so expecting adding support shouldn't be too hard as will only require handling its configuration.

My plan is to acquire a TS80Pas soon as possible and get to work on the firmware side.

I have tried and we are not getting schematics _soon_ (will eventually exist apparently).

I'll update this top message as things happen.
Try to keep on topic in the comments 😆

discip, 2020-05-23:
Ohhhh ... how I have been waiting for this! 👍 😁
Mine arrived yesterday!
I am looking forward to this becoming my **# 1** soldering iron!

1. Although this thing is better than its predecessor regarding *heat-up time* and *max temperature* (Might be even more, I am no expert to evaluate this), I do not like the new *'sliding-handle-piece-thing'*.
2. And I noticed the silicone USB-C to C cable being a bit stiffer, than its counterpart from good old ***TS80***.
3. I do not know much about the internal of an USB-C plug, but it looks like there are missing some pins.

Nevertheless I think you (and everyone contributing to this) are going to make this thing even bigger than its predecessors including the TS100.

a big THANK YOU in advance
kind regards

Ralim, 2020-05-23:
@discip Ah awesome :) 

1. Yeah I'm not sure what i will think of the metal handle slide. I wonder if it will fit in the older TS80 plain case if you prefer its style? (I'm only guessing)

2. I'm sort of expecting that, it would have extra conductors in it i guess :| Still should be better than the ones i have here 😁 

3. I would expect it to be missing the USB3.0/ high speed data pairs as they would be useless. There are USB-C plugs I often use that only have the USB2.0 pins fitted as they are much easier to solder. Possible its that style of connector.

I'm hoping mine will turn up soon. Once it does expect me to start poking at its support 😀

discip, 2020-05-23:
@Ralim
Good evening,
thank you for your instant reaction.

1. Good news: The original *TS80* sleve fits perfectly onto the new ***TS80P*** because the housing is exactly the same! (Because of the fact, that I do not like the slightest wiggle in the tip I am holding directly onto the 'muffler' (dissipator) and that way the tip is even shorter, which I prefer as well. So the most logical call was to replace the new handle with the approved one)

2. Don't get me wrong I am happy to be provided with such elaborate piece of equipment, but the first thing that came to my mind, was about eDesign having used lower quality wiring. Later on I considered the extra mass for extra power would add to the stiffness.
One more thing about the accuracy of fit of the cable plugs: It is very tight!

1. I just thought distributing the 'power' onto as much wires as possible, would get the job done even better. 🤔

I am looking forward to this, almost impatiently. 😅

regards

whitehoose, 2020-05-24:
Well I've retired!  All the virus stuff has given me the opportunity to jump ship on very good terms.
With that in mind I wont be upgrading, because now I have virtually no need for portability - my existing kit will do.  I still have about 20Kg of lead solder so I won't be needing more than 340c either.
Interesting to see the new kit - looks even more like a weller.  If I've got the right end of the stick isn't USB3 already compatible with PD?
I thought if you use a QC3 device with a c to c cable it naively uses all 6 lines because the controller doesn't need to make provision for backwards compatibility with earlier usb standards.
As I understand it (if I do!) the USB3/QC3 and PD aren't in competition they are (and always have been) compatible, you just get given lower voltage\current if you plug qc3 into PD. 
"Traditional" USB C to USB A qc cable converts 6 wire to 4 wire while C to C uses 6 wire comms and can talk to PD even if it cant take 12v.

discip, 2020-05-28:
@Ralim
Good evening,
due to the fact, that I got an US wall plug, although I need an EU wall plug, I have to by one myself.
But as you most likely know, there is an huge amount of wall plugs to choose from.

My focus is to get a reasonably **small**, **well build** (quality wise) and **not** to **pricey** one.
But there is more, I want to consider, before spending the money.

Regarding the specs of the unmodified **TS80P** its max power draw amounts to **`30W`**.
Since I am not familiar with the internals of the iron and its specific capabilities and considering the fact, that your firmware *"overclocked"* the power draw of the original *TS80*, I am wondering, if you think about doing the same to the **TS80P**? (Do you already know, if it is possible to push the limit without compromising the device? I am really all about, not shortening the life to this ***beautiful piece of art**.* 😍)
 
I hope you got my point. 😉

Since there are plugs providing up to **`65W`** of power, but these are somewhat monstrous regarding its size, but also ones with **`30W`**, I am wondering if the later ones are sufficient, regarding the 🎆✨***upgrades*** 👏🎆 you are planing to provide. 😁

thank you in advance
kind regards

Ralim, 2020-05-28:
@discip 
Indeed this is a good question.
By back of the envelope calculation, I think it will peak at around 32-34W. So I would lean towards a 35W or 45W small charger if you can pick up one easily locally. However, the firmware will keep the current ability to limit power, so even a 30W will probably be totally acceptable. 

I have a 30W adapter here that I have totally pushed to 35W (gets a bit warm, but its done it for like 2 years now). So there is probably some room in there too.

I would pick either 45W/35W/30W depending on size and ease of picking it up :)
If the USB-C solution they have used works as I _expect_, then the iron can query for what limit to use, so it should "adjust" to match the adapter.

discip, 2020-05-29:
@Ralim
Good afternoon,
here are some pictures, which, I hope could be helpful. 
These are the specs of the wall plug:
[image]

This depicts the pins of the top and bottom of the same side of the power cable:
[image]

This shows the pins, top and bottom, of the irons inlet:
[image]

This is obviously the top side of the PCB:
[image]

And here is the corresponding bottom:
[image]

This was really not easy to achieve. Especially taking the pictures of the USB-C sockets / plugs was some tedious work.
Nevertheless, I hope you could use these. 😁

regards

Ralim, 2020-05-29:
Glorious!
These are highly appreciated.
I'll probably dig into these more in the coming days, but already those confirm most of my thoughts which is excellent.

Looks like a dedicated USB-C PD IC, so compatibility should be good 😄 .

Those USB-C plugs are like I thought, no high speed signal pairs (USB3/thunderbolt).
Pins present:
* GND
* VBUS
* USB2.0 pair
* CC pins
* SB
* Vconn

Which is great as it looks like _any_ USB-C cable will work, nothing "special" other than the nice silicone outer.

Interesting to see PD+QC on the same plug on the adapter as, as far as I know, you not "meant" to do that 😂 .

discip, 2020-05-29:
@Ralim
That's great! 👍

> Interesting to see PD+QC on the same plug on the adapter as, as far as I know, you not "meant" to do that 😂 .

I do not know, if I got you right 🤪, but the depiction of the specs is actually a collage of the back and, at the far right, the front of the US wall plug:

[image]

Unfortunately on *none* of the sellers pictures the front is to be seen.

The only reason I added this, is so you can see the specs of the originally provided wall plug. 😁

Hmurik, 2020-06-04:
Look forward to. My TS80P got into trouble, something went wrong while trying to change the bootup picture. It seems I accidentally messed up the firmware. When connected to a power source, the soldering iron does not work - only the default bootup picture lights up and nothing else happens. When connected to a computer, I see a virtual disk, the boot module for updating the firmware works, but there is no hex file for TP80P to flash it. A huge request - as soon as you have a soldering iron, lay out first of all the native firmware. Thank!

Ralim, 2020-06-04:
Mine hasn't arrived yet but when it does ill definitely be racing to get a tester firmware up.
You could possibly contact miniware for a firmware for what came on the iron originally. But if not ill try and dump it as well in the meantime for you.

DjordjeMandic, 2020-06-04:
[image]
What is this port protection? OVP protection or what? I cannot figure it out. The iron is around $80 on banggood especially now if you preorder it. Is it good idea to order it right away during the discount or wait a bit for few pcb revisions in case there are hardware flaws?

Ralim, 2020-06-05:
I would be guessing that its just the OVP protection in the USB-C interface being used for marketing?

In the past there hasn't really been PCB revisions (other than the "upgraded" clone irons that are missing protection circuitry).

Hmurik, 2020-06-05:
> contact miniware
I sent a letter to miniware while they are silent. As soon as there is some result I will inform.

discip, 2020-06-05:
@Ralim 
Sorry but I have to ask again, do you think this [one](https://www.aliexpress.com/item/4001026924826.html?spm=a2g0s.8937460.0.0.737c2e0eqbuM3p) would work for the **TS80P**?

ps: What do you think about ***GaN*** technology in general, do have any experience in this field yet?
I read in some comments, that some seem to produce noise, compared to conventional technology.

thanks in advance
regards

gutschke, 2020-06-05:
I have a GaN charger (2*USB-C/100W, 2*QC/18W). Works wonderfully for all my USB powered equipment. Haven't received my TS80P, yet, but should be here in the next couple of days. I'd expect it to work fine with that charger. From my limited experience, I can absolutely recommend GaN technology.

whitehoose, 2020-06-05:
@Discip.
Really it's back to square 1, As things stand, taking what happened when the 80 launched Its possible that a wallwart that works with an 80 won't work with an 80p or vice versa, the only way to know for certain is to try... or if Banggood or Ali are bundling - buy the wallwart they're using.  

Hopefully the manufacturers have learned and improved their kit to expand the list of compatibility - but certainty ... only when you have one plugged in making the iron melt lead can you say yes for certain.

Small with big currents and voltages is never going to be a good idea - it also depends on your standing voltage.  The US use 110v, you can virtually lick the prongs to test them - but UK, EU and Aus uses 240v - that will give you a very hard slap if you extend your tongue, it will even ump the wall to get to you - with 240v you need physical separation and ideally big air gaps between tracks to ensure safety.
I have a 5.2Kw 5v supply I bought from work - it still only delivers 18w to my ts80 - total overkill and would be a waste of money if I'd bought it just for the iron, Ohms law is tough - don't be tempted to over spec power (but you need a bit of wriggle room) - everyone lusts after the 1.21 Gw model - but if you're not time travelling it's just a waste  The pixies only move at their own pace, unless you change the parameters of the path they won't speed up.

Ralim, 2020-06-05:
@discip By my eye that looks like it will work 100% fine. Its USB-PD spec compliant, it has the 12V option, and it has adjustable voltage (which is nice). I see no reason against it.

I've only used some GaN chips, and for all of the power applications I have used them in they have worked _perfectly_ no noticable issues at all. So i wouldn't be worried about it. If there are issues with the charger its probably due to _other_ stuff :joy: .

At least PD is a fairly good spec, so they all should work :)

discip, 2020-06-06:
@gutschke
Thank you!
@whitehoose
Thank you for your advice regarding safety and high voltages.
@Ralim
Thanks that sounds very promising, I think I am going to try the above mentioned one.

Hmurik, 2020-06-08:
Today, firmware for a soldering iron has become available for download:
http://www.minidso.com/forum.php?mod=viewthread&tid=4085&extra=page%3D1
I was able to solve the problem with my soldering iron - it works great!

brencerddwr, 2020-06-17:
I fully understand having a life away from stuff like this.  I am just wondering about progress on this.

Ralim, 2020-06-17:
@brencerddwr 
Your welcome to ask :)
Indeed I _do_ have a life :sweat_smile:  But I have also spent a few hours on this so far; where I'm at is:

Working:
* Accelerometer
* Tip PWm
* Handle raw ADC measurements
* QC3.0
* OLED

Not yet working:
* USB PD
* Handle temp sensor

Have been focusing on the USB-PD and its just a nightmare to debug with the setup in the iron; I have ordered a breakout board for the FUSB302 chip inside the iron; but also will continue to test it as I have time.

Handle temp sensor is odd; not sure if they have moved the pin or changed something in the schematic. Will most likely need to trace the connection to figure that one out; but should be fast.

Main reason its a pain is that the FUSB302 is just a PHY so it doesnt actually contain any of the logic for talking to the power adapter; that all has to be done firmware side.

denimjeans, 2020-06-18:
@Ralim 
Thanks for looking into this. 
I got my TS80P as well. Looking forward to use the boost function :)
Regards from Germany

brencerddwr, 2020-06-18:
@Ralim, when you get to the point of the USB PD and handle temp sensor working, I would be willing to beta test for you.  If I understand correctly, since it uses DSU to update firmware, it should be almost impossible to brick my TS80P.

gutschke, 2020-06-18:
I'm excited that PD is implemented in software. In the long run, that should give a lot more flexibility. But in the meantime, it's just a lot of extra work.

On the other hand, I suspect that an initial release without PD would still be useful. Even if only QC is enabled, the device could be used. Features would be a little limited, but that's OK for beta testing.

yawor, 2020-06-23:
@gutschke it depends. It gives more flexibility in negotiating the PD parameters, but at the same time the firmware is going to need extra space in the STM32's flash.
If I remember correctly, the TS80 used the same STM32F103T8U6 MCU as the TS80P, so the amount of the flash memory is the same, but TS80P is going to need PD logic imlemented.

STM32F103T8U6 has 64KB flash. Current TS80 firmware size is 36KB + 16KB for the DFU bootloader. This leaves "just" 12KB. Some of it is also used for storing user configuration (probably not very much).

I hope @Ralim finds a way to fit everything there :).
BTW I've just received my TS80P, so when there's anything to test, count me in.

yawor, 2020-06-23:
Few interesting facts:
- TS80 doesn't work with a USB-C to USB-C cable - even the one packaged with TS80P.
- TS80P does work with both USB-C to USB-C, as well as USB-C to USB-A cables.
- I've been able to power TS80P with an original 45W PSU for my Dell XPS 13, but the voltage doesn't go higher than 9V - probably because the PSU supports 5, 9, 15 and 20V only, so the iron negotiates a voltage not higher than 12V.
- I've been able to power TS80P using a C to A cable from QC3.0 wall charger and QC3.0 Xiaomi Mi Powerbank 2 - the same I use with TS80. On @Ralim's firmware, the TS80 uses 12V from the wall charger and 9V from the power bank. The stock 1.23 firmware of TS80P shows 9V and 7.7V respectively.
- The TS80P US wall charger packaged with TS80P works ok. Banggood even attached a EU/US plug adapter (without even asking for it). I don't know if this is because of the plug adapter or the charger itself, but when plugging it into a wall socket (230V) it causes quite a spark in the socket. Doesn't happen with most of my other chargers and equipment. I've already ordered another 45W PD wall charger with EU plug instead.

mik-at, 2020-06-23:
> * I've been able to power TS80P with an original 45W PSU for my Dell XPS 13, but the voltage doesn't go higher than 9V - probably because the PSU supports 5, 9, 15 and 20V only, so the iron [...]

Yes, it seems that the stock firmware does only some protocols on QC/PD.
As there are a couple different handshake-protocols for handling voltage and power, I think you just have to get the best combination. For example my powerpack (random aliexpress) does only provide 9V for the TS80P on the USB-C port, but 12V on the USB-A PD port, although 12V on USB-C would be possible.

whitehoose, 2020-06-23:
The wall wart Banggood supply is fine .... they bundle it so it's guaranteed to work, the problem is plugging it in to a non chinese supply.
I'd guess the adaptor you've been given is the standard adaptor everyone bundles with everything.  The front accommodates every plug type in the universe the rear is area dependant - they simply screw on an appropriate set of prongs suitable for your country.

Its a good idea - but the front and back rely on the pressure of plugging something in to make contact.  You'd (and anyone else on the ts80P train) intending to use the included adaptor would be wise to test the continuity of each pin - I usually lift the back flap of live and neutral to ensure a bit of contact pressure. and I've used my original for about 18 months without drama.

Another option is to open the wallwart and solder a lead onto the PCB connection and fit a decent plug.

The main criticism is the earth continuity (or lack of) but so many things these days are double insulated and come with either a 2 pin euro or US connector or a couple of 100 mm of twin in reality you don't need earth except in the UK to open the socket's gate

[image]

yawor, 2020-06-23:
@whitehoose it's even simpler, with only US socket on one side

[image]

I've opened it and pushed out the contact wings a little to have better contact. The electrical connection seems to be good, but the charger itself slide out quite easily, without much force. Also to be honest, I'm a little scared of the US type plugs, especially when they are connected to 230V. It's easy to expose live connection when the plug slide out of the adapter of this type a little.

whitehoose, 2020-06-23:
Gawsh!  That wouldn't work in the UK but it looks well made for 2 pin - and country or region specific too.  Our home electrics are very safe but foreign adaptors for it are death traps.

Ralim, 2020-06-25:
Can confirm it looks like the stock firmware only does P2.0 for 9/12V. Doesnt seem overly smart.
Trying to get a working implementation of 9 & 12V for PD 2.0/3.0 including PPS (lets you do discrete voltages).

Still working on debugging it and suffering from the janky setup i need to break out SWD from the USB-C.

Have got handle temp working I _think_ so may have a non-pd test firmware sooon (work is busy atm as well).

skoehler, 2020-06-25:
I'm looking at purchasing a T80P. According to what I can find, standard USB PD voltages are 5V, 9V, 15V and 20V. However, the TS80P runs at 12V. Most of the chargers I find also just specify the max. amperage for 9V and 15V. Typically, there is no 12V listed in the charger spec. I don't trust their charger (but the cheaper kit without the charger doesn't include the USB silicone cable, duh!).

So is the TS80 really limited to 12V? And will it fall back to 9V if the charger doesn't support 12V?

yawor, 2020-06-25:
@skoehler the stock firmware negotiates highest possible voltage up available not higher than 12v. Like I've mentioned in one of my previous posts, I've been able to power TS80P from an original USB-C power supply for my Dell XPS 13, which doesn't support 12v, but iron has been able to negotiate 9v.

skoehler, 2020-06-25:
I'm just confused why they didn't support 15V or even 20V. Maybe the tips can't handle that much power, but they could have limited the power to the tip by limiting the PWM duty cycle. I am hesitating to buy one because of this weird compatibility issue which could have been avoided.

gutschke, 2020-06-25:
I believe the tip has a resistance around 4.5Ω. If permanently connected to 9V, that'll draw 2A, which chargers generally can do. At 12V, it would draw just over 2.5A (depends on variation in actual resistance). That can actually result in some chargers cutting off power for over-current. And at 15V, it would draw more than 3A. but that's outside of the spec for USB-C. Things only get worse when going up to 20V, where the iron would draw about 4.5. Funnily enough though, for some power supplies, that's again within spec (some devices support higher currents in combination with higher voltage).

With PWM you can bring down average current, but you'd still need to filter the signal so that peak current doesn't exceed 3A. I am not sure how good the hardware is about doing this filtering. The higher the PWM frequency the less filtering is needed.

In practice, this could mean that some power supplies are perfectly happy with powering the iron at 15V, even if things are slightly outside of the official specs, whereas other power supplies won't even work at 12V. And this could also vary from one tip to another, as the difference in resistance might just be enough to push the power supply into over current. It could also depend on whether the soldering iron used PD or QC to negotiate power output, as they have different current limits at the same voltage.

yawor, 2020-06-25:
I've received today a BlitzWolf 45W PD charger. According to information printed on the brick it supports:
- 5V/3A
- 9V/3A
- 12V/3A
- 15V/3A
- 20V/2.25A

This is almost the same as the charger bundled with TS80P - this one also has 18V/2.5A.

I also have 4 different tips:
- TS-B02 (received with TS80P)
- TS-D25 (received with TS80)
- TS-BC02
- TS-J02

I've noticed a different behaviour depending on which tip is plugged in and which charger is used. Below table shows following info: Power displayed in settings, voltage displayed in settings, Power displayed when heating, Voltage displayed when heating

The power displayed in settings can't be changed, it seems to be calculated from negotiated voltage/current right after connecting power. I've reconnected power each time I've replaced the tip.

tip | Bundled charger | BlitzWolf
-- | -- | --
TS-B02 | 33W, 12.1V, 29.8W, 11.7V  | 31W, 11.7V, 28W, 11.4V
TS-D25 | 22W, 9.12V, 19.2W, 8.8V | 20W, 8.85V, 18.5W, 8.6V
TS-BC02 | 28W, 12.1V, 25.8W, 11.8V | 26W, 11.7V, 23.6W, 11.5V
TS-J02 | 31W, 12.1V, 28W, 11.7V | 30W, 11.7V, 26.4W, 11.4V

It seems that the bundled charger is quite good actually. BlitzWolf is rather good charger, but it's a little behind. What's interesting is that the TS-D25 tip is not able to be powered with 12V on TS80P. I've even tried powering the iron with one of the other tips first and then changing to TS-D25 while plugged in. It still shows ~12V in settings, but it drops to 9V when I start heating and stays at that level after that.

I hope this is going to be helpful in some way. The TS-D25 was my favourite tip before I've ordered TS-BC02, which I'm using now most of the time. It would be great to have all of them working on 12V on TS80P. It worked without any issues on TS80 with @Ralim fw.

PedroLamas, 2020-06-26:
I just received a TS80P I ordered a few days ago! It came without a charger so I ordered [this one](https://smile.amazon.co.uk/gp/product/B0829P294W/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) from Amazon (to be delivered later today)!

The specs on it say it supports PD 12V/3A, so once I get it and manage to run a few tests I'll report back my findings.

Ralim, 2020-06-30:
@gutschke  and @yawor are on the money.
I do intend to just "enable" 12V ignoring tip resistance at least initially.
While their smart system is neat to detect resistance and therefore derive the voltage; it chews up flash space and isn't massively helpful.

Goal is to support 9V + 12V. Can confirm that the peak current of 15V will trip some chargers and will also kill the TS80(P) eventually.

If I get PPS (PD3.0) working then that allows 100mV control of the output so should be possible to have a mode to allow for tweaking voltage up or down a bit too.

Zeescez, 2020-07-08:
I tried to load "login.bmp" several times on my ts80p.  Now there is no free space on the soldering iron - 0 bytes free.  I tried to reflash (with current version v1.23)- it didn't help.  How to fix this problem?

skoehler, 2020-07-08:
> I tried to load "logoin.bmp" several times on my ts80p. Now there is no free space on the soldering iron - 0 bytes free. I tried to reflash (with current version v1.23)- it didn't help. How to fix this problem?

It could be that the filesystem is corrupt. Have you checked it?

Zeescez, 2020-07-09:
> It could be that the filesystem is corrupt. Have you checked it?
chkdsk did not find a problem. 8 kB - total, 1 kB - in 2 hidden files, 7 kB - in 5 files.

Edit: while typing this message, I tried to format volume, windows returned error. But also when i reconnect cable, it suggested me to fix errors. now I have 8 kB - total, 1 kB - in 2 hidden files, 1,5 kB - in 5 files, 5,5 kB - free.

Ralim, 2020-07-22:
Status update is that basic PD2/3 is working (not PPS) for SRC devices (not try.SRC like powerbanks - WiP).
So about half of power banks will work and most laptop chargers from what I have tested so far.
Need to shrink code size somehow in a few spots to get it to fit nicely and then will put up a _very_ beta build.

Code needs a chunk of cleanup in general too I think 😅 
But progress _is_ being made, just PD is a pain to debug 😁

Ralim, 2020-07-23:
As promised, here is the _very alpha_ release.
This is probably going to be a bit hit and miss on PD support, I need to setup a better debugging setup and continue testing.
[TS80P_Alpha.zip](https://github.com/Ralim/ts100/files/4965933/TS80P_Alpha.zip)

This _should_ work otherwise, however naturally I'm expecting a fair share of bugs.

discip, 2020-07-23:
@Ralim
First and foremost **THANK YOU** for the effort you put into this! 👍

And ... as you might have expected **_the first_** _this does not work_ and _that does not work_ calls are in the line:

topic | status
-|-
Sleep timeout | only ***Off*** is selectable
Shutdown timeout | only ***Off*** is selectable
motion sensor | no sign of life

I do not know for sure, because I have no equipment for measuring this, but the iron negotiates with the [stock charger](https://user-images.githubusercontent.com/53649486/83251177-86b86d80-a1a9-11ea-9e95-1c7719bd639a.jpg) for *'only'* ***10.3V*** ~ ***21.9W*** is that the expected behavior *(I hope it is a hit!)*?

ps: One a bit of topic issue I ran into recently:
I bought [this charger](https://www.amazon.de/gp/product/B082KVZX85/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1) and I have to admit, that this seems to be ***a decent device***, judged by its looks.
Unfortunately it **sparks** every single time you plug it into the wall socket. Except for the situation, when you plug it out and in again within the next *approx. 5 seconds*.
Are you familiar with such issues?
Do you know an easy solution for this? Solder something *(capacitor?)* in between the prongs?

thank you in advance
kind regards

PedroLamas, 2020-07-23:
@discip I have the same charger (though my has a UK plug) and I too like the results with it, but I have not noticed any sparking issues with mine!

Could you have a faulty one?

akselii, 2020-07-23:
Thank you for the firmware!
Im getting the same bugs as discip, also on Advanced Options after Pulse Power is an selection which just says "ok" 2 times, and when you select it you can change some setting from 21.0 up or down.

discip, 2020-07-23:
@PedroLamas
I do not know, if this is a faulty one, but I read, that someone else had the same issue, so I assume, that it is built like this.
[image] [image]

Could it be, that it better plays along with the US voltage?

PedroLamas, 2020-07-23:
@discip I have no idea, but here in the UK though the plugs are quite different (they have a tighter fit), the voltage is the same as in mainland EU! 🙂

Ralim, 2020-07-23:
Aha,
Looks like I have two issues to chase, one being that I must have used an out of date translations file (broken menu) and also that I need to double check calibration of the input voltage reading.  Miniware shuffled resistor values around a bit so might require a tad more tuning.

Will get back to you with more information sooon :)

whitehoose, 2020-07-24:
You have a couple of possibilities - lots of sparks either capacitor or inductor - unfortunately you have both in the input of a switched PSU.

if you ignore the sparks - does it work? 

If you unplug and immediately apply a short to live and neutral pins on the PSU using a screwdriver with a well insulated handle (carefully!!!) do you get a spark?

There should be a discharge resistor and a smoothing capacitor on the primary winding to stop this - if either has died that would do it 
or
It could be something deeper in  the plug shorting the secondary.

In all cases the only way is to open it up and look for scorch marks which will invalidate any guarantee - and if you spot it - they are always a pig to fix.
Assuming its new - send send it back immediately cause whatever it is - that degree of sparking isn't good and you could kill other delicate electronics with the spikes it is causing.

If you just can't help yourself open it up but usually they make it impossible to do that without using a dremmel and a hammer - so don't expect a good outcome other than knowing why

AEFeinstein, 2020-07-25:
I am the proud owner of a new TS80P, and just flashed the `TS80P_Alpha.zip` firmware. I'll report anything I find, but also let me know if there's anything specific to test.

nikscha, 2020-07-26:
Just flashed the Alpha-firmware, the iron now    "boot-loops":
The miniware logo is shown briefly, then the screen turns off, then the logo is shown again. Sometimes it reaches the main menu, just to turn off again.

Ralim, 2020-07-26:
@nikscha 

Don't panic you should be totally fine to go back to the stock firmware.

Could you tell me what power source you were using for that?

Currently still debugging this issue and trying to figure out what exactly makes the fusb302 act up.

nikscha, 2020-07-26:
I tested the "stock" supply, my laptops usb-c pd charger and my laptops usb-c port. All 3 chargers yield the same behavior.

After trying (didn't work) to flash the original TS80 firmware (couldn't find TS80P, could you help me out?), I flashed your aplha again, this time the "boot loop" is slightly different, see [this](https://photos.app.goo.gl/cQz6kkFwcQp5fkQm6) video. One thing to note as well is that the iron also crashes in DFU mode (also shown in video), holding the lower button for quite a while eventually lead to it staying in DFU mode.

skoehler, 2020-07-26:
Ts80p firmware: 
http://www.minidso.com/forum.php?mod=viewthread&tid=4085

iamLEEROYJENKINS, 2020-07-27:
Thank you Ralim for the ALPHA firmware, and your support for our TS80Ps.   I have a  RAVPower PD 20000 mAh battery pack.  

The TS80P will boot and be ready to go on the USBa iSmart port, but not on the PD USBc port with the AlphaFW.  

I have the TS80p set to 12v  on the AlphaFW.   Plugging into the PD USBc port the TS80P will boot loop.  Using a USBa to USBc cable on the iSmart(USBa port) the TS80P will boot and work running AlphaFW.  

I then flashed the stock v1.23 FW and tried it in both ports and it boots and heats up fine.  

I am using a RAVPower 20000mAh 60W 2 Port power bank.  Ouout 
[image]

NDR008, 2020-07-28:
Hello everyone.
I'm jumping in - dwelled forever on TS80P vs TS100.
Finally went with the TS100 - but PSU-less.
What's the current goal here?
As I understood - stock FW max power is 18W: 9V@2A on QC3, and 30W: 12V@3A (maybe actually 2.5?) on PD 2.0.

Is the goal to get 12V working on QC3 mode too?

Ralim, 2020-07-28:
Hello,
Next updated test firmware.

Changes:

* Bunch of testing of the USB-PD side of negotiation
* -> Should work with more devices, but curious to know models of anything that doesnt work
* -> A non working device will look like a boot loop. Dont panic!
* Should have fixed the input voltage reading calibration
* Should have fixed menus now

If you can please do a settings reset in the advanced menu would be ideal (to make sure you are using updated default calibrations).

### Bootloop

If you get stuck in what looks like a bootloop on some devices and want to jump back to stock firmware you can either:

* Hold down front button and it should stop looping after a while
-OR-
* Use a usb A -> C cable which bypasses all PD and should work totally fine.

### Things to test

* General iron use
* Does PD work for you
* If you have QC devices; does QC work for you
* If you have means to check calibration of tip temp, that would be interesting to know
* Do all menus work for you

[TS80P_Beta.zip](https://github.com/Ralim/ts100/files/4988242/TS80P_Beta.zip)

akselii, 2020-07-28:
Quick test with the stock PD charger that came with the soldering iron seems to show correct volts. 
Selecting 9v or 12V from the settings does not work on the stock charger.
On my QC charger selecting between 9v or 12v does work, and seems to work as intended.
Power limit works on both chargers, at least it looks like it does. My ruideng meter did show the wattage not being very stable but i think it would average at about the 10w that i tested.
Tip temperature still 0 at room temp, i think i did see it flash room temp for a split second on first boot. I did a factory reset.

[242 more replies]