Ralim/IronOS #631: Temperature reading not stable

BNWEIN, 2020-05-22:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report

* **Do you want to request a *feature* or report a *bug*?**

Big

* **What is the current behavior?**
The temperature reading fluctuates by anywhere between 10-15 degrees

* **What is the expected behavior?**
For it to stay steady

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Turn on my iron
2. Set temperature 
3. Heats up to around the temp set but fluctuates between around 15 degrees on the screen 

***Video of problem if hard to reproduce***

Can send a video if need be 

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.09
  - PCB Version: unsure 
  - Power Supply 24v supplied power adapter

* **Other information**

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

ranomier, 2020-05-24:
New user. Just flashed mine. Same issue

Ralim, 2020-05-25:
At the moment this firmware is optimised towards power users who don't like being lied too, so it shows the current tip temperature as it changes. A small amount of oscillation in the PID shows up depending on your tip and power supply (high voltages + small tips being worst). 

This is sorta of a trade off between response time (i.e time from you putting tip on cold surface, and it dumping full power) and how stable it is. The current tuning is just slightly unstable on some tips to keep the device optimised for best performance.

At one point there was the hiding function that you can see on some other irons, where they hide a small deviation on the user interface (even though the tip still does it), but that was not as well liked by the community.

The main issue is that tuning for a wide input power range and various tip thermal weights makes tuning out the small oscillations hard. Especially when you consider that the outside of the tip usually wont see these tiny changes (as the thermal mass will filter them out).

TL;DR I don't have an overly good answer for this at this point :|

whitehoose, 2020-05-25:
On top if the software issues, you have the purely mechanical "wobble" introduced by the tip's components.  Think of a mechanical bi-metal strip - it works to tolerances - heats up to just above - contacts part at maybe +2c switches off till just below, say -2c repeat as required. - this introduces a 5c wobble when its working correctly.  As you sacrifice mass for smaller size the time for heat up / cool down cycle speeds up too.  If the tip was a 1Kg slug you woldn't notice any fluctuations. 

We used to adjust thermal controls manually - if you "nip" the tolerances too tight they began to sound like a swarm of bees as the heat up / cool down cycle speeded up and worse, at higher frequencies the inductance of the coil produces a spike that was like a car's sparkplug, introducing stress into the circuit.  While the modern equivalent doesn't have moving parts, the principle is the same, there needs to be separation to prevent runaway and self destruction.  You can't stop physics happening - but you can hide it!  Exact and immediate temperature measurement isn't always in the best interest of the job, nor is it vital for melting lead.

troisiemetype, 2021-03-07:
I was also surprised by this behavior. I can understand the need to have reactivity internally, but it causes a flicker that is distracting. Woulnd't it be possible to add a filter just for display ?

An exponential filtering would probably do the trick here : use 10-20% of the current reading, and average it with 90-80% of the previous reading. I use it a lot to filter some things that fluctuates a lot, like capacitive touches, it as low processing needs (even less if using fix-point-math) and low memory need as well, with just a second global variable to store the last reading.

discip, 2022-04-23:
@BNWEIN
As @Ralim mentioned above the latest release is far more stable, so please close this issue. 😊

thanks

closed 2022-04-23 by Ralim