Ralim/IronOS #635: No readme/instructions for autostart menu options

Wegerich, 2020-05-29:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [x ] Feature request
  - [ ] Translation

* **What is the current behavior?**
The autostart menu displays F T S 0. I cannot find any guidance about what each of these options is. 

* **What is the expected behavior?**
The menu options readme page should describe what each of the options is. 

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.09 Git 2020-04-04
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) : 19.5V 5A

Ralim, 2020-05-29:
The FTS0 options are defined by the language as well as the readme.
**So yes, definitely need better docs.**

If you have an option you are unsure of, if you pause a description should scroll past that includes a description of what all the options are (Since the actual letters used can change depending on the language)

Wegerich, 2020-05-29:
Aha, I didn't see the scrolling menu. That's slow though ;)

My menu includes "0" but my scroll information doesn't mention that. 
So, 

F=Auto-on disabled
S=Turn into sleep temperature when power is switched on
T=Rise to soldering temperature when power is turned on

discip, 2020-05-29:
Good morning,
it is actually the letter `O` not the value `0`
**F** = *Off*
**T** = *Soldering*
**S** = *Sleep*
**O** = *Sleep at room temperature*

regards

closed 2020-09-14 by Ralim