Ralim/IronOS #639: Repository name

Ralim, 2020-05-30:
Currently thinking that adding support to this code base for other soldering irons in the future feels highly likely to occur. (#634)

As well as ts100 being a bit odd since the TS80 is also supported now as well.

So, opening the floor for anyone's ideas of what to rename the repo to.
_OR_
Do i leave the name as is and just make things more clear in the readme?

Open to ideas 🤔 

Note that if I do change the name GitHub _will_ redirect all old links etc so they shouldn't break.

discip, 2020-05-30:
One of the many suggestions to follow:

### *smart soldering pen*

I am no expert in this field, but I think it would fit the purpose, wouldn't it? 😏

regards

JohnEdwa, 2020-05-30:
Hmm, you can make it be descriptive with a workable abbreviation/acronym, like uh... Custom Iron Firmware Ecosystem (CIFE). It's also usually just called the"TS100 Ralim Custom Firmware" or some mix of those words, so something as simple as RalimCFW could work too. Or you can name it with a completely irrelevant name, like Marlin, Klipper or Cura that you just happen to like, if the idea of branding an open-source project with your name feels too narcissistic.

Another that came to mind is [GadgetBridge](https://github.com/Freeyourgadget/Gadgetbridge), which is a unified app to handle wearable devices. From that I got SolderBridge. Maybe SolderBridgeCFW to make it slightly clearer, I 'unno.

federck, 2020-06-08:
For the reasons you've explained I think that changing name to the repository would make sense.
After some struggle thinking about meaningful acronyms I came up with UNIRE (UNiversal IRon Ecosystem) - "to join" in Latin referring whether at soldering or putting together a community.
Whatever the result, I can also possibly think of a logo.

Regards

discip, 2020-06-08:
@federck
This is a very interesting approach in many ways! 😲🤯
I like it very much! 👍

AntonMadness, 2020-08-02:
I just got my TS80p in and I was confused finding TS80 info in the TS100 repo. So as a newcommer I'd say it's deviniatly good to change the name to minimize confusion.

I like the UNIRE and SolderBridge idea. But is this repo support really going that wide as those names suggest? Ie. will it also support other solder irons like the Pine's one (https://www.pine64.org/2020/07/15/july-updatepmos-ce-pre-orders-and-new-pinephone-version/). If not, maybe a name as simple as TSxxx will suffice.
just my $0.02.

Ralim, 2020-08-18:
@AntonMadness  Indeed it is this repo thats most likely the firmware for Pinecil at this point.

There will be more news on that front _soonish_

@federck I am leaning towards `unire` as it is both fairly unique and also not something boring :D 

Sort of leaning away from using `Ralim` in the name as it does indeed feel a bit... bold :grin: 
Would rather a "irrelevant" name as you mentioned, and can make the tagline (description?) on github descriptive so it _should_ still come up in google searches :)

federicodilo, 2020-08-19:
Hello, @federck here, I had a problem with my previous account.
Since my proposal has been moderately successful (glad to her that 😃), I also thougth to a simple logo in order to make it work both as a logo for the description page and, eventually, as a bootscreen for irons.
More complex logos wouldn't be suitable as bootscreens, in my opinion.

[image]

What do you think about it?

ItsMeAubey, 2020-08-31:
> UNIRE 

Unfortunately, due to how the brain processes words that it already knows, this doesn't really work... Pretty much every single person is going to read this as "urine"...

sreichholf, 2020-08-31:
How about

- Iron (or)
- IronOS

JohnEdwa, 2020-08-31:
Stylizing it, such as using UnIrE or UNire or UnIRE etc would fix the misreading problem.

Though there's also the issue that googling UNIRE will result in finding "Unione Nazionale Incremento Razze Equine", the Italian National Horse Breeding Union :)

Ralim, 2020-08-31:
Another late night thought I had was to use "Ferrocerium" inspired by @sreichholf :thinking:

JohnEdwa, 2020-09-01:
There are also other tin+lead mixtures instead of solder, namely [Pewter ](https://en.wikipedia.org/wiki/Pewter) and [Terne](https://en.wikipedia.org/wiki/Terne). 
Ferroterne. Ternecerium. Pewterne. Ferrone. Ferroter. Pewcerium. Ferter. Ternerium.

Portmanteaus are fun.

discip, 2020-09-01:
If I had to choose, I would probably go with **UnIrE** like @federck & @JohnEdwa suggested.

And the logo could look something like this:
[image]

Maybe someone has a better taste for *font* (something more 'tech-ish') or another *image* for the iron.

By the way: It is ***shorter*** and thus ***easier to remember*** (Maybe due to the very fact @ItsMeAubey mentioned.) compared to *"Ternecerium"* and the like.

ItsMeAubey, 2020-09-01:
As soon as you go from Unire to UnIrE the issue vanishes.

doegox, 2020-09-01:
Thought about **FreeSolder** in echo to Free Software :)

ItsMeAubey, 2020-09-01:
FreeSolder sounds nice. Acronyms work but something with actual meaning (without knowing what the acronym means) would also be good.

ItsMeAubey, 2020-09-01:
OpenSolder? OpenIron?

Ralim, 2020-11-01:
Alrighty, going to stir this up again 🤣 

So the list looks like:

1. Unire / UNIRE 👍 
2. IronOS 😄 
3. Ferrocerium 🎉 
4. Ternerium ❤️ 
5. FreeSolder 🚀 
6. OpenSolder 👀 

To get a feel for shortening this list, I've added an emoji next to these, please react we emoji's as appropriate to vote for the ones you like :)

Ralim, 2020-11-04:
Welp looks like IronOS is a far bigger winner than all the rest :) 
It also has very little similar stuff in Google so that is also a :+1: 

Any complaints against this being the new option perhaps?

GreekGeorge, 2020-11-04:
H.E.A.T.S.
Handheld
Electronics
Algorithm
for the
TS-100 (and clones)
Soldering Iron

tigerclaw989, 2020-11-04:
IronOS sounds very good. I like opensolder but it sounds like it’s an open solder joint.

Urus74, 2020-12-05:
Hello Ralim, thanks for the useful tips!
I have received today my ts100 and tried to upgrade the firmware to the latest one. First I did it by Chinease, but found none on there miniware.com.cn. The solution I thought I found among your releases to the issue. The file had only 61 KB — as if an incomplete data (since I saw a German tutorial on this issue and there were 91 KB on the TS100_DE hex). Anyhow when I dragged it to the disk (after connecting the ts100 to the laptop) I always got the err. and not the rdy suffix. Could you, please, tell me where can I find the latest firmware for the TS100 on your page?
Best regards
Urus

Ralim, 2020-12-05:
> Hello Ralim, thanks for the useful tips!
> I have received today my ts100 and tried to upgrade the firmware to the latest one. First I did it by Chinease, but found none on there miniware.com.cn. The [...]

Hi @Urus74 
Its better to open a new issue than commenting in this one.
As stated in the readme on the main page, they are all at [Releases](https://github.com/Ralim/ts100/releases).
The file not going RDY is a known issue that is not actually related to the firmware. There are _lots_ of issues raised about it, if you have a look through the search you should find them :)

closed 2021-01-05 by Ralim

Urus74, 2021-01-05:
Hi Ralim,
thanks for the answer. The problem has been solved. I had to unpack the zip file and after that its insertion into ts100 worked. 
Best regards