Ralim/IronOS #640: Accelorometer for 2.10 does not work

Gregory1981, 2020-05-30:
Hi.
I am submitting a Bug report.
To reset sleep timeout, accelerometer does not work (need press either button to reset it).
Firmware version 2.10
Power supply: powerbank 3.0 (PD 12V).
On previous firmware 2.09 it did work.
On 2.10 is not working.
For sure I have put back 2.09 and accelerometer does work again.
When I have put 2.10 it is not working.
Becuase this issue, I can not use 2.10 and need go back to 2.09
Thanks for your effort guys.
Greg.

Ralim, 2020-05-30:
Thank you for reporting this soo quickly. Working on a fix now.

Ralim, 2020-05-30:
This should now be fixed: https://github.com/Ralim/ts100/releases/tag/v2.10.1

Gregory1981, 2020-05-30:
Works like a charm now.
Thanks Ben for lightspeed reaction.
I think we can close this issue now.
Thanks.

closed 2020-05-30 by Gregory1981