Ralim/IronOS #641: slight GUI imperfections (scrollbar, clipped text)

discip, 2020-05-30:
* **I'm submitting a ...**
  - [x] Bug report

* **Do you want to request a *feature* or report a *bug*?**
1. Since I do not know the *science* behind the scrolling feature, I do not know if this is a bug or not, but it does certainly (imho) not look right.
2. Layout conflict between text and values.

* **What is the current behavior?**
1. If the last entry in the `Advanced options` menu is reached, the scrollbar is not at the very bottom of the screen.
2. In the `Soldering settings` the **'e'** of the text of `Temp change`, both `long` and `short`, is clipped of, or rather overlayed by the space of the corresponding values *imho*.

* **What is the expected behavior?**
1. The scrollbar reaches the bottom of the screen, when the last entry in a menu is reached.
2. The full text of a menu entry is shown, without being clipped off.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
1. Go to the `Advanced options` menu and whilst doing so, watch the scrollbar especially as you reach it and keep that last position, you just saw, in mind, after entering it, watch the scrollbar again (now somewhat shorter) descending towards the bottom of the screen, but you will realize that it will never reach it.
2. Just navigate to the corresponding entry and you will see for yourself.

* **What is the motivation / use case for changing the behavior?**
My aim is, to improve this, already beautiful, work.

* **What are you running:**
  - **TS80**
  - Firmware Version: **v2.10.1**
  - PCB Version: **2**

Ralim, 2020-06-14:
I think the scrolling thing may be a bug or may be rounding; but I will dig into this deeper when I have time 😓 

The clipping is def a bug, and may be solvable easily by just re-ordering screen drawing :)

discip, 2020-06-14:
@Ralim
Thank you for the promising information. 😁
And as always, I am looking forward to this.

paulfertser, 2020-06-20:
@discip, please test #659 #660 . I saw this issue on my own and sent a fix before I found your report, so it's not mentioned in the commit message yet.

paulfertser, 2020-06-21:
Looks like all the issues raised here are fixed in master now.

discip, 2020-06-22:
@paulfertser
Thank you for looking into this and most likely fixing it. Since I am not able to compile the latest commit and therefore I am not able to test it, I have to rely on your word, that this is fixed now. Could you please confirm, this?
Have you tested it on a **TS80**?
@Ralim 
If this is save to close, I would like to do so, or you could do it as well.

kind regards

paulfertser, 2020-06-22:
@discip my code was supposed to do just two things: 1. Show @Ralim my approach to this problem so that he can judge if it's the right direction; 2. Test (on just two parameters) User Experience of my solution. I tested it on TS100 (I do not have TS80) and for me it seemed adequate, but knowing that I'm biased (as I just coded it I knew what to expect) and also in general my UI preferences are uncommon I feel like someone else (and especially the original issue reported) should actually test that. And I implemented just half of your suggestion (the pause) as I have no idea how to add some kind of visual indicator on such a tiny screen. Why are you not able to compile, do you not have access to a computer under your full control?

paulfertser, 2020-06-22:
Oh heh, I guess I mixed two issues up, sorry. My comment was all about the "sudden transition from the last allowed value to the first" #661. Sorry. The two issues you've brought up should have been fixed (assuming @Ralim tested the code prior to merging). At least on my TS100 with those two changes it all seems proper.

paulfertser, 2020-06-22:
BTW, you do not need to self-compile to test anything. Each "github pull requests" automatically starts a build job. So you can go to the specific page, press "Checks" (e.g. https://github.com/Ralim/ts100/pull/660/checks ), press Artifacts on the right-hand side and download ``compiled.zip`` with all the firmware variants for all target devices.

discip, 2020-06-23:
@paulfertser
Thank you,
so I am going to test the latest commit tomorrow and if it is going to work, I will close this issue.
I was searching for an easy way to get the latest commit, but was not aware of the possibility, you posted. 

THANK YOU 👍

paulfertser, 2020-06-23:
Latest commit to master is always accessible by using this link: https://github.com/Ralim/ts100/commit/HEAD , then you press on the green tick to the left of the commit title, a dropdown appears with ``C/C++ CI / build (push)``, press ``Details`` to the right of it, ``Artifacts``, ``compiled``.

discip, 2020-06-23:
@paulfertser
Thank you!

discip, 2020-06-23:
@paulfertser
That works flawlessly sir. 👍👏
thanks

closed 2020-06-23 by discip