Ralim/IronOS #642: unify behavior of certain menu entries

discip, 2020-05-30:
* **I'm submitting a ...**
 - [x] suggestion

* **Do you want to request a *feature* or report a *bug*?**
I suggest to unify the appearance / handling of certain menu entries.

* **Current behavior  vs  expected behavior?**
In the **`Advanced options`** menu there are at least 2 different solutions for mainly the same thing.
The 'thing' is on one hand to ***activate the feature*** and on the other to ***set*** the corresponding ***value***.

In particular the following 2 entries treat almost the same thing differently:
## 1. Power limit
  * **`Enable power limit [x]`**
  * **`Power limit [value]`**
These 2 could be combined into one entry, just like for example: **`Shutdown timeout`** *(Off, 1M ...)*

## 2. Power pulse
  * **`Power pulse W [[ ],value]`**
Although this one is kind of combined, but not as sleek as for example: **`Shutdown timeout`** *(Off, 1M ...)*
This out of place empty box just 'ruins' it.

* **What is the motivation / use case for changing the behavior?**
This one is as well motivated by the wish to contribute to an even better look and feel of this beautiful piece of software.

* **What are you running:**
  - **TS80**
  - Firmware Version: **v2.10.1**
  - PCB Version: **2**

Ralim, 2020-06-14:
These are all very logical.
I'm currently focusing on the TS80P but after its up to basic functionality these are probably next on the list :)

discip, 2020-06-14:
@Ralim 
Wow that's great news!
If I had to choose, I would definitely go with the **TS80P**.
But I am also curious about the above mentioned menu adjustments, since it would also affect the **TS80P**. 😊

Ralim, 2020-06-15:
Yeah my TS80P did arrive, but my quick (lol - multi day) port of existing code didnt work first go so still waiting for some breakout boards for usb-C to arrive so I can decode whats actually happening on the PD bus.

discip, 2020-06-15:
Ah you have already invested effort into *pre-building* code, that is definitive laudable!
I hope you do not have to rewrite everything! 😅
I cant wait to see the first release, but I appreciate it, if you take your time, since nobody wants the **developers device** being  *bricked*.

Ralim, 2020-09-06:
So what I have done so far is to unify everything to combine the setpoint with the value.

I'm not certain if I want to do the miniware style of entering and exiting setup mode as that feels very _slow_ to do.. But is that because im working through the menu a lot ?

discip, 2020-09-08:
@Ralim
**Thank you** for considering this suggestion. 👍
In my opinion, the *demand for consistency* is covered! 😊

ps: I like the blinking scrollbar when the last entry is reached and the logical behavior of neither pausing nor blinking when the list consists of only 2 entries.

closed 2020-09-08 by discip

Ralim, 2020-09-09:
Glad its a bit more consistent at least :)

Thank @paulfertser for the sweet menu pause and scrollbar work :)

discip, 2020-09-09:
@paulfertser
Thank you!