Ralim/IronOS #647: This will solve problems when moving hex-files to the TS100 from a Mac

piadam, 2020-06-03:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [x] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Enhancement

* **What is the current behavior?**
Upload-errors

* **What is the expected behavior?**
Smooth upload of hex-file to the TS100

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Change the names from e.g. TS100_NL.hex into (UPPERcase-only!) TS100_NL.HEX
2. Drag the HEX-file to the TS100
3. TS100_NL.RDY is the result

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**
I have been struggling for maybe half an hour to get the newest release uploaded.
It kept on failing until I changed the file-suffix into uppercase

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.10
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information**

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2020-06-04:
Oh, this is interesting.
Can definitely add to flashing notes.
Will change makefile to uppercase filenames too.

pocketscience, 2020-06-04:
I didn't encounter this issue today on my Mac - ie I didn't rename the file and it installed just fine... MacBook Pro (2018), macOS Mojave 10.14.5.

piadam, 2020-06-04:
I must admit that on my MacBook Pro running macOS 10.15.5 it appears to do well, even with the lowercase suffix. But on my El Capitan 10.11.6 MacBook Pro failure-rate is about 50%.
Funny since the update from 2.09 to 2.10 never ever succeeded on El Capitan without changing the suffix to UPPERcase.
Mind that the linux-install-script to update the TS100, which inspired me, in my opinion also starts with converting lowercase- to uppercase characters.

agatti, 2020-06-16:
For what's worth I see at least a 50% failure rate on 10.15.5 regardless of the filename case.  From my limited testing it depends on which method you use to copy the file onto the device, ie. dragging and dropping from Finder almost never succeeds (for me), whilst copying the file from Terminal/iTerm2 via `cp` works every other time.  It may be a timing issue somewhere, who knows.

piadam, 2020-06-17:
Good to hear, agatti!
Since I really am a command-line-guy this really sounds like excellent news to me!

valhallaVenus, 2020-09-03:
I'm on a MBP running Mojave 10.14.6 and for some reason my TS-100 isn't even recognized as being connected into the computer. The TS-100 gets power and shows DFU 3.45 but there's no folders or drives in Finder that I can located. Anybody else have that problem?

piadam, 2020-09-03:
Dear valhallaVenus,
It is my opinion that it should work as long as you use the right cable. I am NOT familiar with micro-USB-cables (why should I, I'm a Mac guy), but my experience is that there are power+data cables as well as power-only cables. Please make sure the cable you use is one that supports data-transfers too.
I never had any problems under El Capitan up to Big Sur.
Good luck!

piadam, 2020-09-03:
Don't what happened... I just tried to send you my info just once but the page did not show any changes so I pressed the comment-button (probably) a few times too many..

piadam, 2020-09-03:
Sorry I did not mean to CLOSE the chat, must have been pressing a wrong button.

discip, 2022-04-23:
@piadam
Please close this, since this no issue. 😊
Maybe you could add this information to the documentation section, I can't remember that this was done yet.

thanks

piadam, 2022-04-23:
It has been long since I moved my last hex-file from my MacBook Pro to the TS100. Cannot recall if there stil is a problem. Closing this item now. Thanks for your support! Still using the TS100 but updating it from a Raspberry Pi now.

closed 2022-04-23 by piadam

simeonpashley, 2022-05-25:
Apologies for necro'ing this thread, hoping to help future Googlers.

I came to update a new SEQURE SQ-001 today with the TS100 firmware.

Copying via Finder failed repeatedly with the suffix `.NO` rather than `.ERR`

I changed the suffix case from `.hex` to `.HEX` and finder failed as previously.

Using Terminal, I used the command line `cp TS100_EN.HEX /Volumes/Firmware/` and it worked first time, returning the suffix `.OK` 🚀