Ralim/IronOS #649: I burnt my TS100

chuckyto, 2020-06-08:
You will find photos of my mess.

I decided to use my TS100 with a power supply I had around (thought it was a 12Volt, my fault) and it resulted to be a 48 volt power brick for a POE switch...

The poor TS100 now doesn't power on and apparently C26 and D3 are now broken. Does anybody know what else can be broken?

Can anybody tell me where to find the components in the photos?

[image]
[image]
(There is a crack in this component)

Thank you all.

whitehoose, 2020-06-08:
Lol - been there done something very similar.  I thought it was just me :D
If you find the mini site there is a support page.  While they don't seem to sell spare parts on the site  When I fried my TS80 I e-mailed support and told them what I'd done and asked for a price to buy a mobo and post it my way - they replied, said they wanted to see the damage and told me to post the mobo with a note of my address - I sent it airmail, they kindly posted back a replacement mobo by return for free.  It's worth opening a dialogue.

Faulting the board is just a matter of tapping a meter round the board.  Start at the battery jack and hope the resistors blew fast enough to be useful.  If you don't have smd kit - it's possible to use through hole to prove and then order the bits (ebay do a selection book of 2000 resistors for pennies).

Failing that the price of the new ones is dropping - on YT learn electronics just reviewed an excellent  clone from sequre - It's worth looking on their website - there are some interesting options.
 https://www.youtube.com/watch?v=DnjLT49KQU0

chuckyto, 2020-06-13:
> Lol - been there done something very similar. I thought it was just me :D
> If you find the mini site there is a support page. While they don't seem to sell spare parts on the site When I fried my [...]

Thank you, whitehoose. I tried contacting the seller but they say it's out of warranty. I will try to fix the board before contacting sequremail or maybe get a new ts100... who knows?

whitehoose, 2020-06-13:
Its a real oversight by mini that they don't sell spares, conversations here seem to show the OLED is failing on some TS100 irons, if the 100 is failing its only a matter of time before the 80s follow.  And from these forums its plain that many of the users are experimenting with their kit.

Might not be a massive market - but if you have a few tips - you'd be very unlikely to damage the case - so mobo, top board for the 100s and screens for both could be sold through the usual outlets as spares.  If they are priced right, you cant build a cheaper iron from bits - but a repair shouldn't involve buying all new.  It's like having to buy a new car because you have a flat tyre
I'd be a bit miffed if mine was faulty and the only option was to start again.

In fact I'd jump ship on principle.  There are lots of options - many of them cheaper.

The sequre 001 is apparently compatible with ralim's firmware - so their model is obviously a faithful copy of the 100's innards.  The case is slightly different - but the included cords and connector options are imho better than those included with the original - and they sell 2 cunky PSUs and have a PD option as well.

Ralim, 2020-06-14:
@chuckyto I would reach out to miniware first; they tend to operate on the model of being very helpful; but they delisted spare parts due to companies like Sequre ripping off their products.

I advise against Sequre generally as they are not a known quality; as the design internally is a clone the usual warnings of clones applies (Who knows quality batch to batch).

The official units are definitely more pricy than the alternatives.
_And who knows, there could be other interesting options coming soonish for less 😉 _

At this point i would contact the service@ email with your issue and see what they say. A replacement mainboard is fairly easy to install and almost certain they will either give you one for free or at least for quite cheap.

@whitehoose 
I agree with the jump ship on principle; except i would jump to anyone but Sequre since they are selling a clone using the official firmware. Which has made life hard for future products since now they all come with flash lock bits set (Currently fighting this now on TS80P 😠 ). 
I dont get why they didnt just make their own bootloader to avoid issues 🙄 .

juankr10s, 2020-07-12:
is possible replace smd components? is very easy

whitehoose, 2020-07-12:
i Ralim, never got a flag you'd replied.
I understand your reservations, seems the clone situation is pretty standard in that neck of the woods.  On some levels the poetic justice watching the dragon eat its self makes me smile.  I'm not promoting the sequre kit, and I've resisted buying but I know a couple of ppl who have and have seen the innards.  It looks like a satrday night run on the same production line, the quality is good and the tips are direct from the same sources you see on the usual sites.
I do think the options it opens is good, Mini make some great kit but they know their target market have bottomless pockets, I can't help feeling they are riding the gravy train and exploiting the audience.  Maybe this will shake the tree a little.

closed 2020-09-14 by Ralim