Ralim/IronOS #653: [Feedback] TS80 compatible 65W power supply

nqtronix, 2020-06-13:
I've finally found a USB power supply that works with the TS80 and the 12V mode flawlessly.

- Name: xiaomi USB-C charger 65W
- Model number: CDQ07ZM 
- purchased here: https://www.aliexpress.com/item/32886399121.html

I'd recommed a silicone cable, but because there is no USB-C to USB-C variant available, you'll need a USB-C to USB-A adapter. On the power supply side, the cable must be oriented in the correct direction, otherwise it won't work.

---

I've measured the following heat-up times:

| 9V | 12V
--- | --- | ---
20->310°C | 23 | 13
130->310°C | 13 | 8

---

Originally I wanted to post this into the TS80 wiki, but I didn't figure out how that works (sorry).

discip, 2020-06-24:
@nqtronix 
Good afternoon sir,
you need to put it like this:

`||9V|12V`
`|-|-|-`
`|20->310°C | 23 | 13`
`|130->310°C | 13 | 8`

to make it look like this:

||9V|12V
|-|-|-
|20->310°C | 23 | 13
|130->310°C | 13 | 8

regards

mmehrle, 2020-08-05:
How about this one?

https://www.aliexpress.com/item/4001211694708.html

nqtronix, 2020-08-06:
Most likely it does not work. SUPERVOOC does not work like QC3.0 which is needed for the TS80. The other two power supplys likely onle use USB-C PD for the high power port, QC3.0 is only supported by the USB TypeA ports, which hast a lower poer output.

This https://www.aliexpress.com/item/4001238881915.html might work, it can output 30W to the USB TypeA and the iron needs about 32W to heat up. Obviously I've only tested the one linked in my initial post; if you decide to try any other charger please update us :)

mmehrle, 2020-08-06:
This is really strange. I just tested the two chargers I have. 

This one is supposed to be 30W - it says 43W but I have an older model from Spain which says 30W: https://www.amazon.com/AUKEY-Charger-Foldable-Compatible-Samsung/dp/B01BBZ5LT2/

Anyway, it barely gets to 250C and then stops although it's rated for QC3.0.

But I also have this one:

https://www.amazon.com/AUKEY-Charge-Charger-Compatible-Samsung/dp/B01BBZJ31Y/

And this one is rated QC3 and punches up the temperature relatively quickly - in about 30 seconds or so. 

Apparently not all QC3.0 chargers are cut from the same cloth. But $19 in the US for the one that works ain't bad.

mmehrle, 2020-11-27:
Quick follow up on my previous  comment. After a few uses the AUKEY charger started to act erratically and eventually one of the outputs just died. 

https://www.amazon.com/AUKEY-Charge-Charger-Compatible-Samsung/dp/B01BBZJ31Y/

So not one I recommend. I gave up and ordered the one that nqtronix recommended over at Aliexpress.

closed 2021-04-05 by Ralim