Ralim/IronOS #654: Use TS100 tips on TS80

Depau, 2020-06-14:
I'll just not use the form since this is not really an issue nor a feature request (or maybe not just yet).

Since I noticed prices for TS80 tips are getting quite high at ~20$ on AliExpress, while TS100 tips are ~12$, and [according to Louis Rossmann](https://www.youtube.com/watch?v=ao39bPEyok4) Hakko tips which are even cheaper work fine with the TS100, I was wondering whether it was possible to build an adapter to connect them to the TS80 and save quite some money.

Since I have basic understanding of electronics and zero understanding on how any of these tips work, I thought I'd ask here.

For the adapter, I thought I could just make a separate body with my 3D printer in ABS in which the tip fits, embed a design similar to the [Otter Iron](https://github.com/Jan--Henrik/Otter-Iron) to connect to the tip, then run some wires and connect it to the TS80 with a regular headphone jack.

To thermally insulate the 3D printed part, I think some heat resistant silicone will do the job, as long as the ABS part stays under 105° it should not warp.

I saw #457 and I was wondering whether there is any electrical incompatibility that can't be solved by just patching the firmware.

Ralim, 2020-06-15:
Hello,
Indeed your welcome to scrap the form if it doesn't suit. Its main purpose is to encourage information people sometimes forget :)

So the TS80 cant use the TS100 tips unfortunately because the TS100 and Hakko tips are ~ 8.5 Ohms; where as the TS80 tips are ~ 4.5 Ohms. You could mechanically adapt them but then your power would be dramatically reduced which wouldn't work well on the larger tip sizes.

Just using ohms law here should tell you the reasons why :)

So electrically you _could_ but I wouldn't as the power output would be too low.

Depau, 2020-06-15:
> Just using ohms law here should tell you the reasons why :)

Okay so it's *that* simple.

Got it, thank you! I think I'll consider getting a TS100 or make an Otter iron, since just getting enough TS80 tip cartridges can easily get more expensive.

closed 2020-06-15 by Depau