Ralim/IronOS #656: J02 + other tips?

guerillaengineer, 2020-06-15:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [X ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
I'm looking for an option to set the tip as J02, but can only find B02 and D25

* **What is the current behavior?**

* **What is the expected behavior?**

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1.
2.
3.

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information**

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2020-06-16:
Most likely you are on a much older firmware as the tip selection menu was removed 2 ish firmwares ago at least ?

guerillaengineer, 2020-06-16:
Yes you were correct, I had downloaded the newest one but pushed an older one to my iron. I thank you for your hard work and sorry to bother you~!

closed 2020-06-16 by guerillaengineer