Ralim/IronOS #657: Soldering mode (seems) not working

bodays, 2020-06-16:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
A bug

* **What is the current behavior?**
When I push the button near to the tip, it shows quickly the temperature (around 20/30 C°) and then it goes back to the initial menu... The tip isn't heating

* **What is the expected behavior?**
As I saw in few videos, it should start increasing the temperature till the one I have defined in the settings (300C°)

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
I just upgrade the stock firmware to the ralim one (v2.10) with no issues

***Steps to reproduce:***
N/A

***Video of problem if hard to reproduce***
https://www.youtube.com/watch?v=jZtgLdcJEvc

* **What is the motivation / use case for changing the behavior?**
N/A

* **What are you running:**
Ralim Firmware v2.10

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.10
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :19v - 2100mha

* **Other information**

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2020-06-16:
Heya,
Usually the screen fading out like that is an indication of the power supply to the iron dropping out under load.
Could you please test the iron on a known good power source that can supply the current?

At 19V you need at least 2.3A+ to be able to run the iron correctly without serious power dropouts. 

If the iron works on another supply, you can try using the wattage limiter to reduce the average power draw of the iron to help your power supply cope.

bodays, 2020-06-17:
Hey Ralim,
Thanks for the feedback.
You were right, I have tested with my laptop power spply (19v 4.74A) and it works like a charm... Heating (very) quickly.
I will look for but in the meanwhile, could you please tell me if there is a place explaining each of the parameters of your firmware please ?
Thanks again for your help Ralim

bodays, 2020-06-17:
One last question : at each voltage should I reduce it for my (bad banggood) power supply ?
Thanks

Ralim, 2020-06-17:
> place explaining each of the parameters of your firmware please ?
Will probably be some better docs, but if you pause on each menu item; it should scroll a description to make it a bit easier to understand.

> at each voltage should I reduce it for my (bad banggood) power supply ?
If your power supply is adjustable voltage you could try turning it down or you could try setting the power limiter option. However, often a bad psu isn't great with the spikes of power the iron still needs.

Glad to hear it is working :) The firmware is more aggressive than the stock firmware so it will burn a bit more power but also lets you solder much bigger joints.

bodays, 2020-06-17:
Many thanks again !!

closed 2020-06-17 by bodays