Ralim/IronOS #658: Display content not centered anymore on TS100

fiechr, 2020-06-18:
I've just flashed v2.10.1 onto my TS100. Everything seems to work fine, but all symbol and texts on the screen are shifted to the bottom right (see pictures). On some screens the text on the left is cut off and there is some kind of dot on the right.

[image]

[image]

I did not configure anything, so this is the default state after factory reset.

I don't know what hardware revision my TS100 is, but I can find out, if necessary.

paulfertser, 2020-06-20:
The dot on the right is actually a scrollbar indicator. What version is centered properly on your hardware?

fiechr, 2020-06-20:
v2.09 didn't had that issue.

paulfertser, 2020-06-20:
Photos I find on the Internet show bottom-aligned (not centered) on factory firmware too. Can you please retest with other firmware versions and make some photos to make sure it can indeed be different? The OLED datasheet says it has 16 pixels height and your photo shows all 16 pixels in action so I can't see how it can work any other way.

paulfertser, 2020-06-20:
I see how Temp change short and Temp change long are truncated, that's a bug all right.

paulfertser, 2020-06-20:
Please make photos of other screen anomalies, so far I do not see any.

fiechr, 2020-06-20:
OK, I will downgrade and report back.

fiechr, 2020-06-20:
You are correct! The position of the text is the same with the older version.

The only real bug seems to be the truncated text for these two strings. I couldn't find any other errors. I'm very sorry for the confusion! :-(

I understand now the purpose of the dot, but I'm unsure if it is actually useful. But that's just me.

PR #659 (2020-06-21): Fix "Temp change short/long" menu display
The values are only two digits so do not overwrite the end of the menu
item string.

Fixes: #658.
commit 021b8e2f (2020-06-21): Fix "Temp change short/long" menu display

The values are only two digits so do not overwrite the end of the menu
item string.

Fixes: #658.
closed 2020-06-21 by Ralim