Ralim/IronOS #662: 2.10.1 shows wrong temperatur

skoehler, 2020-06-24:
I just updated by TS100 to version 2.10.1. Now, after powering on, the temperature shown is 520°C. The tip is cold!
When powering on after the update, all settings were reset.

I was running the Ralim firmware before and the soldering iron was working just fine. But I don't know which version I was using before.

skoehler, 2020-06-24:
I'm using the TS-C1 tip

skoehler, 2020-06-24:
v2.05.01 works

Ralim, 2020-06-24:
Uhhhh.... That is _very_ weird I have to say.

Almost feels like your iron has slower timings than normal. Out of curiosity could you use the debug menu and tell me what the tip raw measurement says in that menu with tip in and tip removed when on the new firmware ?

skoehler, 2020-06-24:
My TS100 is from April 2018, bought it via Banggood. I will supply you with the values from the debug menu.

skoehler, 2020-06-24:
````
no DC power, only USB power

TS100 firmware v2.05.1
debug menu shows:
  v2.05 PCB2
  TS-C1: Rtip=1193 Ctip=10
  TS-BC2: Rtip=1182 Ctip=8
  no tip: Rtip=16379 Ctip=470

TS100 firmware v2.10.1
debug menu shows:
  v2.10 git
  PCB2
  TS-C1: Rtip=830 Ctip=6528
  TS-BC2: Rtip=700 Ctip=6529
  no tip: Rtip=10407 Ctip=6980
````

skoehler, 2020-06-24:
Does the provided information tell you anything?

Ralim, 2020-06-25:
A fair bit of head scratching in confusion :confused: .
Ill look at it a bit tonight :)

Firebie, 2020-06-25:
Ctip = (Rtip - TipOffset) / TipGain
May be TipOffset is negative somehow?

Ralim, 2020-09-14:
@skoehler 

Apologies for this getting lost.
Is this still occuring for you with latest firmwares?

skoehler, 2020-09-14:
Version 2.12 still shows 520 °C when the tip is cold.

Ralim, 2020-09-14:
It should only show that when the tip is disconnected :/ 

Have not been able to reproduce this at all across an entirely new set of tips..

Out of curiosity if you run the calibration when cold what number does it show you ?

skoehler, 2020-09-15:
I will try the calibration. But it seems that my Ctip and Rtip values are completely bogus, even when no tip is connected. See my post above.
Do you think calibration will help?

Ralim, 2020-09-15:
Your rtip values are reading lower then normal, which most likely means that your iron does not use the exact same op-amp part as other irons, but with calibration _should_ be in range as that _should_ derive a lower calibration number for you that should compensate i believe.

skoehler, 2020-09-17:
I tried v2.12-beta, but the calibration changes nothing!
The iron shows four dots and then shows 781 next to some box. Seems like that is the new calibration value. Then, after existing the menu, the iron still shows >500°C. I'm confused.

Ralim, 2020-09-17:
The calibration of 781 make sense, that is close to your offset.
If you power cycle after the calibration is it still showing >500C when the tip is connected?

skoehler, 2020-09-17:
Yes, even after a power cycle, the TS100 still shows >500°C. The TS-BC2 tip was connected the whole time.

tttunci, 2020-09-17:
Hello, I think I am having a similar issue. I happily used your firmware for a day however while I was using the soldering iron today, it started to not heat up and showing 498°C. While I was trying it to get it to work, it did some spikes where it actually got hot enough to melt the solder. But these spikes only happen for few seconds then it drops its temperature. Would appreciate it if you can give me some ideas. 
After I calibrated it, It heated to 490+°C again where it was able to melt the solder then droped its heat while still displaying 490+°C for the tips temperature.

skoehler, 2020-09-17:
As you can see, a downgrade to v2.05.1 fixes it for me. Have you tried that?

tttunci, 2020-09-17:
I just tried it and it gives me the BAD tip error. I just assume my tip somehow got damaged. Thank you for your quick response.

dodox1, 2021-01-01:
I have ts100 pcb version V2.5**B**. There is R1 value 850k, unlike version 2.5**A** where the value is R1 750k.
Do you think this should affect the temperature reading?
My temperature was shifted a lot.
set - measured on FG100 (tip gain default, tip BC2 and )
330 - 300
300 - 266
250 - 225
200 - 182
If I set tip gain to 24 it seems to be correct.
(version 2.11 used)

Ralim, 2021-01-01:
Ah heck,

Absolutely that will completely change the response of the unit. This is going to be a mess if they have changed a value again without warning.

Where did you buy yours from ?

dodox1, 2021-01-01:
OMG, sorry it is EIA-96 marking :see_no_evil: so the R1 value is the same 85D = 750k 
(It was from aliexpress in 2019, but seems to be original package)

Firebie, 2021-01-01:
Did you try latest 2.13? It uses lookup table instead of linear convertion for tip temp

dodox1, 2021-01-02:
TS100 V2.13 test with tip BC2
(room temp calibrated)
set - measured
200 - 197
220 - 218
240 - 238
250 - 248
280 - 279
300 - 300
320 - 322
( I'll make test with B2 )

Ralim, 2021-01-03:
@dodox1 those are fairly close. The system has about 1.5-2% max achievable accuracy without much more calibration that is really not worthwhile.

discip, 2022-04-23:
@skoehler
Do you still experience this issue?
Please close this otherwise. 😊

thanks

skoehler, 2022-04-24:
I will test and inform you.

closed 2023-07-21 by Ralim