Ralim/IronOS #663: Quick Charge 4.0+ power supply not working

VanKurt, 2020-06-24:
* **I'm submitting a ...**
Bug report

* **What is the current behavior?**
I've bought a Belkin Quick Charge 4.0+ 27W power supply. QC 4+ should be fully backwards compatible with Quick Charge 3.0, so I was expecting it to work with the TS 80 soldering iron. But the TS80 stays fully dead when I connect it to the supply (display does not turn on).

The power supply and cable seem OK, since I can charge my smartphone and camera witout any problems. The TS 80 seems to be the culprit here.

Link to the power supply: https://www.belkin.com/de/p/P-F7U074/

* **What is the expected behavior?**
I expectd the TS80 to turn on and heat up when connected to the power supply.

* **What are you running:**
  - Firmware Version: 2.10.1
  - PCB Version: ? 
  - Power Supply (Voltage and Current Rating) : Belkin QC 4.0+ 27W

Ralim, 2020-06-24:
Most likely the issue here is that you are using a USB-C to C cable; these only work with the TS80P. This is a hardware flaw in that the iron is missing the pulldown on the CC pin that is required for a USB-C power supply to turn on.

You can buy USB C-C cables with this resistor in them (designed for this exact situation).

DjordjeMandic, 2020-06-24:
> Most likely the issue here is that you are using a USB-C to C cable; these only work with the TS80P. This is a hardware flaw in that the iron is missing the pulldown on the CC pin that is required [...]

Is it possible to bodge this resistor inside ts80?

Ralim, 2020-06-25:
@DjordjeMandic Indeed that is correct; for non-pd aware you should have a 5.1k resistor to ground on each CC pin.

It falls in the category of "you can, but its very fine soldering" as you would be soldering to the back of the pins on the CC socket. So generally I dont recommend this.

closed 2020-07-22 by Ralim