Ralim/IronOS #664: Missing Tip Selection

fanatictentacle, 2020-06-29:
* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**
In the settings menu and submenus there is no "Tip Model Selection"

* **What is the expected behavior?**
There should be a menu for that regarding to https://github.com/Ralim/ts100/blob/master/menu.md

***Steps to reproduce:***

1. Plug-In the iron
2. Go to settings
3. Scroll through settings and submenus
4. Dont find the Tip Selection option
5. Go to a corner and cry

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

  - Firmware Version: 2.10 git (DE translation)
  - PCB Version: 2
  - Power Supply (Voltage and Current Rating) : 24V 3.8A

Ralim, 2020-06-30:
> Go to a corner and cry
Sorry man :(

Tip model selection was removed and I forgot to update that file :(

fanatictentacle, 2020-06-30:
Okay thanks!
Maybe the corresponding lines of code in the translations should be deleted too. Because I have looked into that to see if maybe the translation was missing something
btw: didnt really went to a corner to cry, but thanks for the sorry ;)

closed 2020-06-30 by fanatictentacle