Ralim/IronOS #666: Blank Screen (Video Included)

wespyen, 2020-07-07:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**

* **What is the current behavior?**
Blank Screen on boot.

* **What is the expected behavior?**
Logo and menu options to be viewable on boot.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. I plugged my TS100 into my PC while pressing the A button. That shows a full LCD panel reading "DFU: 3.45"
2. I downloaded the TS100_EN.hex file and moved it into the drive that the iron is plugged into
3. The hex file name changes to TS100_EN.RDY.
4. When I unplug the usb cable and plug in my 12V power supply, the screen only shows some of the dots lighting up.

***Video of problem if hard to reproduce***
[Here](https://youtu.be/GhmdzyoRVwM) is a video of the behavior after copying over the firmware file.

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information**

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2020-09-14:
Hi @wespyen 

Could you confirm if this issue still occurs on the latest release?

marcos-arsoares, 2020-10-06:
I do have this issue on the latest release.

paulfertser, 2020-10-07:
This video is actually telling an interesting story:

1. The vendor bootloader has unstable control over OLED, as each time
it runs the display is garbled in different ways;

2. This firmware is communicating with OLED almost properly, the
firmware version shown for a split-second is a good evidence for that.

Please try
https://github.com/Ralim/ts100/suites/1300796821/artifacts/20410913
that was auto-built from my test "github pull request" and see if the
behaviour is any different.

Ralim, 2020-12-29:
@wespyen 
@marcos-arsoares 

Have you been able to test @paulfertser's version to double check?
Not much has changed for this really; most likely part here is due to the Miniware code using a software i2c while this code uses the hardware i2c peripheral; and as such requires that ack's etc are correct.

This should not normally cause issues at all, though on dying screens it _may_ cause them to not work while they "sometimes" do on the original firmware.

SuperPrower, 2021-02-19:
@paulfertser , I am experiencing similar issues on the latest build of Iron OS (some dots and then blank screen, but Iron gets hot), plus graphical artifacts on the stock firmware (after returning to standby mode, sometimes). I tried building from source with reduced I2C speed (75000 -> 50000) as in your commits, but I still only saw dots on the screen at boot and part of some image for a split second. I tried reducing it further to the 25000, but got no image at all. My TS100 is recently purchased from AliExpress.

discip, 2022-04-23:
@wespyen
Is this issue still a thing?
Otherwise please close this. 😊

thanks

NoSkillSammy, 2023-02-15:
I've tried everyone singles person idea forwards and backwards, nothing , absolutely nothing. Tried looking for a cheap replacement board off ebay but seemed like a risk so i just bought a whole new $80 ts100!!!! yay open source firmware :(

Ralim, 2023-02-15:
How old was your iron?
Blank screens can be caused by a few things as mentioned or age.
Did you test with the miniware firmware (as it does different workarounds).
Did you open your own issue to help us diagnose what the issue is?

closed 2023-07-21 by Ralim