Ralim/IronOS #669: Firmware 2.10.1 stuck on 25C

bigadz87, 2020-07-18:
***I'm submitting a ...***
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

***What is the current behavior?**
After testing a brand new Ts100 with official 2.18 software, I flashed 2.10.1 onto my device. The iron allows me to modify settings, set temps etc however the temp stays at 25C and does not increase at all.

I am using a Toshiba laptop dock 19v 6amp DC power supply. 

* **What is the expected behavior?**
The iron should increase and decrease in temperature as per desired settings. 

***If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Flash the device with latest firmware through DFU 
2. Power on device, device goes straight to displaying 25C
3. Hold B button to get to settings menu, set settings as I see fit
4. Return to soldering mode
5. Device stays at 25C, holding a button shows arrow (I'm assuming for boost) but also does nothing 

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.10.1
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) : 19.2V 6.2A

* **Other information**

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2020-07-22:
Just to check, does this unit work on the official firmware?
Could you try doing a reset of all settings in the advanced settings menu ?

bigadz87, 2020-07-23:
Yes works perfectly on the official firmware. I will try reflashing and performing a reset

closed 2020-09-14 by Ralim