Ralim/IronOS #671: Firmware does not work on SainSmart Pro 32 (TS100 clone)!

andreich26, 2020-07-23:
The firmware v2.10.1 is installed without errors, but the soldering iron does not warm!

Ralim, 2020-07-23:
Hello,

Can you provide some more details?
I do not own any clone units to test against so curious as to what is different on these.

Could you list out all of the items in the debugging menu?

andreich26, 2020-07-23:
Video https://youtu.be/D66w7igqEzo
the soldering iron does not turn off on the B button
the soldering iron does not heat

agatti, 2020-07-23:
That's odd, I've got the very same iron and it has always worked with this firmware for the past couple years...

Ralim, 2020-09-14:
If you are still looking to solve this:

If you load the miniware firmware onto the unit does everything work correctly?
Could you try the latest firmware to see if the issue still persists?

closed 2020-11-01 by Ralim