Ralim/IronOS #673: Solder Tip database

mikko70, 2020-07-26:
Beacuse there is official /with logo) and unofficial replacement tips (some unknown manufacturer). I hope some database identity and measure tip values , maybe firware can also warn if specs (recistance, temperature) go out of range.

Ralim, 2020-09-14:
The TS100 cannot measure the resistance sadly.
Even with the official tips resistance is +-1 ohm depending on tip model.
Almost all of the tips I've picked up so far have all worked totally fine. Though some of the unbranded ones have failed earlier.

Generally if your tip starts to go out of spec, you _very quickly_ realise as the tip stops working 😅 

Not against this though, I _think_ you could start this in the wiki if you desired?

ntindle, 2021-01-19:
Do you know if the Pinecil can measure resistance?

Ralim, 2021-01-19:
Pinecil does not have tip resistance measurement built in at the moment. I was voting in the favour of not including it due to it not being overly accurate anyway and more hassle to support than anything else.

River-Mochi, 2023-03-23:
@mikko70   willl you be making a database of tips in the wiki page,  otherwise consider closing this ticket since it's 3 years old.

The firmware can not measure tip tesistance  on most Irons because they lack the hardware.   even the newest Pinecil V2  although it has hardware to measure resistance, it's noisy  so it just uses this feature to toggle between  showing people if they have a short tip (6.2-6.5 ohms)   or a normal  tip (7.8 ohms - 8.3ohms). 
IronOS uses the Pinecil V2  hardware as toggle and doesn't  display accurate resistance because of limitations.